wled/WLED #95: 'NeoEsp8266Uart800KbpsMethod' was not declared in this scope

Beliar-666, 2019-01-01:
Hi,
I just Updated Boardinformation and some Libraries to newest Builts and now i get:

`In file included from C:\Users\MatzeB\AppData\Local\Temp\arduino_build_119662\sketch\WS2812FX.h:42:0,

                 from C:\Users\MatzeB\AppData\Local\Temp\arduino_build_119662\sketch\WS2812FX.cpp:42:

NpbWrapper.h:27: error: 'NeoEsp8266Uart800KbpsMethod' was not declared in this scope

   #define PIXELMETHOD NeoEsp8266Uart800KbpsMethod
`

Whats wrong here?

Beliar-666, 2019-01-01:
Changed #define PIXELMETHOD NeoEsp8266Uart800KbpsMethod to #define PIXELMETHOD NeoEsp8266Uart1800KbpsMethod

and it works again. Seen on Discord

Aircoookie, 2019-01-01:
I'll reopen this for now so other users can find the fix more easily until I release the next version.

closed 2019-10-25 by Aircoookie