wled/WLED #101: Sample for Macro

RaiderXXL, 2019-01-25:
Please,
provide me a few samples for macros.. dont know how to do marcos (Wiki dont help me) ;)
possible to do a reboot at 3:00am or like that...

Thanks a lot...

Pascal

Mariu86, 2019-01-25:
Will be a great help , I don't know how to do it.  Aircoookie can you help me with a big problem :D , it is possible to set wled to search for wifi every hour or so?

Aircoookie, 2019-01-25:
No problem, I realize that macros (as well as presets and sync) aren't really intuitive currently. Basically, you just set your action in one of the 16 textfields. For on, you can type in "T=1", for example. Then, you use the index number next to the textfield you used to activate the macro (or "0" to have no macro). Here is a little example that will turn the light on a minute before 7pm, turn it blue at 7pm and off a minute later. Don't forget to enable "Get time from NTP server" at the top of the time settings page!
Rebooting via the API is not yet supported, but I could add that.

@Mariu86 what do you mean by "search for wifi every hour". Do you mean that your setup doesn't reconnect to your WiFi automatically? I've noticed that too sometimes and will add some code soon that tries to reconnect if the connection fails. For now it may help you to set a static IP for the device.

[image]

Mariu86, 2019-01-25:
Thanks Aircoookie! My problem it's my wife she unplug the router when she go to sleep some times, if wled do not find wifi it's going to API and if I plug back my router the wled remain in API mode, I must do a restart to connect back to the wifi. Hope you understand my English :D

Mariu86, 2019-01-25:
If wled will search for wifi every hour and connect to my router will be a very good thing for me. Thanks again Aircoookie for everything!

RaiderXXL, 2019-01-27:
Thank you!

Thats easy...

Pascal

closed 2019-10-02 by Aircoookie