wled/WLED #102: ioBroker Adapter

Aircoookie, 2019-02-04:
Hi again, @DutchmanNL!

I would also choose the HTTP API option. UDP broadcast is not as reliable and won't provide feedback on whether the command was successful to ioBroker, but is very fast. MQTT is not bad as well, but it requires user configuration of WLED settings (adding the broker IP), and keeping the connection, which might make things more unstable, so HTTP is probably the way to go.

About JSON, I was planning to add an API for it anyways (it is useful for the new segment feature I plan on adding soon-ish), but I might need a little time to implement it properly. https://github.com/bblanchon/ArduinoJson seems like the go-to library for this job!
What values would be useful to you to have in the initial JSON response?

On a side note, have you heard of mDNS/Bonjour/Zeroconf? I don't know the capabilities of ioBroker, but it is supported by WLED and allows for automatic device discovery (my Android app uses it, for example). Of course, just having the user enter the IP address is an option as well.

Looking forward to working on this together :)

DutchmanNL, 2019-02-04:
thank you !

Yeah look for the integration part I am thinking on 2 ways, one would be a dedicated adapter for WLED and the other way an integration to the existing multicast adapter I am currently working on.
The benefit of UDP is the package size and little overhead on the network. About reliability, yea as far I see in practice is very stable and sometime (1 of > 100) requests fails. But here we build and error loop to handle the situation by ensuring send commands are acknowledged and if not the message is send again.

For the JSON, yea that's the way to go we did the same with temperature settings looks very well.
Need to thin about the values, give me some investigation time for that.

Bonjour is an interesting part, thank you pointing it out!
Currently we do not have an bonjour "listener" adapter as far I know, we are able to emulate a device with Yahka adapter which enables adoption of state from ioBroker into apple HomeKit.

@andiling

DutchmanNL, 2019-02-05:
Hi, 

I had a closer look to the structure and how we could proceed.

In basic I would like to handle it by broadcast that would make adoption very easy!

My basic idea :
- I would like to have all actions controllable by API as an state object within iObroker
- this includes all post request which can be handled by http APU
- WLED announces itself by UDP which allows iObroker to automatically adopt the device and required states

Issues I see, or parts we need to discuss :
- having all states is great, but I will need the current values. 
- we could send all values by JSON within an UDP broadcast package.
- if that's not possible we could use the UDP announcement of change to trigger an http get receiving all values

Sending would run the same, a state change in iObroker would trigger an UDP broadcast (or for time being http post action).

For this it's important to have feedback, so sending a command should be committed by an OK or NOK, NOK would trigger resending the command again.
This can be handled by an timeout loop or by comparing the new received UDP message with the current data.

So basic question would be : 

1) would it be possible to have all possible switch values exposed as JSON ?
2) would it be possible to send it by UDP package
3) if not, is it an idea to let the UDP package trigger a http get to receive the JSON

If 3 is the option, we need to discuss the layout of JSON to have all relevant information for the post request in one message.

Happy to hear from you, this will be awesome!
I already managed to have an "quick and dirty" implementation which allows me to control all lights based on my IIT sensor (temperature, doorbell and all this kind of funny stuff).

Btw, one additional question : what do you think about making separation of led stripes possible within WLED ?
The library is able to handle separation if that could be done by web interface (or states/http post) it would be great.
Thinking about having like 6 5m stripes in 1 room but than separating them to have individual control capabilities

raymiec, 2019-02-06:
something like MDNS.addService("xxxx", "tcp", xx); will work for advertising the service on the network. Think the required library is already included. I more or less have a native android app / controller finished. I'll upload it and you can pull the discovery function i wrote if needed. OTA upload of a config file Wled could parse would be sweet too, would finish my project nicely.

DutchmanNL, 2019-02-20:
sorry for the long Delay i will come back the Details later this week.

Just as a Small. question for now : the JSON Looks cool, how are you processing with the Separation idea ?

Aircoookie, 2019-02-20:
Awesome! Don't worry, I'll still need some time as well for the JSON API. Segment support will be even further in the future. 

There's another user who requested driving two strips with mirrored output. That's really easy to implement, so I will provide an option for doing so really shortly.

DutchmanNL, 2019-02-20:
> There's another user who requested driving two strips with mirrored output. That's really easy to implement, so I will provide an option for doing so really shortly.

cool ! If you need any assistence with the JSON Part please give me a shout. i know a Person who is quite familiar with it if you need a helping Hand

Aircoookie, 2019-03-06:
JSON API is now added! (although HTTP only, I still have to add it via UDP as well as create a doc)

If you want to check it out, install the latest master!
- `[IP]/json` returns everything
- `[IP]/json/state` returns state object (every value updatable)
- `[IP]/json/info` returns device info (no value updatable)
- `[IP]/json/palettes` and `[IP]/json/effects` return palette and effect name lists.

To update the state object, just send a POST request to `[IP]/json` or `[IP]/json/state` with the values you want to change (for example `{"bri":255,"seg":[{"col":[[0,255,200]]}]}` for maximum brightness teal)

debsahu, 2019-03-06:
This is great, adding to Home Assistant will be working much easier. I saw you were working on JSON API, is it available for MQTT interface as well?

meelow, 2019-03-06:
+1 for mqtt interface ;-)

Aircoookie, 2019-03-06:
Yes, I will make it available via MQTT and via UDP :)

DutchmanNL, 2019-03-07:
> If you want to check it out, install the latest master!

great, I tried to build the master branch but looks like we have an issue here :

`Warning! Library `{'requirements': '1.2.0', 'name': 'EspAsyncWebServer'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': '1.2.0', 'name': 'EspAsyncWebServer'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.`

`WLED/wled00/wled00.ino:42:31: fatal error: ESPAsyncWebServer.h: No such file or directory`

DutchmanNL, 2019-03-07:
was able to fix this issue by changing the name of dependency from 

EspAsyncWebServer@1.2.0

to

Esp Async WebServer@1.2.0

still compile fails :

`*** [.pioenvs/nodemcuv2/src/wled00.ino.cpp.o] Error 1`

/Users/rafal/Programming/WLED/wled00/wled00.ino:369:31: warning: right operand of comma operator has no effect [-Wunused-value]
IPAddress realtimeIP = (0,0,0,0);
^
/Users/rafal/Programming/WLED/wled00/wled00.ino:561:19: error: new declaration 'bool reconnectHue()'
}
^
```
/Users/xxx/xxx/WLED/wled00/wled00.ino:555:6: error: ambiguates old declaration 'void reconnectHue()'
DEBUG_PRINT("State time: "); DEBUG_PRINTLN(wifiStateChangedTime);
```

DutchmanNL, 2019-03-07:
> This is great, adding to Home Assistant will be working much easier. I saw you were working on JSON API, is it available for MQTT interface as well?

i woud suggest than to create an issue for HA ?
I cannot focus on both systems but basically what we do here (and no NOT MQTT!) is feasible for any automation system

DutchmanNL, 2019-03-07:
> JSON API is now added! (although HTTP only, I still have to add it via UDP as well as create a doc)
>
> If you want to check it out, install the latest master!
>
> * `[IP]/json` returns everything
> * [...]

Works like a charm good job !

> To update the state object, just send a POST request to `[IP]/json` or `[IP]/json/state` with the values you want to change (for example `{"bri":255,"seg":[{"col":[[0,255,200]]}]}` for maximum brightness teal)

i have to try this one. Basically I like the way of implementation basically I can always just send the complete JSON and the device will handle the rest that nice.
I will work on the adapter during the next week I got no everything I need to start working on it :)

Aircoookie, 2019-03-07:
Nice! I hope it will also stay stable for you. It gets tight really quickly on the ESP8266 heap-memory-wise, we are at a point where only the latest esp core will even work well. Need to look into optimizing the code so more heap is available

Aircoookie, 2019-03-07:
Also, just pushed the PIO fix :)

DutchmanNL, 2019-03-07:
> Also, just pushed the PIO fix :)

and it works like a charm thank you !

DennisH1978, 2019-04-19:
Hello there... Is this a New Adapter for controlling ws2812b Stripes? I read the whole Thread but I am not Sure if this Adapter is for ws2812b and if it is availble... Could you please tell me? Thx 😀

Aircoookie, 2019-04-19:
Hi @DennisH1978!
Yes, my software WLED can display various colors and effects on ws2812b strips. It is standalone but offers integration capability with other services via MQTT or JSON. There is already a Homeassistant autodiscovery integration, but no native adapter for ioBroker yet.

DutchmanNL, 2019-04-19:
Hey @Aircoookie 

How can we proceed on that ? I basically have an multicast adapter ready which offers an generic API for communication on multicast.
It expect a JSON in specific format to automatically create the states and return values.

We currently have an binary available which communicates on that principle with Vansware SmartHome plugs.
The adapter and binary are in beta-test tase.

The API is capable to :

- Handle initialization command
 - when an message is received an not state present, the adapter will request a complete overview of all states
 - sends feedback if all Data is handled
- create state structure with devices / channels and states
- each state can have a role, writable yes/false, min and max value configured
- on writable states each change is send by udp multicast to the partner
- if no "success" comes back the change will be send again with a maximum of 5 times to timeout

Beliar-666, 2019-07-08:
Hey Guys, any news about the Adapter?

DutchmanNL, 2019-07-09:
@Aircoookie any idea based on my previous suggestion ?

Beliar-666, 2019-09-16:
no news here?

Aircoookie, 2019-10-25:
So sorry for ignoring this for half a year! This thread totally fell under the table somehow...
@DutchmanNL sounds very good! If you are still up for implementing this, send me the JSON format you need for discovery, I should be able to add it as multicast!

DutchmanNL, 2019-10-26:
@andiling can you assist please

gallipower, 2019-11-02:
Sounds great! 
I have 6 wleds from 10 to 160 WS8212 working and controlled via wled-app. Awsome Work!

But also running all the rest of my home via iobroker and it would be great if an easy integration in iobroker via adapter is possible!

I attemted to run it via mqtt but i failed to bring it up running.

So, if i could provide help in testing the adapter please let me know!
(Further communication better in German because of my weak english!)

DutchmanNL, 2020-02-04:
stay tuned, integration will be ready within. the next 2 days alpha already in dev-test :D

Aircoookie, 2020-02-04:
@DutchmanNL awesome, good work!
By the way, I've created a new discussion topic on the brand new WLED forum for the integration!
https://wled.discourse.group/t/iobroker-integration/33

DutchmanNL, 2020-02-05:
https://github.com/iobroker-community-adapters/ioBroker.wled

beta ready, happy testing ! 👯‍♂

lorenz24, 2020-02-07:
Hi just one information.
I think the name of the field "seg.o.col.sx" should be speed and not ID of the effect.

Thanks you for creating this adapter for iobroker.

copperfield74, 2020-06-12:
@DutchmanNL Thanks very much for the adapter. But I get A LOT of warnings within IOBroker. They say:
(998) State attribute definition missing for + lor
(998) State attribute definition missing for + lip
(998) State attribute definition missing for + lm
Is there a fix for this or did I misconfigure something?

Thanks in advance.

Aircoookie, 2020-06-13:
@copperfield74 those are some newly added attributes to the JSON api. It's annoying that a warning is thrown for that. @DutchmanNL can fix it by including these attributes, but once I add more in a future version the issue would re-appear...

DutchmanNL, 2020-08-28:
fixed in 0.5.2, the warning is to indicate the developer about an issue will be moved to sentry in an later version and excluded from client logggin