wled/WLED #103: mds service type

raymiec, 2019-02-11:
Can you change  MDNS.addService("http", "tcp", 80);   to MDNS.addService("wled", "tcp", 80); 

Didn't realise this was actually in it already, couldn't discover with android NDP,

Don't think there would be any issues changing?

Changing would make it easier to discover from android and other devices if we are specifically looking for it, http as a service type will return all services on the network with service type http.

I can't think of any other way to just return wled devices when discovering on the network.

Thanks. 

Parsing the name is unreliable.

Aircoookie, 2019-02-11:
Good suggestion!
Today I will finally release 0.8.3. mDNS will then advertise _http._tcp as well as _wled._tcp.
I ran into the same issue when writing the app, my workaround was to send an API call and see if the response is valid, but I too realized it would be way easier to have a separate service type for WLED!

raymiec, 2019-02-11:
Excellent, i can fire an android app on the play store to will let you list and switch between devices.?

Aircoookie, 2019-02-11:
Sure! :) (Although my app https://play.google.com/store/apps/details?id=com.aircoookie.WLED pretty much does the same thing)

raymiec, 2019-02-11:
Haha it does so I never even knew you released one!   I have a few things that could be added to yours. I also have accounts / roles / permissions etc, i.e kids account can only control bedroom. You could also use it for premium features etc. That side of things does require an active internet connection / or host server on pi etc.

closed 2019-02-12 by raymiec