wled/WLED #104: Multiple led strip control from single ESP

N1nja98, 2019-02-11:
Wow!! I must congratulate you on your awesome work! It is absolutely amazing! I was working on a similar project when i stumbled across WLED. It had almost everything i was trying to make! 

One requirement of my project was to connect multiple led strips of various lengths to a single esp and control them individually or grouped. Would this be possible? If you could give me some pointers I could try and implement it myself. 

Thanks!

Aircoookie, 2019-02-11:
Hi, thank you! I'm very glad you like it that much! Multiple strips are a bit difficult, since there are only 2 pins available on the ESP8266 for "hardware accelerated" LED driving (software bit-bang can be unreliable), but I can consider rewriting the library a bit to at least allow those two strips to be connected.

On a related note, the next major feature I will work on is segment support! It will allow you to set different segments of the same strip to different colors, so you could also connect strips in series and control them separately!

N1nja98, 2019-02-11:
Segment support sounds like an awesome idea! 

I’m currently using an ESP32 on which I understand more pins can be used than the ESP8266 so the lack of pins would not be an issue on that part. 

The reason in my case for using multiple pins instead of one continuous strip divided into segments is that the strips are somewhat isolated from one another (but too short in length to add another ESP) and would require much more wiring to connect in serial.

Again, your work is greatly appreciated!

N1nja98, 2019-02-19:
OK... I tried adding a second strip to the code that would be controlled same as the original strip. But I noticed that whichever strip is initialized last, gets updated, the first strip initialized remains blank; even when I change the order I initialize the strips.  

I've tried and succeeded running 8 different strips at a time with the original WS2812FX library which uses the Adafruit NeoPixel Library so I assume it has something to do with NeoPixelBus. Is multiple strip output supported with NeoPixelBus? 

Thanks!

Aircoookie, 2019-02-19:
Yes, having two strips (one on DMA, one on UART) should be supported with NeoPixelBus.
Can you send me your modified NpbWrapper.h so I can take a look?

N1nja98, 2019-02-20:
Only modifications are I added a third argument for ledpin to Begin() and accordingly to WS2812FX::init(). Do also need to change the PixelMethod?

//this code is a modified version of https://github.com/Makuna/NeoPixelBus/issues/103
#ifndef NpbWrapper_h
#define NpbWrapper_h

//#define WORKAROUND_ESP32_BITBANG
//see https://github.com/Aircoookie/WLED/issues/2 for flicker free ESP32 support

//PIN CONFIGURATION
#define LEDPIN 2  //strip pin. Any for ESP32, gpio2 is recommended for ESP8266
#define LEDPIN2 12  //strip pin. Any for ESP32, gpio2 is recommended for ESP8266
#define BTNPIN 0  //button pin. Needs to have pullup (gpio0 recommended)
#define IR_PIN 4  //infrared pin.
#define AUXPIN 15 //unused auxiliary output pin

//automatically uses the right driver method for each platform
#ifdef ARDUINO_ARCH_ESP32
 #ifdef WORKAROUND_ESP32_BITBANG
  #define PIXELMETHOD NeoEsp32BitBangWs2813Method
  #pragma message "Software BitBang is used because of your NeoPixelBus version. Look in NpbWrapper.h for instructions on how to mitigate flickering."
 #else
  #define PIXELMETHOD NeoWs2813Method
 #endif
#else //esp8266
 //autoselect the right method depending on strip pin
 #if LEDPIN == 2
  #define PIXELMETHOD NeoEsp8266UartWs2813Method //if you get an error here, try to change to NeoEsp8266Uart1Ws2813Method or use Neopixelbus v2.3.5
 #elif LEDPIN == 3
  #define PIXELMETHOD NeoEsp8266Dma800KbpsMethod
 #else
  #define PIXELMETHOD NeoEsp8266BitBang800KbpsMethod
  #pragma message "Software BitBang will be used because of your selected LED pin. This may cause flicker. Use GPIO 2 or 3 for best results."
 #endif
#endif

//you can now change the color order in the web settings
#define PIXELFEATURE3 NeoGrbFeature
#define PIXELFEATURE4 NeoGrbwFeature

#include <NeoPixelBrightnessBus.h>

enum NeoPixelType
{
  NeoPixelType_None = 0,
  NeoPixelType_Grb  = 1,
  NeoPixelType_Grbw = 2,
  NeoPixelType_End  = 3
};

class NeoPixelWrapper
{
public:
  NeoPixelWrapper() :
    // initialize each member to null
    _pGrb(NULL),
    _pGrbw(NULL),
    _type(NeoPixelType_None)
  {

  }

  ~NeoPixelWrapper()
  {
    cleanup();
  }

  void Begin(NeoPixelType type, uint16_t countPixels, uint8_t ledPin)
  {
    cleanup();
    _type = type;

    switch (_type)
    {
      case NeoPixelType_Grb:
        _pGrb = new NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>(countPixels, ledPin);
        _pGrb->Begin();
      break;

      case NeoPixelType_Grbw:
        _pGrbw = new NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>(countPixels, ledPin);
        _pGrbw->Begin();
      break;
    }
  }

  void Show()
  {
    #ifdef ARDUINO_ARCH_ESP32
     #ifdef WORKAROUND_ESP32_BITBANG
      delay(1);
      portDISABLE_INTERRUPTS(); //this is a workaround to prevent flickering (see https://github.com/adafruit/Adafruit_NeoPixel/issues/139)
     #endif
    #endif
    
    switch (_type)
    {
      case NeoPixelType_Grb:  _pGrb->Show();   break;
      case NeoPixelType_Grbw: _pGrbw->Show();  break;
    }
    
    #ifdef ARDUINO_ARCH_ESP32
     #ifdef WORKAROUND_ESP32_BITBANG
      portENABLE_INTERRUPTS();
     #endif
    #endif
  }
  
  bool CanShow() const
  {
    switch (_type) {
      case NeoPixelType_Grb:  _pGrb->CanShow();  break;
      case NeoPixelType_Grbw: _pGrbw->CanShow(); break;
    }
  }

  void SetPixelColor(uint16_t indexPixel, RgbColor color)
  {
    switch (_type) {
      case NeoPixelType_Grb: _pGrb->SetPixelColor(indexPixel, color);   break;
      case NeoPixelType_Grbw:_pGrbw->SetPixelColor(indexPixel, color);  break;
    }
  }

  void SetPixelColor(uint16_t indexPixel, RgbwColor color)
  {
    switch (_type) {
      case NeoPixelType_Grb:  _pGrb->SetPixelColor(indexPixel, RgbColor(color.R,color.G,color.B));   break;
      case NeoPixelType_Grbw: _pGrbw->SetPixelColor(indexPixel, color);   break;
    }
  }

  void SetBrightness(byte b)
  {
    switch (_type) {
      case NeoPixelType_Grb: _pGrb->SetBrightness(b);   break;
      case NeoPixelType_Grbw:_pGrbw->SetBrightness(b);  break;
    }
  }

  RgbColor GetPixelColor(uint16_t indexPixel) const
  {
    switch (_type) {
      case NeoPixelType_Grb:  return _pGrb->GetPixelColor(indexPixel);     break;
      case NeoPixelType_Grbw: /*doesn't support it so we don't return it*/ break;
    }
    return 0;
  }

  // NOTE:  Due to feature differences, some support RGBW but the method name
  // here needs to be unique, thus GetPixeColorRgbw
  RgbwColor GetPixelColorRgbw(uint16_t indexPixel) const
  {
    switch (_type) {
      case NeoPixelType_Grb:  return _pGrb->GetPixelColor(indexPixel);  break;
      case NeoPixelType_Grbw: return _pGrbw->GetPixelColor(indexPixel); break;
    }
    return 0;
  }

  void ClearTo(RgbColor color)
  {
    switch (_type) {
      case NeoPixelType_Grb: _pGrb->ClearTo(color);   break;
      case NeoPixelType_Grbw:_pGrbw->ClearTo(color);  break;
    }
  }

  void ClearTo(RgbwColor color)
  {
    switch (_type) {
      case NeoPixelType_Grb:    break;
      case NeoPixelType_Grbw:_pGrbw->ClearTo(color);  break;
    }
  }

private:
  NeoPixelType _type;

  // have a member for every possible type
  NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>*  _pGrb;
  NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>* _pGrbw;

  void cleanup()
  {
    switch (_type) {
      case NeoPixelType_Grb:  delete _pGrb ; _pGrb  = NULL; break;
      case NeoPixelType_Grbw: delete _pGrbw; _pGrbw = NULL; break;
    }
  }
};
#endif

debsahu, 2019-02-20:
Interesting hardware method to achieve this https://youtu.be/oGB7Yrbeo_Y

Aircoookie, 2019-02-21:
@debsahu those are really cool!

@N1nja98 here you have a `NpbWrapper.h` capable
 of driving two strips: [NpbWrapper.zip](https://github.com/Aircoookie/WLED/files/2888988/NpbWrapper.zip). It just mirrors them for now, so it's should work just like if you were connecting two strips to the same pin in parallel.
Please make sure not to use more than ~200 LEDs per strip with this, since you will run into memory issues with the DMA method and high LED counts.
Also, the auto brightness limiter will only work for this if you divide the amp rating by 2, since it only calculates the draw for one of the strips.
Once I added segment support, the two strips could display different content, which would make this approach that's only a demo right now actually useful.

Mariu86, 2019-02-21:
Aircoookie , how many segments will support wled ? Can we make a nanoleaf? Thanks

N1nja98, 2019-02-21:
@Aircoookie Thanks for the help! I now see what I did wrong. I figured every WS2812FX strip created a new NeoPixelWrapper object when in fact you have to duplicate everything in the NeoPixelWrapper.
 
@debsahu I wonder if similar drivers are available somewhere.

N1nja98, 2019-02-28:
I found a solution for driving more than 2 strips on ESP32 using FastLED. I'm currently driving 4 strips of 284 leds each (total 1136) on 4 separate pins, addressed as one long strip.  It still uses NeopixelBus to set the pixel colors in the array but instead of using  NeopixelBus' show function I use FastLED's. FastLED's led array is just pointing to NeopixelBus' array. Not the most efficient solution but it suits my purpose. Here is the modified code if anyone's interested:

This is placed at the top of "WS2812FX_fcn.cpp":
```
//toggle for multi-strip
#define Use_FastLED //FastLED already included in WS2812FX.h

#ifdef Use_FastLED

#define LED_TYPE    WS2812B
#define COLOR_ORDER RGB //Actually GRB. Explained later.
#define NUM_STRIPS 4
#define NUM_LEDS_PER_STRIP 284
//#define NUM_LEDS NUM_LEDS_PER_STRIP * NUM_STRIPS

CRGB* leds;
#endif // Use_FastLED
```

Modified "WS2812FX::init":
```
void WS2812FX::init(bool supportWhite, uint16_t countPixels, bool skipFirst)
{
	if (supportWhite == _rgbwMode && countPixels == _length && _locked != NULL) return;
	RESET_RUNTIME;
	_rgbwMode = supportWhite;
	_skipFirstMode = skipFirst;
	_length = countPixels;

	uint8_t ty = 1;
	if (supportWhite) ty = 2;
	uint16_t lengthRaw = _length;
	if (_skipFirstMode) lengthRaw += LED_SKIP_AMOUNT;
	bus->Begin((NeoPixelType)ty, lengthRaw);	
	if (_locked != NULL) delete _locked;
	_locked = new byte[_length];

	_segments[0].start = 0;
	_segments[0].stop = _length - 1;

	unlockAll();
	setBrightness(_brightness);

#ifdef Use_FastLED

        //Pointer to NeoPixelBus led array	
        leds = (CRGB*)bus->GetPixels(); //GetPixels function added to "NpbWrapper.h"
	
	// tell FastLED there's 284 NEOPIXEL leds on pin 12, starting at index 0 in the led array
	FastLED.addLeds<LED_TYPE, 12, COLOR_ORDER>(leds, 0, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
	
	// tell FastLED there's 284 NEOPIXEL leds on pin 14, starting at index 284 in the led array
	FastLED.addLeds<LED_TYPE, 14, COLOR_ORDER>(leds, 1 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);

	// tell FastLED there's 284 NEOPIXEL leds on pin 27, starting at index 568 in the led array
	FastLED.addLeds<LED_TYPE, 27, COLOR_ORDER>(leds, 2 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);

	// tell FastLED there's 284 NEOPIXEL leds on pin 26, starting at index 852 in the led array
	FastLED.addLeds<LED_TYPE, 26, COLOR_ORDER>(leds, 3 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
#endif // Use_FastLED
}
```

 "bus->Show()" at bottom of "WS2812FX::show" is replaced with: 
```
#ifndef Use_FastLED
	bus->Show();
#else
	FastLED.show();
#endif // Use_FastLED
```

This function is added into "NpbWrapper.h"
```
uint8_t* GetPixels() 
  {
	  return _pGrb->Pixels();
  }
 ``` 

The FastLED latest release (3.2) doesn't work and will result in flickering. Use tag 3.2.6 or newer.

 My color order was wrong at first. I think it's because the order of color values in NeopixelBus' led array is the same as the color order you set for your leds, while FastLED's led array is RGB no matter what color order you set. So in my case with GRB leds, showing NeopixelBus' array with FastLED results in reversed green and red. So in the COLOR_ORDER definition for FastLED I put RGB instead of GRB to change them back.
 
 Also when exceeding the 600 led limit on the ESP32 you need to bypass the max led safety in "wled03_set" and "wled05_init". Not sure what the maximum led count is  using this method. I've been running 1136 leds for 12 hours straight without issues so far. 
@Aircoookie What issues are encountered when exceeding 600 leds with the original code?

niroth81, 2019-09-16:
I know this is an old Issue. But is it possible for you to create a binary that I can do an OTA upload with to include the strip segmenting? I'm not very good with updating these files and I've been at it for days without success.

Jafterdark, 2019-11-12:
I just seen this thought I'd leave it might be helpful
https://www.reddit.com/r/esp32/comments/bkyeq0/20000_ws2812b_pushed_at_130fps_with_esp32_and/

fijit586, 2019-12-04:
I too am looking for more than one data signal. I just posted a "Feature Request" about it. Segments could work for my situation as the 2 channels I would want would be located next to each other (windows). 
I am also going to put leds on my roofline and want my animations to emanate from the peaks. I could accomplish this by wiring FROM the peak and having 2 strings, one on each side of the roof, with both led #0 at the peak. But "segments" could make this easier to wire IF I can call out a pixel and have the animation start there and go down stream on subsequent pixels but reverse and go upstream on previous ones. Then I wouldn't have to start 2 strings at the peak. I could start one really long string at the bottom of the roof, on one side.

goatchurchprime, 2020-03-04:
I've heard that the state of the art with regards to fast controls is the [method on the teensy](https://www.pjrc.com/teensy/td_libs_OctoWS2811.html#tech).

I can't find the WLED code where the actual hardware accelerated pin driving is done, other than the setting "NeoEsp8266UartWs2813Method", but this leads to the [NeoPixelBus](https://github.com/Makuna/NeoPixelBus/wiki/Library-Comparisons) library which implements a huge number of different [hardware hacks](https://github.com/Makuna/NeoPixelBus/tree/master/src/internal).  

In an ideal world, shouldn't NeoPixelBus be used as an external library rather than copied in?

There is so much more to it than what you get in Micropython (my favourite firmware for ESPs), which is just a single unsophisticated bitbanging-timer-delay function.

Aircoookie, 2020-03-04:
Working on multi strip support right now! Promising nothing, but we might be able to get 4 outputs on 8266 and 8 on ESP32 😄

@goatchurchprime NeoPixelBus is used as an external library! I'd never be as good as Makuna with low level driving goodness. `NpbWrapper.h` is just for dynamically changing between RGB and RGBW modes (and soon LED pins and number of strips)

Timgerr321, 2020-03-16:
@Aircoookie I'm glad to hear that you're working on the multiple led feature. Do you have an indication on how long it takes before we can use it?

CaptainDario, 2020-03-18:
Yes, I would be very interested in a timeline too. Even if it would be very vague.
I could do some testing if that would somehow help?

Thanks again for the awesome software!

Aircoookie, 2020-03-18:
Hi, thank you!
I'm glad to hear that the multi-pin support is in high demand!
Can't promise anything right now, but very likely in software selectable LED pin in early and multiple strips in late April!

PaulNBerlin, 2020-03-20:
Hi @Aircoookie, will your multi strip approach also support different types of led strips controlled by the same esp? E.g. Led Pixel Lights (RGB in my case) and LED strips (GRB in my case)? 

[image]

sjude68, 2020-03-20:
> Hi @Aircoookie, will your multi strip approach also support different types of led strips controlled by the same esp? E.g. Led Pixel Lights (RGB in my case) and LED strips (GRB in my case)?
> 
> [image]
> 
> Would be great if they can be connected in row using a single data wire.

Yes. I have used it my led strip and connected the one you showed

PaulNBerlin, 2020-03-20:
Hi @sjude68, I can connect but the collors are wrong
[image]

sjude68, 2020-03-20:
Yes @PaulNBerlin the color is different because that strip uses Rgb color coding. I faced the same problem and posted the query. Something to do with segments. There should be way to in next segment.

Xeyk, 2020-04-03:
Can't wait for this feature too! I am controlling three different data lines with Esphome right now and want to switch to wled. This is the only thing stopping me.

Xeyk, 2020-04-14:
Hi Mr. Aircoookie. Did you need any help testing this or coding help on this? Please let me know.

Aircoookie, 2020-04-14:
@Xeyk thank you for offering your help! I will let you know once I have something to work with :)
Here is what I'm working on, if you are interested: https://gist.github.com/Aircoookie/974ef271fa9b73db45aebc5199302ee5
But it's not yet in a state where it would be useful, there is still quite a bit to do

grainz, 2020-04-19:
I just started with WLED yesterday to experiment with RGB in my PC with fans and strips. They use WS2812B. Having the ability to interact with each device on their own would be a great feature. I've been reading up on shift registers for more GPIO but this doesn't seem to apply to SPI type devices. I look forward to see how this progresses! Thank you for the great software!

jordanadania, 2020-05-16:
Anyone know why Ninja's solution won't work for me? I get the error:
`CRGB does not define a type`

Aircoookie, 2020-05-17:
@jordanadania have you compiled WLED without modifications successfully? You might not have the FastLED library installed yet.

jordanadania, 2020-05-17:
i definitely have fastled installed and have compiled wled without mod and with mod. this is my only issue so far.

Aircoookie, 2020-05-17:
@jordanadania looked at the above modification again, probably in `FX_fcn.cpp` the `#include "FX.h"` is below the pasted-in code, it needs to remain at the very top. That might be the issue.

Configurable multi-pin output will be the main feature of v0.12.0, the next major version after 0.11.0 where filesystem support will be added.

jordanadania, 2020-05-17:
now its 
`'_rgbwMode' was not declared in this scope`

Aircoookie, 2020-05-17:
Ahh, great, above code is of course outdated 😅 
Try this init (no guarantee from me that it will work!)
```cpp
void WS2812FX::init(bool supportWhite, uint16_t countPixels, bool skipFirst)
{
  if (supportWhite == _useRgbw && countPixels == _length && _skipFirstMode == skipFirst) return;
  RESET_RUNTIME;
  _useRgbw = supportWhite;
  _length = countPixels;
  _skipFirstMode = skipFirst;

  uint8_t ty = 1;
  if (supportWhite) ty = 2;
  _lengthRaw = _length;
  if (_skipFirstMode) {
    _lengthRaw += LED_SKIP_AMOUNT;
  }

  bus->Begin((NeoPixelType)ty, _lengthRaw);
  
  _segments[0].start = 0;
  _segments[0].stop = _length;

  setBrightness(_brightness);

#ifdef Use_FastLED

	//Pointer to NeoPixelBus led array	
	leds = (CRGB*)bus->GetPixels(); //GetPixels function added to "NpbWrapper.h"
	
	// tell FastLED there's 284 NEOPIXEL leds on pin 12, starting at index 0 in the led array
	FastLED.addLeds<LED_TYPE, 12, COLOR_ORDER>(leds, 0, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
	
	// tell FastLED there's 284 NEOPIXEL leds on pin 14, starting at index 284 in the led array
	FastLED.addLeds<LED_TYPE, 14, COLOR_ORDER>(leds, 1 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);

	// tell FastLED there's 284 NEOPIXEL leds on pin 27, starting at index 568 in the led array
	FastLED.addLeds<LED_TYPE, 27, COLOR_ORDER>(leds, 2 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);

	// tell FastLED there's 284 NEOPIXEL leds on pin 26, starting at index 852 in the led array
	FastLED.addLeds<LED_TYPE, 26, COLOR_ORDER>(leds, 3 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
#endif // Use_FastLED
}
```

jordanadania, 2020-05-17:
> This function is added into "NpbWrapper.h"
> 
> uint8_t* GetPixels() 
>   {
> 	  return _pGrb->Pixels();
>   }

Says I can't overload that function. You must have added it since?

jordanadania, 2020-05-17:
Yeah, I am getting crazy random colors after updating LED count. On startup, the first 30 were correct.

Aircoookie, 2020-05-17:
> You must have added it since?
Yes, albeit unused.

Try rebooting, it will re-initialize everything, then it might work (weird, the pointer should get updated every time the LED count is updated)

jordanadania, 2020-05-17:
when i swap the GetPixel functions, the AP doesnt show up

jordanadania, 2020-05-17:
resetting changes the random colors

jordanadania, 2020-05-17:
nope. thats correct answer. Rebooting. THANK YOU!!

jordanadania, 2020-05-17:
After upload, I had to change led count in LED preferences again. then save. then reboot. works perfect. THANK YOU!

robinzelders, 2020-05-17:
I tried to use multiple strips using @N1nja98 's code. On boot the two strips do indeed light up but checking the serial monitor it shows the following. Does anyone know what happens or even better how to fix it?
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Ada
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008570e  PS      : 0x00050031  A0      : 0x40084d40  A1      : 0x3ffbe7a0  
A2      : 0x00000019  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x02000000  
A6      : 0x02000000  A7      : 0x00060623  A8      : 0x3ff56000  A9      : 0x3ffba264  
A10     : 0x00000015  A11     : 0x3ffbd6b0  A12     : 0x00060023  A13     : 0x3ffbc2b0  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000007  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008570e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400873c1

Backtrace: 0x4008570e:0x3ffbe7a0 0x40084d3d:0x3ffbe7d0 0x4017aa97:0x3ffbc740 0x4010497e:0x3ffbc760 0x4008b875:0x3ffbc780 0x4008a091:0x3ffbc7a0

Rebooting...
ets Jun  8 2016 00:22:57

```

robinzelders, 2020-05-18:
> I tried to use multiple strips using @N1nja98 's code. On boot the two strips do indeed light up but checking the serial monitor it shows the following. Does anyone know what happens or even better [...]

That is on my ESP32
on my ESP8266 I get:
```
Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 01a0
3ffffec0:  00000000 3fffdad0 3fff0fac 3fff0db4  
3ffffed0:  3fffff30 3fffff3c 00000000 40220e1f  
3ffffee0:  00000050 3fffc6fc 00000000 3fff0bab  
3ffffef0:  00000003 3fff0d88 3fff041c 3fffff3c  
3fffff00:  3fffff30 00000050 3fff0db4 40220e8a  
3fffff10:  3ffe9924 3fffff30 3ffe9920 3fff0bab  
3fffff20:  3ffe9924 3ffe9920 3fff0db4 40214004  
3fffff30:  00706374 df01a8c0 83c6a7f0 64656c77  
3fffff40:  3fff4000 8421ad40 40100af4 3fff0fac  
3fffff50:  3fff0bb1 00000000 3fff0188 40214286  
3fffff60:  00000000 a0aa6235 3fff0100 3fff0fac  
3fffff70:  00000000 00000000 3fff0188 3fff0fac  
3fffff80:  3fffdad0 00000000 3fff0188 402142f9  
3fffff90:  3fffdad0 00000000 3fff0f6c 40214bc0  
3fffffa0:  feefeffe feefeffe feefeffe 40227f54  
3fffffb0:  feefeffe feefeffe 3ffe87f0 4010088d  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
```

Which decodes to: 
`Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4000bdc8
EXCVADDR: 0x00000000`

But still, no idea what that means exactly

N1nja98, 2020-05-18:
Not sure if this mod works on ESP8266. I've only used it on ESP32. I've since had to modify some code after some updates. I've had to change 
`#define PIXELMETHOD NeoEsp32Rmt0Ws2812xMethod`
to
` //#define PIXELMETHOD NeoEsp32Rmt0Ws2812xMethod`
 `#define PIXELMETHOD NeoWs2813Method`

PaulNBerlin, 2020-05-18:
Hello @Aircoookie,

I would like to know in which direction the development regarding this issue will go.
In my opinion there are 2 options:

1. Controlling multiple types of strips using multiple output pins of the esp 
2. Controlling multiple types of stript using 1 output pins + segments (e.g. first segment uses GRB color and second RGB color) --> This option would result in only 1 data wire.

Thanks 
Paul

robinzelders, 2020-05-18:
> Not sure if this mod works on ESP8266. I've only used it on ESP32. I've since had to modify some code after some updates. I've had to change
> `#define PIXELMETHOD NeoEsp32Rmt0Ws2812xMethod`
> to
> ` //#define PIXELMETHOD NeoEsp32Rmt0Ws2812xMethod`
> `#define PIXELMETHOD NeoWs2813Method`
> in NpbWrapper.h

Still get the same error unfortunately

jordanadania, 2020-05-18:
It works perfectly on the 8266. Robinzelders error sounds like out of range. Either and array went too far or a variable like uint8_t went beyond 255. Check for something like that.

robinzelders, 2020-05-18:
I only get the error if I use FastLED.show() instead of bus->Show(); I assume the problem might be there

robinzelders, 2020-05-18:
Is it because something has changed in V0.10.0 or does it work on that version for you guys as well?

jordanadania, 2020-05-18:
Can I see your init?

robinzelders, 2020-05-18:
Switched to the Arduino IDE and now it works on my ESP2866 but still not on my ESP32

robinzelders, 2020-05-18:
> Can I see your init?

`void WS2812FX::init(bool supportWhite, uint16_t countPixels, bool skipFirst)
{
  if (supportWhite == _useRgbw && countPixels == _length && _skipFirstMode == skipFirst) return;
  RESET_RUNTIME;
  _useRgbw = supportWhite;
  _length = countPixels;
  _skipFirstMode = skipFirst;

  uint8_t ty = 1;
  if (supportWhite) ty = 2;
  _lengthRaw = _length;
  if (_skipFirstMode) {
    _lengthRaw += LED_SKIP_AMOUNT;
  }

  bus->Begin((NeoPixelType)ty, _lengthRaw);

  _segments[0].start = 0;
  _segments[0].stop = _length;

  setBrightness(_brightness);
#ifdef Use_FastLED

  //Pointer to NeoPixelBus led array
  leds = (CRGB*)bus->GetPixels(); //GetPixels function added to "NpbWrapper.h"

  // tell FastLED there's 284 NEOPIXEL leds on pin 12, starting at index 0 in the led array
  FastLED.addLeds<LED_TYPE, 2, COLOR_ORDER>(leds, 0, 165).setCorrection(TypicalLEDStrip);

  // tell FastLED there's 284 NEOPIXEL leds on pin 14, starting at index 284 in the led array
  FastLED.addLeds<LED_TYPE, 3, COLOR_ORDER>(leds, 165, 42).setCorrection(TypicalLEDStrip);

#endif // Use_FastLED
}`

robinzelders, 2020-05-20:
Also I get the following error in my compiler when trying to upload to the ESP32 in Arduino IDE. Maybe this is the problem, any idea how to fix this?

```
Sketch uses 1375524 bytes (104%) of program storage space. Maximum is 1310720 bytes.
text section exceeds available space in board
Global variables use 56996 bytes (17%) of dynamic memory, leaving 270684 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board DOIT ESP32 DEVKIT V1.
```

N1nja98, 2020-05-25:
@robinzelders I just tried updating mod with latest WLED code and received the same LoadProhibited exception. The latest working version I have is v0.9.0-b1. I’ll have to figure out what changed in newer code. If you still want it I can post the full list of mods   necessary to get it working on ESP32.

robinzelders, 2020-05-25:
@N1nja98 That would be great actually, I was rewriting the code myself but to no avail.

bjornbergenheim, 2020-05-25:
I'm trying this as well, and I'm also seeing the same problems. So go N1nja98!

N1nja98, 2020-05-25:
Ok the latest working version I have is this commit: https://github.com/Aircoookie/WLED/tree/558811e4f9fd0a546e28275a6982d6768ae50d74

Here are the mods I had to make:

This is placed at the top of "FX_fcn.cpp" under the include statements:
```
//toggle for multi-strip
#define Use_FastLED //FastLED already included in FX.h

#ifdef Use_FastLED

#define LED_TYPE    WS2812B
#define COLOR_ORDER RGB //Actually GRB. Explained later.
#define NUM_STRIPS 4
#define NUM_LEDS_PER_STRIP 284
//#define NUM_LEDS NUM_LEDS_PER_STRIP * NUM_STRIPS

CRGB* leds;
#endif // Use_FastLED
```

This is placed in same file in at the bottom of  "WS2812FX::init" right under ```setBrightness(_brightness);```:

```
//An example for driving 1036 leds in four different strips with 284 leds in each.
#ifdef Use_FastLED

        //Pointer to NeoPixelBus led array	
        leds = (CRGB*)bus->GetPixels(); //GetPixels function added to "NpbWrapper.h"
	
	// tell FastLED there's 284 NEOPIXEL leds on pin 12, starting at index 0 in the led array
	FastLED.addLeds<LED_TYPE, 12, COLOR_ORDER>(leds, 0, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
	
	// tell FastLED there's 284 NEOPIXEL leds on pin 14, starting at index 284 in the led array
	FastLED.addLeds<LED_TYPE, 14, COLOR_ORDER>(leds, 1 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);

	// tell FastLED there's 284 NEOPIXEL leds on pin 27, starting at index 568 in the led array
	FastLED.addLeds<LED_TYPE, 27, COLOR_ORDER>(leds, 2 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);

	// tell FastLED there's 284 NEOPIXEL leds on pin 26, starting at index 852 in the led array
	FastLED.addLeds<LED_TYPE, 26, COLOR_ORDER>(leds, 3 * NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
#endif // Use_FastLED
```

 ```bus->Show()``` in same file at bottom of "WS2812FX::show" is replaced with: 
```
#ifndef Use_FastLED
	bus->Show();
#else
	FastLED.show();
#endif // Use_FastLED
```

This function is added into "NpbWrapper.h"
```
uint8_t* GetPixels() 
  {
	  return _pGrb->Pixels();
  }
 ``` 

```#define PIXELMETHOD NeoEsp32Rmt0Ws2812xMethod``` on line 56 in "NpbWrapper.h" is replaced with:
```
#define PIXELMETHOD NeoWs2813Method
//#define PIXELMETHOD NeoEsp32Rmt0Ws2812xMethod
```

Version number ```Esp Async WebServer@1.2.0``` on line 40 in "platformio.ini" is changed to ```Esp Async WebServer@1.2.2```

 Version number ```platform = espressif32@1.11.1``` on line 104 in "platformio.ini" is changed to ```platform = espressif32@1.7.0```

I use VS Code with PlatformIO for compiling and uploading. Here is the full version of the modified code. Let me know if it works for you.
[WLED Multi-Strip Mod.zip](https://github.com/Aircoookie/WLED/files/4678724/WLED.Multi-Strip.Mod.zip)

I'll try to get it updated to latest version of WLED.

bjornbergenheim, 2020-05-25:
Cool, before I go and change git around into your version, have you tried head and not get it to work? Also, the pin definition in NpbWrapper_h
`#define LEDPIN 2`
How does that pin play into the pins defined in the WS2812FX::init method?

N1nja98, 2020-05-25:
@bjornbergenheim I’ve tried both releases 0.10.0 and 0.9.1. Both didn’t work. So something between 0.9.1 and the commit I mentioned previously. 

> Also, the pin definition in NpbWrapper_h
#define LEDPIN 2
How does that pin play into the pins defined in the WS2812FX::init method?

I’m not sure actually. No data is sent to the pin because you are never calling ``bus->Show()`` but I just use different pins with the FastLed initialization.

bjornbergenheim, 2020-05-25:
I'm using Exception Decoder, and though I'm out on a very thin limb here (not really knowing what I'm doing) if the /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c  file is actually esp32-hal-rmt.c then line 621 is close to pinMode(pin, INPUT);
I have nothing connected to this ESP32 right now. Could that causes the problem? It runs OK without crashes (also withouth LEDs connected to any pins) with the stock WLED.

The crash is decoded to this
```
PC: 0x4008570e: rmt_driver_isr_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 621
EXCVADDR: 0x00000014

Decoding stack results
0x4008570e: rmt_driver_isr_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 621
0x40181ec7: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x4010b502: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x4008b875: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x4008a091: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

bjornbergenheim, 2020-05-25:
There is this in NbpWrapper.h that makes the error of overloading if you try to add your own code 
```
  uint8_t* GetPixels(void)
  {
    switch (_type) {
      case NeoPixelType_Grb:  return _pGrb->Pixels();  break;
      case NeoPixelType_Grbw: return _pGrbw->Pixels(); break;
    }
    return 0;
  }
```
It was added  with the heartbeat feature
https://github.com/Aircoookie/WLED/commit/5befcd24b526cb5822f0dec3de071ba44af6fe2e

bjornbergenheim, 2020-05-25:
I tried your zipfile, and my esp32 still crashes.

```
entry 0x400806b8
Ada
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008559e  PS      : 0x00050031  A0      : 0x40084bd0  A1      : 0x3ffbe7a0  
A2      : 0x00000018  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x01000000  
A6      : 0x01000000  A7      : 0x00000000  A8      : 0x3ff56000  A9      : 0x3ffb1ce0  
A10     : 0x00000000  A11     : 0x3ffbc4c0  A12     : 0x00000020  A13     : 0x80000020  
A14     : 0x00000000  A15     : 0x3ffb8058  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008559e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40087251

Backtrace: 0x4008559e:0x3ffbe7a0 0x40084bcd:0x3ffbe7d0 0x40176a67:0x3ffbc7d0 0x401008aa:0x3ffbc7f0 0x4008b705:0x3ffbc810 0x40089f21:0x3ffbc830
```
```
PC: 0x4008559e: rmt_driver_isr_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 580
EXCVADDR: 0x00000014

Decoding stack results
0x4008559e: rmt_driver_isr_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 580
0x401008aa: IRrecv::decodeCarrierAC(decode_results*, unsigned short, unsigned short, bool) at E:\Mina dokument\Arduino\libraries\IRremoteESP8266\src\ir_Carrier.cpp line 101
0x4008b705: vTaskPrioritySet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 1677
0x40089f21: xQueueReceiveFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1616
```

[182 more replies]

closed 2021-04-06 by Aircoookie