wled/WLED #107: Various Errors in UI

raymiec, 2019-02-12:
Initial error in devtools 

(index):1 Uncaught ReferenceError: feedback is not defined
    at HTMLIFrameElement.onload ((index):1)

Under fx mode it just says loading and the slider is missing.

Fastled options not loading,  under fastled palette is just says error loading list.
devtools error (index):29 GET http://10.10.1.52/json/effects net::ERR_CONNECTION_TIMED_OUT

Aircoookie, 2019-02-12:
I'm sorry, I totally messed up and left the IP of my testing device in there instead of a relative URL. Will need to re-release. Expect the fix in about an hour.

raymiec, 2019-02-12:
no worries man, just giving you the heads up.  

If you want to fire your android app source on github i can add some stuff to it also just finished a Kiosk mode to lock the app at boot. works pretty sweet.

Aircoookie, 2019-02-12:
It is a Xamarin.forms app, not sure you can work with that, but go ahead! http://app.wled.me

Thanks again for the heads-up, the fix is released now :)

raymiec, 2019-02-12:
oh no i'll keep away from that thank you very much, will stick with android studio :).  Was xamarin just so you could do cross platform?  

Think it's only android and IOS that really matter?  IOS should be easy enough, i already have an app that has enough native functionality to pass apple app review, network discovery should be easy enough i think as bonjour service is already there. I could submit one but it would be under my business name?  (saves £99 a year).

Mariu86, 2019-02-12:
The Android app will be a better choice 😀please work ok that . Thanks

raymiec, 2019-02-12:
Still issues with the ui after that update errors are gone but ui a bit screwed.
[image]

Aircoookie, 2019-02-12:
Yes, I chose Xamarin because it's so easy for cross platform! (also because I like C#)

Sure, it'd be amazing if iOS users had a native app! If you are already paying the store fee, absolulely, submit one :)

The UI screenshot looks fine to me, except for being off-center. Is it just cropped that way or is it actually not in the center? Also, what browser do you use?

raymiec, 2019-02-12:
ah wait thats been intentionally moved, scroll wasnt activating on my mac

raymiec, 2019-02-12:
Right i'll do IOS, i've actually got a couple developers accounts and theres one with only one app so would be good to get more use of that anyway.

I've not really any experience with Xamarin except a few weeks at college so would rather stay away, i'm guessing less compatibility with existing libraries /packages though. I'll continue with my android app as its a better option for me, i can't imagine kiosk mode etc working with Xamarin,

Mariu86, 2019-02-12:
@raymiec when you finish the Android app can you share with us ? Thanks

raymiec, 2019-02-12:
@Mariu86 Im going to work on the IOS app so not planning on doing much with android for a couple of weeks. My own build is targeted mainly towards tablets in kiosk mode and runs alongside a laravel app theres a lot of stuff that nobody will want and custom to me so needs a proper cleanup.  

That being said, i have some issues on my pixel 3 so i plan on knocking up another quick app thats suits my own needs more.

What sort of features are you after? if theres something missing from Aircoookies i can maybe add it.

something along these lines with more info on the cards and more controls beside the colour chooser.
[image]

closed 2019-02-13 by raymiec