wled/WLED #109: Compilation of DMA mode (pin3) problems

demlak, 2019-02-14:
Hey.. i got those errors.
Any hint on this?

> In file included from /home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBrightnessBus.h:29:0,
                 from sketch/NpbWrapper.h:42,
                 from sketch/WS2812FX.h:30,
                 from sketch/WS2812FX_fcn.cpp:27:
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h: In instantiation of 'bool NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::CanShow() const [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>]':
sketch/NpbWrapper.h:113:46:   required from here
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:161:40: error: passing 'const NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>' as 'this' argument of 'bool NeoEsp8266DmaMethodBase<T_SPEED>::IsReadyToUpdate() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]' discards qualifiers [-fpermissive]
         return _method.IsReadyToUpdate();
                                        ^
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h: In instantiation of 'bool NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::CanShow() const [with T_COLOR_FEATURE = NeoGrbwFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>]':
sketch/NpbWrapper.h:114:47:   required from here
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:161:40: error: passing 'const NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>' as 'this' argument of 'bool NeoEsp8266DmaMethodBase<T_SPEED>::IsReadyToUpdate() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]' discards qualifiers [-fpermissive]
In file included from /home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBrightnessBus.h:29:0,
                 from sketch/NpbWrapper.h:42,
                 from sketch/WS2812FX.h:30,
                 from sketch/WS2812FX.cpp:27:
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h: In instantiation of 'bool NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::CanShow() const [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>]':
sketch/NpbWrapper.h:113:46:   required from here
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:161:40: error: passing 'const NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>' as 'this' argument of 'bool NeoEsp8266DmaMethodBase<T_SPEED>::IsReadyToUpdate() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]' discards qualifiers [-fpermissive]
         return _method.IsReadyToUpdate();
                                        ^
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h: In instantiation of 'bool NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::CanShow() const [with T_COLOR_FEATURE = NeoGrbwFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>]':
sketch/NpbWrapper.h:114:47:   required from here
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:161:40: error: passing 'const NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>' as 'this' argument of 'bool NeoEsp8266DmaMethodBase<T_SPEED>::IsReadyToUpdate() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]' discards qualifiers [-fpermissive]
In file included from /home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBrightnessBus.h:29:0,
                 from sketch/NpbWrapper.h:42,
                 from sketch/WS2812FX.h:30,
                 from /home/demlak/tmp/WLED-master/wled00/wled00.ino:76:
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h: In instantiation of 'bool NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::CanShow() const [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>]':
sketch/NpbWrapper.h:113:46:   required from here
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:161:40: error: passing 'const NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>' as 'this' argument of 'bool NeoEsp8266DmaMethodBase<T_SPEED>::IsReadyToUpdate() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]' discards qualifiers [-fpermissive]
         return _method.IsReadyToUpdate();
                                        ^
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h: In instantiation of 'bool NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::CanShow() const [with T_COLOR_FEATURE = NeoGrbwFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>]':
sketch/NpbWrapper.h:114:47:   required from here
/home/demlak/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:161:40: error: passing 'const NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>' as 'this' argument of 'bool NeoEsp8266DmaMethodBase<T_SPEED>::IsReadyToUpdate() [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]' discards qualifiers [-fpermissive]
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

Aircoookie, 2019-02-15:
Hi!
Normally all you'd need to do to use DMA is to set `#define LEDPIN 3` in NpbWrapper.h.
If you did that and still got this compilation error, please try updating the NeoPixelBus library (I use version 2.3.5).
I've got two lights running in DMA mode really well with that!
Out of interest, how would you describe the flicker you have with the UART method? I use it on most of my lights and it was really reliable so far.

demlak, 2019-02-16:
Hi,
i was using neopixelbus-lib by makuna in the newest version 2.4.2. UART mode compiled without problems with this newest version.

Downgrading to 2.3.5 results in a working compilation for DMA mode, too.

Using such an old version is, maybe, not a good choice at all?

The flickering maybe depends also on my wiring testing-solution right now.. much LEDs and just a breadboard with jumperwires.. maybe not good for high current =)

richyxi, 2019-05-30:
@demlak I'm facing a little flicker and I think that is also my wiring set up, do you solve this problem? Thanks in advance!

demlak, 2019-05-30:
i'm sorry.. i wasn't into this since my last post.

closed 2019-10-25 by Aircoookie