wled/WLED #110: Blue = Blue + White

demlak, 2019-02-14:
Hi,
i tried to get a clean blue color (for example in the webinterface in RGBWmode) with my sk6812 RGBW Stripe. 
But always the white LED is also used, when changing value for Blue.
Is it possible to clearly seperate the blue and the white LED?

thx in advance
demlak

Mariu86, 2019-02-15:
You have set something not right. I have the sk6812 and it's working ok !

Aircoookie, 2019-02-15:
Do you have a separate slider for white value in the UI? If yes, adjust it to all the way left to turn off the white LEDs. Otherwise they are mixed with the RGB color. You can also enable the "Auto-calculate white from RGB instead" option in LED settings!

demlak, 2019-02-16:
hey.. thx for your reply.

my settings alow me to adjust the white LED seperated. But it is still used on the blue LED, too.

if i set blue to 100% it is also not as bright as when i set red or green on 100%. maybe this is because blue and white LED share the current all the time. 

Setting blue is always setting blue + white at the same time. 

i tried to make a photo.. but impossible with my cam. =)

i tried to make a video of it.. difficult task =)
link to video (132mb big file!!): https://0ma.de/TRIM_20190216_104950.mp4

timeline of video:
00s-23s - slow dimming from black to 100% Blue
23s-30s - fast dimming from black to 100% green
30s-36s - fast dimming from black to 100% red
36s-41s - fast dimming from black to 100% white

Aircoookie, 2019-02-16:
How weird! Do you use the mobile or classic user interface? Please try selecting an effect like rainbow or colorloop, and see if that achieves a pure blue color or also mixes blue and white. If it mixes, it is an issue with the LEDs or their protocol. If it doesn't, please try opening the URL `http://[esp-ip]/win&FX=0&B=255&W=0` in your browser and see if that achieves a pure blue.

demlak, 2019-02-17:
i used the classic interface. on effects like rainbow or colorloop it is still mixing.. 
nevertheless, result of `/win&FX=0&B=255&W=0` is also mixing.. 

```
<vs>
<ac>255</ac>
<cl>0</cl>
<cl>0</cl>
<cl>255</cl>
<cs>0</cs>
<cs>0</cs>
<cs>0</cs>
<ns>1</ns>
<nr>1</nr>
<nl>0</nl>
<nf>1</nf>
<nd>60</nd>
<nt>0</nt>
<fx>0</fx>
<sx>75</sx>
<ix>128</ix>
<fp>0</fp>
<wv>0</wv>
<ws>0</ws>
<md>1</md>
<cy>0</cy>
<ds>WLED Light</ds>
</vs>
```

Aircoookie, 2019-02-17:
Ok, if the effects produce unwanted white as well, it's definitely the white LEDs setting themselves to the same brightness as the blue ones for some reason. Do you know the exact type name of the LEDs or have a purchase link. I'd like to take a closer look. Something must be wrong as you are still able to control the white LEDs while leaving the blue ones off.

demlak, 2019-02-17:
I got them from a friend who says, they are `sk6812`

Aircoookie, 2019-02-17:
Thought so, I'm not aware of any type which supports RGBW but `SK6812`. Still, I have them as well and they work as expected for me (and @Mariu86 above as well). If they are old, maybe the first generation behaved differently, although I haven't heard of protocol changes.

Can you flash https://codebender.cc/sketch:318150#Neopixel%20RGBW%20Test.ino or a similar test sketch and see if it achieves blues without mixing?

demlak, 2019-02-17:
i got a compilation error on your sketch.. so i used this one: https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/RGBWstrandtest/RGBWstrandtest.ino

i took a look more in detail.. and maybe i/we got fooled..
when only the White LED is brighten up, there is a smal spot inside the area which is brighten up.. 
when the blue LED is brighten up, the whole surface and not just a little spot of the white LED is brighten up.
i think the light of the blue LED is lighten-up the surface of the white LED. so, it seems, that not the white LED itself is shining, but just the surface is enlighted by the blue LED. crazy =)

so... it seems everything is ok with this stripe..

nevertheless.. blue is not as bright as the other colors.. maybe this is just physics?

sansillusion, 2019-02-17:
I would guess the blue phosphorus material is week or damaged and the uv leek from the blue led and react with the phosphorus material on the adjacent white led.

demlak, 2019-02-17:
over all, the blue is a very light blue.. this sucks.. i'll check if i can change the stripe at my friend.

thx a lot guys!

Mariu86, 2019-02-17:
You have a diode or a capacitor between the esp and LEDs ?

demlak, 2019-02-17:
i'm using a resistor between data-in of the LED-stripe and pin of ESP.
Also i use a capacitor parallel to the power of the LED-stripe. 
The LED-stripe power does not come from the ESP. both, LED-stripe and ESP are driven from the same source.. parallel.

Mariu86, 2019-02-17:
Try to connect LEDs direct from esp,if I remember correctly I think I have some strange color effects when I used a capacitor and a diode or resistor I don't remember exactly. Now I'm using a level shifter from 3,3 to 5v and I connect the esp - level shifter- 300 LEDs with no problem. Try to disconnect everything and connect directly from esp and set the nr of LEDs to 1-2 and see if it's any change. Sorry for my English .

demlak, 2019-02-18:
No difference

Mariu86, 2019-02-18:
Sorry for you. You have some rgbw stress test in the Arduino try them and see if it's any difference. I have 2 different  sk6812 strips  whit 30/60 LEDs per meter from different suppliers and work very well, the white color it's a big difference when you set only white.

demlak, 2019-02-18:
yes.. did you read this? https://github.com/Aircoookie/WLED/issues/110#issuecomment-464500625

Mariu86, 2019-02-18:
:( sorry for you then.

closed 2019-03-27 by Aircoookie