wled/WLED #112: Led Count?

raymiec, 2019-02-16:
Any way to pull to led count?

Aircoookie, 2019-02-16:
Not at the moment, but next build will feature an `[ip]/json/info` page which will include `{"ledcount":150}` among other parameters.

raymiec, 2019-02-16:
now worries, can inject a js function for now.  Do you plan on exposing all settings? i plan on doing both android and ios as full native so would be handy. I have a first draft of android if you want me to send the apk over for you to have a look?

Aircoookie, 2019-02-16:
Maybe at some point all settings will be exposed if I switch from JS injection to JSON for all settings pages.
For now, only the settings that I think might be relevant for a 3rd party application using the API will be included. Prime examples would be ledcount, rgbw support and the likes. Boot defaults also seem like a good idea.

Awesome, great work! I would be super keen on taking a look at your draft apk :)

raymiec, 2019-02-16:
Nice one! i'll let you know if anything else could be useful.v
I sent you an email with link to apk.

Aircoookie, 2019-02-17:
Took a quick look at the apk. Awesome work, the app has the potential to become really useful!(especially considering native performance, sound reactiveness... )! Discovering and controlling lights worked as expected out of the box!

For my HTC U11 (1440p 16:9 screen) there is a slight layout issue:
(Nothing major, just so you are aware)

[image]

raymiec, 2019-02-17:
Nice one at least it ran! Lol.

The audio reactive stuff if going to be difficult for me, but the plan is basically just to try port ledfx over. Same library so shouldn't be too hard.  Added more notes and colours today and made it fill the whole strip, seemed to work pretty well.

I've added some controls for ledfx as well, should make both Wled and Ledfx to work together.  

But yeah loads of ideas for it. The layouts I just did basic constraints so it looked ok on my pixel 3 but will get that sorted shortly.

Also if any one reading this wants to help, get in touch :)

Mariu86, 2019-02-17:
Very good guys ! Hope to see a final version! Thanks

raymiec, 2019-02-18:
@Aircoookie I updated the apk, if you fancy downloading again.  Used a new library to manage network requests, makes it a lot faster and updates to the ui are now almost instant. Seems pretty solid,

I added some more constraints, checked in emulator at your screen res, looks ok.

The audio reactive page you can set the ip to test, seems pretty responsive as well.

and it should look like this. Can focus on the design later, functionality for now.

[image]

Aircoookie, 2019-02-21:
@raymiec looks great! I'm really excited how this app will evolve!
I just updated the master, you can now get the led count in the `/json/info` page (`jsonobj.leds.count`)!

raymiec, 2019-02-21:
Excellent, that will save a call to the settings page and make it easier for me to check if the led count has been changed.

I'm halfway through the settings pages, you inject all the settings in one string so was really easy to parse. I just need to make them submit now, injecting the new values and hitting submit should do it.

Removed that scrolling cardview with the hosts from the main page to give room for more features, Added SQLite and moved away from shared preferences for storing data(Much better).  

But yeah it's coming together and easier to implement things as theres a bit of a framework now. Another week or so on the code and same for the design and we should have something good to go.

MatFl, 2019-02-21:
Hi @raymiec, are you making a native android app? If yes I would be interested in helping you out a bit. Do you mind sharing your progress?

raymiec, 2019-02-21:
@MatFl Yeah, give me a week or two to finish off this first version and i'll get the source on github.

Really there's not that much more that can be done with it to control WLED, so it would be more implementing features to take advantage of WLED.  I.e the audio reactive side of things. Currently can get beat, pitch etc but we need effects (Which can probably be ported over from LedFx).

Today i tested ambilight, worked ok on videos and images so this is possibly another feature to implement.

If you do have any ideas of your own that you would like to add let me know and i can make sure to keep it in mind so it will be easier for you to add.

Here's some more screenshots of where it's at.

[image]

[image]

[image]

closed 2019-03-01 by raymiec

raymiec, 2019-03-02:
Some progress with this, should be close to a beta i would think. Still to finish a couple of the settings pages but other than that everything works.

Added the LedFx stuff, you can scan for servers, it will automatically add the WLED device to LedFx if it doesn't exist and you can start and stop effects.  Effects options will come soon.

IOS app will be coming shortly, a good bit of work has been done on that too.

if anyone wants to try it, https://led24x7.com/wled01.apk

Should look something like this, will do more on the design soon.
[image]
[image]
[image]
[image]
[image]

sansillusion, 2019-03-02:
I had the first version installed as a reminder to try it when I can. I just tried to install that new version but it my nexus 6 running latest Nitrogen os but it says cannot install. BTW you and aircookie are making a great app and code for our esp32 ! Thanks a lot !

raymiec, 2019-03-02:
I switched the apk. try downloading again, just tried on a couple of devices, seems ok. It should be on the app store during the week once i get some more work done on the layouts.

sansillusion, 2019-03-02:
This one works, thanks a lot !

Aircoookie, 2019-03-02:
Awesome, great work on this! That's some serious progress! Everything works for me as expected as well!
One tip, you can get JSON lists of the effects and palette names with `/json/effects` and `/json/palettes`. Then, you won't have to update the app if I add new FX!

Keep up the good work!

raymiec, 2019-03-02:
Nice one, i'll get it tidied up and on the play store and we can get a beta out.

Ah yeah i just forgot to uncomment the function for that, i literally copied the json files from there to the assets folder just to stop the calls during dev.

Actually been enjoying using it, makes the lights a lot more usable.

raymiec, 2019-03-03:
I updated the apk again, fixed some stuff on the ui and tidied up some loose ends, also effects & palettes are loaded dynamically.

I'm in the middle of writing a weather service, seems to be working well, lights / effects can be set depending on the weather, this is fine for me as i have dedicated android device on the network, but not much use if just using on your phone as it won't work when your mobile isn't on the same network, saying that if it is of interest let me know and i'll put it in next release.

Plan is just to finish of the setting pages and put it on the app store during the week, if yous spot any issues let me know, also if there's anything you want added.

Cheers

Mariu86, 2019-03-03:
@raymiec the app need to connect to wled with 0.8.3 version ? I have the 0.8.2 version and I have some problem with the wled effects

raymiec, 2019-03-03:
it should work with 0.8.2 i'll try it out myself,  what issue are you having ?

Aircoookie, 2019-03-03:
Ah yes, sorry for the confusion! The JSON lists were only introduced in v0.8.3.

Mariu86, 2019-03-03:
The search it's not working but I was connected with the Ip address , the problem it's with the effects and with the side bars that In my case it's showing only 2 blue dots  , and in settings the wifi setting are a little bit to the Left

raymiec, 2019-03-03:
@aircookie, devices can still be added manually to the app so still need to support. 

i've sorted it anyway, just set to load from assets folder if older than 0.8.3.

I'll upload an apk soon, just finishing off some stuff for one click OTA updates. Also working on a function to flash the sketch from your phone over usb with otg adapter.

In the case of the two blue dots it sounds like a constraint issue, What model is your phone and android version? i'll load it up in the emulator and have a look.   Wifi /LED Settings etc ain't finished yet.

You can drag and drop screenshots here.

Cheers

Mariu86, 2019-03-03:
I don't know how to send the pictures from the phone, it's saying to drag and drop but I can do that , I try to copy to the clipboard but I can't .

raymiec, 2019-03-03:
@Mariu86 i've updated the apk if you want to give it a go, effects & palettes should load on older versions now.   What phone do u have and i can check it in the emulator ?

Mariu86, 2019-03-04:
I have a Huawei p9 with Android 7.0, I will try the app later, I must download the app from the same link from above?

raymiec, 2019-03-04:
yeah same link

Mariu86, 2019-03-04:
Thanks for the good job ! Hope the app will evolve in something very good !

raymiec, 2019-03-04:
I should add, when installing over an older version you will probably need to clear app data or uninstall app first!.

raymiec, 2019-03-05:
Little update tonight if anyone has an OTG cable and fancies trying it out. You can plug your esp in to your phone and monitor the serial output.

[image]
[image]
[image]

Mariu86, 2019-03-05:
Very nice ! :D can you tell me how to do a update from the phone ?

raymiec, 2019-03-05:
Updating the device from the app will hopefully be in the next build, i'm still having issues with that at the moment.

Aircoookie, 2019-03-05:
Great job on the terminal! Works flawlessly for me, it even did a little pop-up asking me if i wanted to open it with Device Control when I plugged in the ESP. This is a very helpful feature for debugging if a PC is not in reach!

Something else: In the next few hours I'll push the new JSON API! Unfortunately, a small change was required to make it work nicely and efficiently. This change is that `/json/effects` and `/json/palettes` now just return the array `["Solid","Blink","..."]` instead of the respective object `{"effect":["Solid","Blink","..."]}` . I know that this is inconvenient for you, but there is also an upside! You will be able to just GET `/json` and it will include all values from `/json/info`, `/json/effects`, `/json/palettes` and the new `json/state`, so just one request is required for all values.

The JSON response looks like this:

```json
{
	"state": {
		"on": true,
		"bri": 255,
		"transition": 3,
		"nl": {
			"on": false,
			"dur": 60,
			"fade": true,
			"tbri": 0
		},
		"udpn": {
			"send": false,
			"recv": true
		},
		"seg": [{
			"start": 0,
			"stop": 150,
			"len": 150,
			"col": [
				[252, 52, 1],
				[0, 0, 0],
				[0, 0, 0]
			],
			"fx": 65,
			"sx": 20,
			"ix": 128,
			"pal": 6,
			"sel": false,
			"rev": false,
			"cln": -1
		}]
	},
	"info": {
		"ver": "0.8.4-dev",
		"vid": 1903052,
		"leds": {
			"count": 150,
			"rgbw": false,
			"pin": [3],
			"pwr": 3019,
			"maxpwr": 5700,
			"maxseg": 1
		},
		"name": "PartyLights",
		"udpport": 21324,
		"live": false,
		"fxcount": 80,
		"palcount": 47,
		"arch": "esp8266",
		"core": "2_5_0",
		"freeheap": 13576,
		"uptime": 16561,
		"opt": ["alexa", "blynk", "cronixie", "huesync", "mobile-ui", "ota"],
		"brand": "wled",
		"product": "DIY light",
		"btype": "dev",
		"mac": "ecfabc20ff67"
	},
	"effects": [
		"Solid", "Blink", "Breathe", "Wipe", "Wipe Random", "Random Colors", "Sweep", "Dynamic", "Colorloop", "Rainbow",
		"Scan", "Dual Scan", "Fade", "Chase", "Chase Rainbow", "Running", "Saw", "Twinkle", "Dissolve", "Dissolve Rnd",
		"Sparkle", "Dark Sparkle", "Sparkle+", "Strobe", "Strobe Rainbow", "Mega Strobe", "Blink Rainbow", "Android", "Chase", "Chase Random",
		"Chase Rainbow", "Chase Flash", "Chase Flash Rnd", "Rainbow Runner", "Colorful", "Traffic Light", "Sweep Random", "Running 2", "Red & Blue", "Stream",
		"Scanner", "Lighthouse", "Fireworks", "Rain", "Merry Christmas", "Fire Flicker", "Gradient", "Loading", "In Out", "In In",
		"Out Out", "Out In", "Circus", "Halloween", "Tri Chase", "Tri Wipe", "Tri Fade", "Lightning", "ICU", "Multi Comet",
		"Dual Scanner", "Stream 2", "Oscillate", "Pride 2015", "Juggle", "Palette", "Fire 2012", "Colorwaves", "BPM", "Fill Noise", "Noise 1",
		"Noise 2", "Noise 3", "Noise 4", "Colortwinkle", "Lake", "Meteor", "Smooth Meteor", "Railway", "Ripple"
	],
	"palettes": [
		"Default", "Random Cycle", "Primary Color", "Based on Primary", "Set Colors", "Based on Set", "Party", "Cloud", "Lava", "Ocean",
		"Forest", "Rainbow", "Rainbow Bands", "Sunset", "Rivendell", "Breeze", "Red & Blue", "Yellowout", "Analogous", "Splash",
		"Pastel", "Sunset 2", "Beech", "Vintage", "Departure", "Landscape", "Beach", "Sherbet", "Hult", "Hult 64",
		"Drywet", "Jul", "Grintage", "Rewhi", "Tertiary", "Fire", "Icefire", "Cyane", "Light Pink", "Autumn",
		"Magenta", "Magred", "Yelmag", "Yelblu", "Orange & Teal", "Tiamat", "April Night"
	]
}
```

Every value of the state object can be changed by sending a POST to `/json` or `/json/state`. For example,
`{"bri":255}` will set maximum brightness, `{"seg":[{"col":[[0,255, 255]]}]}` will set a teal-ish color. I will create a full documentation of the API soon!

raymiec, 2019-03-05:
Yeah it's handy as hell, I have the lights on my bike, sometimes issues connecting to hotspot etc, so nice just to be able to plug my phone in.

Nice one, that will make everything a lot simpler, less calls too and lets me easily implement more features.

Just finished off the scenes, you can now save a scene and set it based on the weather. Will add more functionality to it soon.

I'm having issues with OTA updates on ver 0.82 & 0.83, When i send the bin file the ESP just hangs, can't even see anything when debugging. I'm not doing anything special just sending a multipart request with the file. It works on 0.8.4, i can send the 0.8.3 bin file and it downgrades fine. If you have any ideas on that one ?

Aircoookie, 2019-03-05:
Weather based scenes sound great!

About the issues with OTA updates, do they fail via browser as well? I did not fully write the updater code myself,  and when I switched to ESPAsyncWebserver I also changed the updater. Just took a look at the /update html though and it seems like 0.8.4 has `form action='/update'` while 0.8.3 has `form action=''`. So the URL is probably the culprit...

raymiec, 2019-03-05:
Ah nice one, i never even spotted that, will give it a go now and see if it makes a difference.

raymiec, 2019-03-09:
Just switching over to use the new json api now, regarding this segment stuff can you explain it when you get a sec, I see an array of segments with only one, i gather when segments are added this can increase to 3? under col theres an array with rgb, with the first one setting the color, whats the purpose of the other 2 ?    thanks

Aircoookie, 2019-03-09:
Yes, that's correct. "seg" is an array because WLED will support 8-16 segments soon. If you then want to set the 4th segment, you could either do `{"seg":[{}{}{}{"fx":1}]}` or `{"seg":[{"id":3,"fx":1}]}`, if you don't want to send the empty objects before that. But for now, anything but the first segment object will be ignored.

Color has three rgb arrays, because they are primary, secondary, and tertiary color. Secondary is essentially a background color, for example in "Blink", it will not blink to off, but to the secondary color.
Tertiary is very seldomly used at the moment (only in the "Tricolor" FX) and I'm not yet too sure about it.

raymiec, 2019-03-09:
Nice one i'm almost up to date with the app, one thing i have noticed is when setting fx even though i get back success the effect doesn't actually change.
{"seg":[{"id":0,"fx":21}]}
System.out: {"success":true}

sending again will change it.   It happens randomly but a lot.

raymiec, 2019-03-10:
setting bri i'm assuming should be changing the brightness, the value gets set but brightness never gets changed.

Aircoookie, 2019-03-10:
Good work! Just tried the brightness, works fine for me. Try if setting "on" to true or false changes anything (it shouldn't, `{"on":false}` is supposed to have the same effect as {"bri":0}. I'll take a closer look tomorrow.

raymiec, 2019-03-10:
I need to start pulling the latest master every time, its working now without any code change.  Is "bri" suppose to remember its last position? if you set it to 0 from say 200, it will set "on" to false but "bri" will stay at 200, i can just do a check to see if "on" is true or false before loading, was just wondering if this was intended ?

With the json success messages it seems to return success on successful connection and not if the command was actually executed successfully. The latter would be much better for dev.

Thanks man, new api is great, its literally cut the codebase in half.  (Was almost crying at the amount of code i had to delete lol).

Aircoookie, 2019-03-10:
Great! Yes, it's designed to work that way, since UIs that have power button + brightness slider can now set the correct slider position even if the light is off.

I agree that the success messages are not very useful. Basically, my json parser looks for any keys it expects, if it finds one, it will update the internal state, if not, it will keep it. It presently can't detect if anything was actually executed or if the json contained any extra/invalid keys. While I could implement "awareness" of which values were actually changed and responding with them, the performance overhead would not be worth it on ESP8266. However, you can be sure that the command was applied if you sent valid data and get `{"success":true}` (it means that the actual parsing was successful, so no JSON syntax errors). What I could however add is a command like `{"stateresp":true}`, which would send the (updated) state object instead of the success code. Let me know if that would be useful.

Awesome! Only shows that JSON is a superior protocol for an API compared to URL requests and XML!

raymiec, 2019-03-10:
To be honest i'm just about done so personally i won't benefit much more from it unless you plan to introduce a lot in the future.

JSON is great, especially when using php or something, Java it can be a pain because of all the excess code, creating the objects and arrays etc. I copied and pasted from my other function so i ended up sending something like {"state":[{"id":0,"fx":0}]} instead of {"seg":[{"id":0,"fx":0}]} which would still return {"success":true} as it's still valid JSON. 

Obvious mistake by me but sometimes hard to notice at 4.20am :).  Anything to mitigate silly things like that would be good but if it's costing on overheads then maybe not worth it.

Will tidy up android app as is and upload it later, i think it's pretty useable now. I have the segment stuff prepared so whilst waiting on WLED release for that, it gives me plenty of time to get IOS up to scratch. Also are you going to keep going with your apps or you wanna replace with mine?

raymiec, 2019-03-10:
unless im totally missing something here I think when bri is set to 0 that the actual value should be set as well. In the app if i set the slider to zero and the light goes off and refresh that device then the slider is going to goto the last position before i set it to zero i don't think we should be changing that, If i set a slider i kinda expect it to be at the same position when i go back. 

I'd maybe make the power and brightness independent of each other, start brightness from 1, and power actually turns it on and off.  Your thoughts?

raymiec, 2019-03-10:
Any chance of reverting on the changes with the brightness and power control, kinda seemed fine before, the way it is now means i need to check the state every time and then send the state along with any command.  Also if the slider is set to zero then refreshed it will take the last value before it was set. If i set a slider i kinda expect it to be in the same place when i go back, i can handle this in the background but seems like a lot of extra work for one extra feature of being able to set the brightness value when off, even just make them independent start brightness from 1 or something.  Your thoughts?

Aircoookie, 2019-03-10:
Once again, awesome work getting this all implemented so quickly! I'm not going to remove my app from the store for now, but I'll definitely add a link to yours once it's up :)

I can absolutely relate to such frustrating mistakes! Still, I can't easily add error handling, because ArduinoJSON will only look for expected keys and ignore the invalid ones. If I wanted to detect an unrecognized key, I'd have to go through the JSON and test for every key if it matches with something.

After thinking about it for some time, I agree that setting `{"on":false}` should be the preferred method for turning off as compared to `{"bri":0}`, because the state object instantly "jumps" back to {"bri":200} or similar. I still want to keep the dedicated "on" variable, since I don't like it that the brightness slider jumps when turning on or off via the power button. Still, I'd like the lights to turn on when the user modifies the "bri" value, even when `{"on":false}`.
There are two options I can think of:
- Leaving the API how it is now, but recommending the client to implement brightness sliders with range [1, 255]. That way the slider keeps its value when toggling with the on/off button. `{"bri":0}` wouldn't be forbidden, but not the preferred way as it would "jump back" if refreshed.
- Changing the API to the brightness 0-255 behavior (that the HTTP API has). This would essentially make the "on" field pointless, since it would just be `on =  (bri >0)`. In this case I'd add a "lastbri" field which contains the data "bri" has currently. Not the most elegant solution, but it can be really useful to have the "last" brightness when the light is off, so I'd like to maintain that.

This really comes down to whether you want to be able to turn the light off with the brightness slider. I still prefer the current way a bit.

By the way, just found a bug with this. Sending any API command that doesn't contain `{"on":false}` or `{"bri":0}` turns on the lights if they were off. Will fix this regardless of what option we'll decide on :)

raymiec, 2019-03-10:
Glad that's a bug and not a feature lol, it was that that was getting annoying, so if thats gets fixed it solves my issues and i don't need to send the state in every call. I'll try it when you've fixed, and with that the current intended functionality makes sense.

Obviously i'm only approaching this from my side and what my needs are, for me Ideally we shouldn't be moving the slider for the user, so setting the range from 1 - 255 solves that.  Power button is always in reach so not a big deal turning off completely with the slider, thats actually more effort.

No worries, i have someone i can pass it off to who will maintain and update, knowing me i'll get bored and jump onto something else in a few weeks lol.

Aircoookie, 2019-03-10:
Great to hear! The bug is now fixed in latest master :)

raymiec, 2019-03-10:
Perfect, tested, solves all my current issues.  Cheers

FoxExe, 2020-01-10:
@raymiec , any updates to your app?

I searched the entire web but I never found an Android app that would allow do something like your LedFX.
More precisely, I didn 't find any (free) application at all for Android, compatible with WLED and allowing to do "Music Visualization"...

am-cell, 2020-08-07:
@raymiec I can't seem to open the link . Could you possibly suggest me how to find your app in the play store or resend the link