wled/WLED #113: ESP-01 Problems

quorle, 2019-02-20:
I have problems with an ESP-01. At first everything wanted to flash with ATOM and PlatformIO, which unfortunately does not work. That's why I flashed everything with Arduino 1.8.8. I have set the boardcore to 2.3.0. The ESP-01 connects to the Wi-Fi and MQTT, but I can not do anything about MQTT. I send "on" or "off" and nothing happens. I have a black ESP-01, but it looks like it has only 512K memory (memory chip: PUYA P24Q80H) Now can help me?
I want to use 209 LED's WS2812b. Connected to GPIO3 of the ESP-01

Settings Script wled00:

```
/*
 * Main sketch, global variable declarations
 */
/*
 * @title WLED project sketch
 * @version 0.8.3
 * @author Christian Schwinne
 */

//ESP8266-01 (blue) got too little storage space to work with all features of WLED. To use it, you must use ESP8266 Arduino Core v2.3.0 and the setting 512K(64K SPIFFS).

//ESP8266-01 (black) has 1MB flash and can thus fit the whole program. Use 1M(64K SPIFFS).
//Uncomment some of the following lines to disable features to compile for ESP8266-01 (max flash size 434kB):

//You are required to disable over-the-air updates:
#define WLED_DISABLE_OTA

//You need to choose 1-2 of these features to disable:
#define WLED_DISABLE_ALEXA
#define WLED_DISABLE_BLYNK
#define WLED_DISABLE_CRONIXIE
//#define WLED_DISABLE_HUESYNC
#define WLED_DISABLE_INFRARED    //there is no pin left for this on ESP8266-01
#define WLED_DISABLE_MOBILE_UI

//to toggle usb serial debug (un)comment following line(s)
//#define WLED_DEBUG

//library inclusions
#include <Arduino.h>
#ifdef ARDUINO_ARCH_ESP32
 #include <WiFi.h>
 #include <ESPmDNS.h>
 #include "src/dependencies/webserver/WebServer.h"
 #include <HTTPClient.h>
 /*#ifndef WLED_DISABLE_INFRARED
  #include <IRremote.h>
 #endif*/ //there are issues with ESP32 infrared, so it is disabled for now
#else
 #include <ESP8266WiFi.h>
 #include <ESP8266mDNS.h>
 #include <ESP8266WebServer.h>
 #include <ESP8266HTTPClient.h>
 #ifndef WLED_DISABLE_INFRARED
  #include <IRremoteESP8266.h>
  #include <IRrecv.h>
  #include <IRutils.h>
 #endif
#endif

#include <EEPROM.h>
#include <WiFiUdp.h>
#include <DNSServer.h>
#ifndef WLED_DISABLE_OTA
 #include <ArduinoOTA.h>
 #include "src/dependencies/webserver/ESP8266HTTPUpdateServer.h"
#endif
#include "src/dependencies/time/Time.h"
#include "src/dependencies/time/TimeLib.h"
#include "src/dependencies/timezone/Timezone.h"
#ifndef WLED_DISABLE_ALEXA
 #define ESPALEXA_MAXDEVICES 1
 #include "src/dependencies/espalexa/Espalexa.h"
#endif
[... 139 lines cut ...]
bool analogClockSecondsTrail = false;         //Display seconds as trail of LEDs instead of a single pixel
bool analogClock5MinuteMarks = false;         //Light pixels at every 5-minute position

char cronixieDisplay[7] = "HHMMSS";           //Cronixie Display mask. See wled13_cronixie.ino
bool cronixieBacklight = true;                //Allow digits to be back-illuminated

bool countdownMode = false;                   //Clock will count down towards date
byte countdownYear = 19, countdownMonth = 1;  //Countdown target date, year is last two digits
byte countdownDay  =  1, countdownHour  = 0;
byte countdownMin  =  0, countdownSec   = 0;

byte macroBoot = 0;                           //macro loaded after startup
byte macroNl = 0;                             //after nightlight delay over
byte macroCountdown = 0;                      
byte macroAlexaOn = 0, macroAlexaOff = 0;
byte macroButton = 0, macroLongPress = 0;

//Security CONFIG
bool otaLock = false;                         //prevents OTA firmware updates without password. ALWAYS enable if system exposed to any public networks
bool wifiLock = false;                        //prevents access to WiFi settings when OTA lock is enabled
bool aOtaEnabled = true;                      //ArduinoOTA allows easy updates directly from the IDE. Careful, it does not auto-disable when OTA lock is on

uint16_t userVar0 = 0, userVar1 = 0;
```
[image]

Aircoookie, 2019-02-20:
Hi! First of all, it seems like your install is valid (although there is no need to set WiFi and MQTT server via the wled00.ino, the "official" way is to connect to the "WLED-AP" and input your data in the web settings interface).

I don't know if the MQTT connection was successful. Can you try uppercase "ON" and "OFF" or a number 0-255? You should also be able to open the esp8266 IP to access the Web UI.

GPIO2 is the default LED output in WLED. Please reconnect the LEDs or set LEDPIN to 3 in `NpbWrapper.h`

quorle, 2019-02-20:
Thank you for your hint. I did everything as you wrote. The Wi-Fi is displayed to me, but I can not log in, because the password was wrong. But that can not be so.

I also do not understand that with the MQTT. But I could not send anything about the topics or was not done on "off" or "on"

quorle, 2019-02-22:
Ok, the ESP-01 does not build a wifi.
So I can not do anything. Is there a trick? At the above settings, I have changed nothing more, except that:

> Hi! First of all, it seems like your install is valid (although there is no need to set WiFi and MQTT server via the wled00.ino, the "official" way is to connect to the "WLED-AP" and input your data [...]

Aircoookie, 2019-02-22:
Have you changed

```
char clientSSID[33] = "**MY-WLAN-SSID**";
char clientPass[65] = "**MY-WLAN-PASSWORD**";
```

back to:

```
char clientSSID[33] = "Your_Network";
char clientPass[65] = "";
```

?
Otherwise it will still (try to) connect to your network and not open its access point. You can also try to uncomment `#WLED_DEBUG` and you should see whether the connection was successful and the ESP IP address in the serial monitor.

quorle, 2019-02-25:
Hello, thanks for the info and thank you for helping me. I have downloaded a new, fresh version 0.8.3.

Here is the configuration from wled00:

```
/*
 * Main sketch, global variable declarations
 */
/*
 * @title WLED project sketch
 * @version 0.8.3
 * @author Christian Schwinne
 */

//ESP8266-01 (blue) got too little storage space to work with all features of WLED. To use it, you must use ESP8266 Arduino Core v2.3.0 and the setting 512K(64K SPIFFS).

//ESP8266-01 (black) has 1MB flash and can thus fit the whole program. Use 1M(64K SPIFFS).
//Uncomment some of the following lines to disable features to compile for ESP8266-01 (max flash size 434kB):

//You are required to disable over-the-air updates:
#define WLED_DISABLE_OTA

//You need to choose 1-2 of these features to disable:
//#define WLED_DISABLE_ALEXA
#define WLED_DISABLE_BLYNK
//#define WLED_DISABLE_CRONIXIE
//#define WLED_DISABLE_HUESYNC
#define WLED_DISABLE_INFRARED    //there is no pin left for this on ESP8266-01
#define WLED_DISABLE_MOBILE_UI

//to toggle usb serial debug (un)comment following line(s)
#define WLED_DEBUG

//library inclusions
#include <Arduino.h>
#ifdef ARDUINO_ARCH_ESP32
 #include <WiFi.h>
 #include <ESPmDNS.h>
 #include "src/dependencies/webserver/WebServer.h"
 #include <HTTPClient.h>
 /*#ifndef WLED_DISABLE_INFRARED
  #include <IRremote.h>
 #endif*/ //there are issues with ESP32 infrared, so it is disabled for now
#else
 #include <ESP8266WiFi.h>
 #include <ESP8266mDNS.h>
 #include <ESP8266WebServer.h>
 #include <ESP8266HTTPClient.h>
 #ifndef WLED_DISABLE_INFRARED
  #include <IRremoteESP8266.h>
  #include <IRrecv.h>
  #include <IRutils.h>
 #endif
#endif

#include <EEPROM.h>
#include <WiFiUdp.h>
#include <DNSServer.h>
#ifndef WLED_DISABLE_OTA
 #include <ArduinoOTA.h>
 #include "src/dependencies/webserver/ESP8266HTTPUpdateServer.h"
#endif
#include "src/dependencies/time/Time.h"
#include "src/dependencies/time/TimeLib.h"
#include "src/dependencies/timezone/Timezone.h"
#ifndef WLED_DISABLE_ALEXA
 #define ESPALEXA_MAXDEVICES 1
 #include "src/dependencies/espalexa/Espalexa.h"
#endif
#ifndef WLED_DISABLE_BLYNK
[... 163 lines cut ...]
**I've turned on the debug mode under 11500:**

```
14:16:15.899 -> ---MODULE DEBUG INFO---
14:16:15.899 -> Runtime: 420623
14:16:15.899 -> Unix time: 420
14:16:15.899 -> Free heap: 18664
14:16:15.899 -> Wifi state: 3
14:16:15.899 -> State time: 0
14:16:15.899 -> NTP last sync: 999000000
14:16:15.899 -> Client IP: 192.168.2.207
14:16:20.002 -> Not-Found HTTP call:
14:16:20.002 -> URI: /win&nocache=171836.90569045517
14:16:20.002 -> Body: 
14:16:20.002 -> API req: /win&nocache=171836.90569045517
14:16:20.913 -> ---MODULE DEBUG INFO---
14:16:20.913 -> Runtime: 425626
14:16:20.913 -> Unix time: 425
14:16:20.913 -> Free heap: 18664
14:16:20.913 -> Wifi state: 3
14:16:20.913 -> State time: 0
14:16:20.913 -> NTP last sync: 999000000
14:16:20.913 -> Client IP: 192.168.2.207
14:16:23.872 -> Attempting to connect MQTT...
```

quorle, 2019-02-25:
Update:

If you set this screen so the LEDs flicker.
The mx. current ist 65000, because external power supply.

[image]

Aircoookie, 2019-02-25:
> But how can it be that he immediately connects to the network? when I type in the IP address in the browser comes at once the web interface. But why?

The ESP saves this in flash. It is reading the WiFi configuration from your previous WLED install.
Is MQTT working now if you enter the broker IP in sync settings? (and reboot once)

> Update: If I want to assign a static IP in the configuration with the browser, it will not be saved.

Are you filling in the required gateway field as well with the IP of your router? If it still doesn't work, try to go to security settings and do a factory reset. It resets all settings, maybe something broke.

> If you set this screen so the LEDs flicker.

Thank you, I confirmed this! The LEDs get updated too often in the nightlight mode, which can cause flicker. Will fix soon.

quorle, 2019-02-26:
Hello, So I flashed the ESP-01 now with a blank.bin and then added your script in its original state. Now it's working.

But something is wrong with the MQTT. I can change the LED's color only with the topic: "ALL". If you enter the hexadecimal value for the respective lamp, the whole thing does not work.

Aircoookie, 2019-02-26:
Hi, can you try to subscribe to `wled/b4e62d44bdb3/g` in your client? (replace b4e62d44bdb3 with your esp's hexadecimal mac). The ESP should publish the current brightness there if it changes. If that works, you should also be able to update it with the `wled/b4e62d44bdb3` topic.

What do you mean by "partially single or multiple LEDs are controlled from time to time with only one color"? Are some LEDs getting stuck on a particular color or flickering?

quorle, 2019-02-27:
1. Hello, I tried to reach the ESP under: wled/dc4f224262e6 / g, unfortunately this is not possible. Only the IP leads to the ESP. If you now enter the IP/g in the browser, "not found" comes.

2. I've sent you a video with the settings I've just made that shows how the LEDs behave. I hope now you know what I mean?

[image]

Video:

[LED Video.zip](https://github.com/Aircoookie/WLED/files/2909280/LED.Video.zip)

Aircoookie, 2019-02-27:
Ok, as for MQTT, that's not how it works :) With IP/g you can't reach it, you need to _subscribe_ to the _topic_  `wled/dc4f224262e6/g` using an _MQTT client_ software, connected to an _MQTT broker_. It is a control protocol very different from HTTP (what browsers use), so entering it in the browser won't do anything. If you'd like to control the device via HTTP, either use the webinterface or the [HTTP API](https://github.com/Aircoookie/WLED/wiki/HTTP-request-API).  For example, you may send `IP/win&A=0` for off and `IP/win&A=255` for maximum brightness.

Perhaps what you mean with reaching the ESP via the hexadecimal is mDNS, not MQTT. You can in fact reach the webinterface using the browser with `http://wled-dc4f224262e6.local`, if you either use an Apple device or have Bonjour software installed on Windows. You may also change the mDNS address to something easier to remember in the WiFi settings!

About the LEDs behaving that way, likely the input voltage of the ESP->LED data signal is not high enough, so the LEDs have trouble interpreting the signal sometimes. I suppose you have the LEDs connected to GPIO2? Using any other pin other than GPIO2/3 might lead to such behavior as well, although unlikely.
You can try shortening the wire, or powering the LEDs from the end of the strip. Driving WS2812B with ESP8266 is technically out of spec, so if nothing else works, you might need a 3.3V->5V levelshifter.

quorle, 2019-02-28:
Thanks for the detailed answer. I'll try that with the level shifter at times. Since an ESP-01 outputs only 3.3V voltage and a Wemos D1 yes 5V voltage. The LEDs are all powered by an external 5V 12A power supply.

MQTT is affiliated with an MQTT broker under ioBroker. Personally, I find MQTT easier than sending anything via HTTP request.