wled/WLED #114: ESP8266 12-E crash at 300+ LEDs

MaxHD25, 2019-02-20:
Everything extra disabled already and I stepped my lights up incrementally until 300 was the failing point.

Aircoookie, 2019-02-20:
Thanks for the feedback! Did this happen with v0.8.3 or with the latest master? Also, what's your version of ESP8266 Arduino core?

MaxHD25, 2019-02-20:
2.3.0 and it was the version off master, I’ll try a release version. It does the same to my own web server code. I’m running 40amp external power supply and a logic shifter.  It might be an issue with running out of RAM to instruct the LEDs.

Aircoookie, 2019-02-20:
Yes, I just tried the full 1200 LEDs and instantly got my light bricked! Really gotta provide a "safety" sketch that resets user settings. That said, try using arduino core v2.5.0, it makes 8kb more ram available compared to 2.3.0, so it should be less of an issue.

However, 300 LEDs should be way too few to start crashing. I need to investigate this, supporting at least 600 LEDs is a must!

MaxHD25, 2019-02-20:
Well I’m glad it’s not just an issue with mine. It’s a hard crash until power is reset. The reset button doesn’t function. If it wasn’t on a friends wall i’d be able to tell you if serial gave an reason for the crash. I’ll have him try 2.50 and see if it works. The final number we want is 586. Should I try clocking the chip to 160MHz or would that be for stability reasons?

Aircoookie, 2019-02-20:
I've identified this as a likely memory issue.
Did you, by any chance use the DMA mode (LEDPIN 3)?
My findings (using latest master and core 2.5.0) are:

- GPIO2 UART mode works with 1200 RGBW LEDs and Huesync, MQTT activated
- GPIO3 DMA works with 300 LEDs, but even enabling RGBW starts to show issues.

This actually makes sense, because NeoPixelBus docs state that each RGB pixel needs 5 times as many RAM in DMA mode as a buffer than in UART mode.

You can tell if the ESP is running out of memory. Either it completely crashes, or some parts of the UI might not load or display nonsense.

I will do my best to improve the ram usage, so that more than 300 LEDs will become possible using DMA.

MaxHD25, 2019-02-21:
So I updated the board and it definitely freed up some memory. However the issue seemed to root from the power supply. When powered from only USB and no VIN it doesn’t crash anymore. It’s really odd to me considering that I feel like it would just flat out not work or be fine in a case where VIN is messed up. However, when we incrememnted it up it was fine till about 250 last night before we figured this out.

debsahu, 2019-02-22:
Move all `DEBUG.print(xxx)` to `DEBUG.print(F(xxx))` and println as well. Saves a lot of RAM.

jsmath, 2019-10-18:
I see that the code is still limiting to 300 LEDs when using DMA.  I am hoping to use 360 LEDs with DMA.  Have there been any improvements in this regard?  What is the limit on LEDs with DMA if using arduino core v2.5.0 and changing the DEBUG.print as mentioned above?

closed 2020-04-05 by stale[bot]