wled/WLED #115: Error: \wled00.ino.cpp: No such file or directory with Atom/PlatformIO

thomasklein1975, 2019-02-22:
Hi,

I am a beginner. Hope that my question is not too stupid.

By using Atom to flash my wemos d1 mini I got this message while uploading/compiling:
`Compiling .pioenvs\d1_mini\src\wled00.ino.cpp.o
Generating LD script .pioenvs\d1_mini\ld\eagle.app.v6.common.ld
xtensa-lx106-elf-g++: error: wled00\wled00.ino.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.`

What can i do? 

Greetings
Thomas

Aircoookie, 2019-02-22:
Unfortunately I don't really know much about PIO, so I don't know what exactly is going wrong here.

@wiesendaniel @debsahu Do you have any ideas?

debsahu, 2019-02-22:
I personally use VSCode and PIO. But I'm guessing the idea is same for ATOM and PIO. PIO was not installed correctly:

* uninstall PIO
* close ATOM
* Remove . platformio from home folder (C:\Users\<username>\.platformio)
* Reopen ATOM
* Install PIO and restart ATOM
* Wait patiently for 5-10 min until PIO installation is complete

thomasklein1975, 2019-02-22:
Thx for very fast response!

Now I got this one:
`..
Z:/Software/Rapsi/wemos/WLED-master/wled00/wled00.ino:367:27: warning: left operand of comma operator has no effect [-Wunused-value]
IPAddress realtimeIP = (0,0,0,0);
..

Z:/Software/Rapsi/wemos/WLED-master/wled00/wled00.ino:563:19: error: new declaration 'bool reconnectHue()'
Z:/Software/Rapsi/wemos/WLED-master/wled00/wled00.ino:555:6: error: ambiguates old declaration 'void reconnectHue()'
..
Z:/Software/Rapsi/wemos/WLED-master/wled00/wled05_init.ino:256:16: error: 'class EspClass' has no member named 'getMaxFreeBlockSize'
..
Z:/Software/Rapsi/wemos/WLED-master/wled00/wled18_server.ino:232:20: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
oappend(";--cFn:");               oappend(cssFont);
....

*** [.pioenvs\d1_mini\src\wled00.ino.cpp.o] Error 1`

thomasklein1975, 2019-02-22:
where can i find the *.bin file to flash?

While using Arduino I get this message:
`C:\Users\tklein\Documents\Arduino\WLED-master\WLED-master\wled00\wled00.ino:83:22: fatal error: IRrecv.h: No such file or directory

   #include <IRrecv.h>

But there is this folder in lib "IRremoteESP8266-master"
                      ^

compilation terminated.

exit status 1`

Aircoookie, 2019-02-22:
You need to download the IRremoteESP8266 library. Alternatively, you can also uncomment `#define WLED_DISABLE_INFRARED` in wled00.ino. The bins are on the github [releases](http://bin.wled.me) page!

thomasklein1975, 2019-02-22:
mmhhmm. so it is in file:
//#define WLED_DISABLE_INFRARED    //there is no pin left for this on ESP8266-01

thomasklein1975, 2019-02-22:
@Aircoookie thx a lot. With the bin it works. Cool tool! 
Can I trigger by using a url like 192.168.0.44?power=1. I have activated mqtt and alexa and the device freezes. I would like to trigger device from my fhem server.

DuschdrBabbe, 2019-02-22:
FHEM and mqtt work also have it running without problems... 
config your mqtt server and topic...

`defmod Testlicht MQTT_DEVICE
attr Testlicht DbLogExclude .*
attr Testlicht IODev Mosquitto
attr Testlicht alexaName Testlicht
attr Testlicht devStateIcon ON=hue_filled_lightstrip@green OFF=hue_filled_lightstrip@#FF8000
attr Testlicht event-on-change-reading .*
attr Testlicht eventMap OFF:off ON:on
attr Testlicht genericDeviceType light
attr Testlicht group WLED
attr Testlicht icon hue_filled_lightstrip
attr Testlicht publishSet wled/testlicht
attr Testlicht publishSet_api wled/testlicht/api
attr Testlicht stateFormat Helligkeit:brightness
attr Testlicht subscribeReading_brightness wled/testlicht/g
attr Testlicht subscribeReading_color wled/testlicht/c
attr Testlicht webCmd on:off`

debsahu, 2019-02-22:
@wiesendaniel @Aircoookie Can we add this in **platformio.ini** under **lib_deps_external**
```
IRremoteESP8266@2.5.5
```
How is Travis CI doing? It should have caught this missing library.

debsahu, 2019-02-22:
@thomasklein1975 After successful building, the firmware.bin will be inside **.pioenvs**. You can use the precompiled binaries built by @Aircoookie

Aircoookie, 2019-02-22:
@thomasklein1975 sure, just use the url `192.168.0.44/win&T=1` About MQTT freezes, are you sure the mqtt broker URL is correct and the broker accepts connections without auth on port 1883? In the next version (actually already in master) MQTT is async, so it should no longer cause freezes on unsuccessful connections.

@debsahu I will update the `platformio.ini` with that, thanks! I disabled travis because it was failing for every build and I'm unsure how to configure it. Let me know if you want me to re-enable it and, if you have the time (this is also for @wiesendaniel , if you are interested) send me a PR with its proper configuration. Thank you!

thomasklein1975, 2019-02-23:
wow. many thx for all these very fast and friendly answers and tips!! this is really cool and make a beginner much more easy to be helped!
hopefull i can test further about this tool this weekend.

best wishes
thomas

closed 2019-10-25 by Aircoookie