wled/WLED #116: Initital Setup-Tutorial??

lightrunner86, 2019-02-28:
Hello, i'd really like to know if this project is aktually able to run? I already tried 3 versions of arduino (1.8.8. last) and only getting anoying messages that this&that header file is missing (after finaly gathering all seperate files from different locations?)
now I'm only getting this:

In file included from C:\Users\user1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Arduino.h:38:0,

                 from C:\Users\user1\Desktop\arduino-1.8.8\libraries\NeoPixelBus-master\src/NeoPixelBus.h:28,

                 from C:\Users\user1\Desktop\arduino-1.8.8\libraries\NeoPixelBus-master\src/NeoPixelBrightnessBus.h:29,.............................................

and so on...

would anyone please be so kind and wrap All additional needed files in one *Zip /*.Rar & Share here

lightrunner86, 2019-02-28:
What really help allot. Is there a list of possible Boards? esp8266Mod d1 Mini Supported? 
thank you very much in advance :-)

lightrunner86, 2019-02-28:
I also tried the *.bin Update Variant via WiFi First. After Uploading it restarts at Firmware Upload Page?

Aircoookie, 2019-02-28:
Hi @lightrunner86 ,
you can try to install version 2.3.5 of NeoPixelBus! It is really easy via the Arduino Library manager.
The bin update should work in any case, if you still end up at the upload page after reboot, try deleting the `/update` part from the URL and you should see the welcome page or control UI!

lightrunner86, 2019-02-28:
Hello, thx for your quick response, i "downgraded to v.2.3.5:

Build-Optionen wurden verändert, alles wird neu kompiliert
In file included from sketch\WS2812FX.h:30:0,

                 from sketch\WS2812FX_fcn.cpp:27:

NpbWrapper.h:159:39: error: 'PIXELMETHOD' was not declared in this scope

   NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>*  _pGrb;

                                       ^

NpbWrapper.h:159:50: error: template argument 2 is invalid

   NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>*  _pGrb;

                                                  ^

NpbWrapper.h:160:39: error: 'PIXELMETHOD' was not declared in this scope

   NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>* _pGrbw;

                                       ^

NpbWrapper.h:160:50: error: template argument 2 is invalid

   NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>* _pGrbw;

                                                  ^

sketch\NpbWrapper.h: In member function 'void NeoPixelWrapper::Begin(NeoPixelType, uint16_t)':

NpbWrapper.h:67:57: error: 'PIXELMETHOD' was not declared in this scope

         _pGrb = new NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>(countPixels, LEDPIN);

                                                         ^

NpbWrapper.h:67:68: error: template argument 2 is invalid

         _pGrb = new NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>(countPixels, LEDPIN);

                                                                    ^

NpbWrapper.h:68:16: error: request for member 'Begin' in '*((NeoPixelWrapper*)this)->NeoPixelWrapper::_pGrb', which is of non-class type 'int'

         _pGrb->Begin();

                ^

NpbWrapper.h:72:58: error: the value of 'PIXELMETHOD' is not usable in a constant expression

         _pGrbw = new NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>(countPixels, LEDPIN);

                                                          ^

sketch\NpbWrapper.h:67:57: note: 'PIXELMETHOD' was not declared 'constexpr'

         _pGrb = new NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>(countPixels, LEDPIN);

                                                         ^

NpbWrapper.h:72:69: error: type/value mismatch at argument 2 in template parameter list for 'template<class T_COLOR_FEATURE, class T_METHOD> class NeoPixelBrightnessBus'

         _pGrbw = new NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>(countPixels, LEDPIN);

                                                                     ^

NpbWrapper.h:72:69: error: expected a type, got 'PIXELMETHOD'
[... 482 lines cut ...]
sketch\NpbWrapper.h: In member function 'void NeoPixelWrapper::ClearTo(RgbColor)':

NpbWrapper.h:142:37: error: request for member 'ClearTo' in '*((NeoPixelWrapper*)this)->NeoPixelWrapper::_pGrb', which is of non-class type 'int'

       case NeoPixelType_Grb: _pGrb->ClearTo(color);   break;

                                     ^

NpbWrapper.h:143:38: error: request for member 'ClearTo' in '*((NeoPixelWrapper*)this)->NeoPixelWrapper::_pGrbw', which is of non-class type 'int'

       case NeoPixelType_Grbw:_pGrbw->ClearTo(color);  break;

                                      ^

sketch\NpbWrapper.h: In member function 'void NeoPixelWrapper::ClearTo(RgbwColor)':

NpbWrapper.h:151:38: error: request for member 'ClearTo' in '*((NeoPixelWrapper*)this)->NeoPixelWrapper::_pGrbw', which is of non-class type 'int'

       case NeoPixelType_Grbw:_pGrbw->ClearTo(color);  break;

                                      ^

'PIXELMETHOD' was not declared in this scope

...with this effect?

Aircoookie, 2019-02-28:
Thanks for sending the error message! Did you change anything in the file NpbWrapper.h?
Also, are you trying to compile 0.8.3 or the current master?
Lastly, what is your installed ESP8266 arduino core version?

Don't worry, if we can't get compiling to work, you can still install the binary :)

lightrunner86, 2019-02-28:
..... Sorry so sooo much :-/ ... I'm new on github, and somehow mixed up 2 versions... *shame* 
i finally got it running :-) 
A great Idea !! hope i'm skilled enough soon to also help here :-)

closed 2019-02-28 by lightrunner86

Aircoookie, 2019-02-28:
No worries :) Have fun with the software!

lightrunner86, 2019-03-01:
The issue for my search findung this projekt was that i built a ws2812b Led
matrix in a 3*27 glassbrick wall. ^^ , -but stand-alone....
making value changes from analog input visiblile in the matrix/ ws2812b ist
no big deal.
I'd like for example getting descrete notifications in light (Effekt/color)
change if oil level runs low, mail in the mailbox, motion on PIR, Hooked
via esp8266, but thats just a thought ;-)