wled/WLED #118: Effects list not loading

amstan, 2019-03-04:
Something in commit https://github.com/Aircoookie/WLED/commit/6e76fc0aa7e914d01ab7fc37813f8bfb2f91d97d broke the effects list.

Now it just says "Loading...", but it never does.

A quick look at the code suggests that the new way is probably some javascript that loads the list. I guess it might be be working yet?

Aircoookie, 2019-03-04:
That's right, the effect and palette lists are dynamically loaded in JS.

Here are some steps if you want to help debug why it isn't working for you:

- What is your Arduino IDE version, Arduino ESP8266 core version and LwIP version? It's ok if you don't know some of it, but I use 1.8.7 with Core 2.5.0 and LwIP 1.4 High Bandwidth

- Try opening `[ESP-IpAddress]/json/effects` in your browser. It should display a JSON list of the effect names (from https://github.com/Aircoookie/WLED/commit/6e76fc0aa7e914d01ab7fc37813f8bfb2f91d97d on)

- You could also try the [0.8.3 binary](http://bin.wled.me), it is directly flashable via WLED in "Security Settings -> Manual OTA update". If it works with the binary, it is an issue with how the firmware is compiled for you, if it doesn't, it is a network or browser issue.

Thank you!

Dgemily, 2019-08-22:
thank you very much for this wonderful wled !!!
for peolple having the same issue. I use to have it and I just change the uplaod speed from 921600 to 115200 and it fixed it !

timothybrown, 2019-08-22:
I get this same issue and have been meaning to take a look at this. It seems to only be an issue in Safari for me. I also notice that if I leave the effects page once the “Loading...” entry comes up (and go to settings or the preset page for example) and then hit my browser’s refresh button, then go back to the effects page, the list will be loaded and work fine.

I’m not using the Arduino IDE, but instead PlatformIO with Core 2.5.0 and LwIP 1.4 High Bandwidth.

closed 2020-04-05 by stale[bot]