wled/WLED #119: Wifi only stays up for 10 minutes

amstan, 2019-03-04:
On my setup, wifi only seems to stay alive for 10 minutes. Then it eventually disconnects, cannot access it anymore. I need to reboot the board to connect to it again.

Here's some serial logs:
    rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    configsip: 0, SPIWP:0xee
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:1
    load:0x3fff0018,len:4
    load:0x3fff001c,len:1100
    load:0x40078000,len:10312
    load:0x40080400,len:6460
    entry 0x400806a4
    Load EEPROM
    CSSID: MyAPName NO AP
    C_NC
    C_NC
    C_NC
    C_NC
    
    Connected! IP address: 10.255.0.160
    HTTP server started
    mDNS responder started
    Ada
    Not-Found HTTP call:
    URI: /win&nocache=37805.69845868653
    API req: /win&nocache=37805.69845868653
    Not-Found HTTP call:
    URI: /win&nocache=106202.2194270038
    API req: /win&nocache=106202.2194270038
    Not-Found HTTP call:
    URI: /win&nocache=171716.20874267467
    API req: /win&nocache=171716.20874267467
    ######### I closed the web ui here
    ---MODULE DEBUG INFO---
    Runtime: 5001
    Unix time: 5
    Free heap: 223420
    Wifi state: 3
    State time: 0
    NTP last sync: 999000000
    Client IP: 10.255.0.160
    ---MODULE DEBUG INFO---
    Runtime: 10004
    Unix time: 10
    Free heap: 228340
    Wifi state: 3
    State time: 0
    NTP last sync: 999000000
    Client IP: 10.255.0.160
    ########## Snip, boring
    ---MODULE DEBUG INFO---
    Runtime: 190112
    Unix time: 190
    Free heap: 228340
    Wifi state: 3
    State time: 0
    NTP last sync: 999000000
    Client IP: 10.255.0.160
    ---MODULE DEBUG INFO---
    Runtime: 195115
    Unix time: 195
    Free heap: 228340
    Wifi state: 3
    State time: 0
    NTP last sync: 999000000
    Client IP: 10.255.0.160
    ---MODULE DEBUG INFO---
    Runtime: 200118
    Unix time: 200
    Free heap: 229412
    Wifi state: 6
    State time: 200118
    NTP last sync: 999000000
    Client IP: 0.0.0.0
    ---MODULE DEBUG INFO---
    Runtime: 205123
    Unix time: 205
    Free heap: 229412
    Wifi state: 6
    State time: 200118
    NTP last sync: 999000000
    Client IP: 0.0.0.0
    ---MODULE DEBUG INFO---
    Runtime: 210128
    Unix time: 210
    Free heap: 229412
    Wifi state: 6
    State time: 200118
    NTP last sync: 999000000
    Client IP: 0.0.0.0
    ---MODULE DEBUG INFO---
    Runtime: 215133
    Unix time: 215
    Free heap: 229412
    Wifi state: 6
    State time: 200118
    NTP last sync: 999000000
    Client IP: 0.0.0.0

amstan, 2019-03-04:
I tried another sketch that gives a small server, also has the same problem.

Well. Turns out it's just a matter of my router.

I run openwrt on a WRT3200ACMv1.

I upgraded the firmware to latest (18.06). Things got a little more stable (without the 10 min disconnects). But it was way laggier than before.

Eventually I managed to make it work properly by adjusting my settings. I made the 2.4GHz network kinda suck now, but I don't use much else on there anyway.

Here's another page that talks about issues like this between the esp8266 and the same router, from the wifi driver's issue tracker actually. https://github.com/kaloz/mwlwifi/issues/278

Here's my final openwrt config that makes things stable:
    root@router:~# cat /etc/config/wireless 

    config wifi-device 'radio0'
            option type 'mac80211'
            option hwmode '11a'
            option country 'US'
            option distance '30'
            option htmode 'VHT80'
            option channel '100'
            option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'

    config wifi-iface 'default_radio0'
            option device 'radio0'
            option network 'lan'
            option mode 'ap'
            option encryption 'psk2'
            option key '***********'
            option ssid '*************'

closed 2019-03-04 by amstan

Aircoookie, 2019-03-04:
Thanks for the report! Definitely something to be aware of, even though it is not a direct issue with WLED.