wled/WLED #120: Crash.

raymiec, 2019-03-07:
Tried your latest and the closest i can get to working is, 

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
Load EEPROM
CSSID: Theinternet NO AP
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC

Connected! IP address: 192.168.1.26
HTTP server started

Exception (2):
epc1=0x3fff5104 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff5104 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb80 end: 3fffffc0 offset: 01a0
3ffffd20:  3fff6c72 0000002e 00000010 4027da4f  
3ffffd30:  3ffffda3 4023ed29 3fff2f40 00000000  
3ffffd40:  3fff3380 3fff3348 3fff6c2c 4023efab  
3ffffd50:  00000000 3fff3348 3fff6c2c 4023f39c  
3ffffd60:  3ffffd90 3fff3300 3fff6894 4023efab  
3ffffd70:  00000002 3fff3300 3fff6894 3fff6c2c  
3ffffd80:  3fff6c2c 3fff6c18 3fff3348 40246c95  
3ffffd90:  005e0001 fffffb00 ffffffff 3fff6894  
3ffffda0:  3fff6c80 3fff6c18 3fff3348 40247784  
3ffffdb0:  0000005a 40248250 00000074 40100c78  
3ffffdc0:  00000018 3fff334c 00000001 00000000  
3ffffdd0:  00000002 3fff32a8 00000004 3fff2dd4  
3ffffde0:  00000010 00000010 00000000 3fff6c14  
3ffffdf0:  00000000 3fff6c2c 3fff6c98 402477cc  
3ffffe00:  3fff3348 3ffffe30 00000004 3fff6c14  
3ffffe10:  00000016 3fff3348 3fff6c14 40246ef1  
3ffffe20:  3fff3348 3ffffe30 00000004 4023f3f0  
3ffffe30:  00000494 00000000 3fff350c 4023f42c  
3ffffe40:  3fff6c18 3fff3348 000006f6 40246fc2  
3ffffe50:  3fff6ba4 00000001 3fff6bc4 00000020  
3ffffe60:  3ffffec8 3fff3348 3fff6c14 4024719b  
3ffffe70:  3ffffec8 3ffebb4c 3fff3348 40247227  
3ffffe80:  3fff6ba4 00000001 40100376 3ffebb4c  
3ffffe90:  3fff29dc 00000000 3fff29dc 40221908  
3ffffea0:  00000000 00000000 3fff2a20 40221e5c  
3ffffeb0:  3fff29c0 00000000 3fff2a14 40222990  
3ffffec0:  3fff29c0 00000001 fb0000e0 40220215  
3ffffed0:  3fff29c0 4023ebe4 3fff29f0 3fff1988  
3ffffee0:  3fff29c0 00000000 3fff29dc 4022026e  
3ffffef0:  3fff29c0 00000000 3fff29dc 4021ffdd  
3fffff00:  3fff5f34 3fff6ab4 4021fb0c 4021fbec  
3fffff10:  00000000 00000000 3fff2a64 40233484  
3fffff20:  00000000 3fff29dc 3fff2a64 40229d24  
3fffff30:  3fff3b74 3ffe8746 40229630 40234228  
3fffff40:  3fff2a64 3ffe8746 3fff2b58 4021b65d  
3fffff50:  3fff5f34 1a01a8c0 4020d460 4020d3bc  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  00000001 feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe 3fff2c7c  
3fffff90:  3fffdad0 00000000 3fff2c4c 4021b75c  
3fffffa0:  feefeffe feefeffe feefeffe 40232ea8  
3fffffb0:  feefeffe feefeffe 3ffe8880 40100ec9  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

If you have any ideas, i tried a bunch of different neopixelbus versions as well.

Cheers

meelow, 2019-03-07:
I got the bootloader messages when I configured the wrong led pin ...

Aircoookie, 2019-03-07:
oh dear! I will try and reproduce this once i get home. It crashes directly after server init, so probably a memory or async issue. What esp8266 core version did you use?

raymiec, 2019-03-07:
I was on 2.42 I think, just updated and tried again but it's the same, I'll maybe clear everything and update and try again in case it's an issue on this mac.

Aircoookie, 2019-03-07:
Try this binary: 
[1903071_nodemcu.zip](https://github.com/Aircoookie/WLED/files/2942203/1903071_nodemcu.zip)

There is currently a very weird issue. Sometimes the module works flawlessly and other times the internal webserver will hang like crazy. I'm still investigating on that one.

raymiec, 2019-03-07:
Yeah that binary worked.  

---WLED 0.8.4-dev 1903071 INIT---
esp8266 2_5_0
heap 26800
LEDs inited. heap usage ~1088
Load EEPROM
CSSID: Theinternet NO AP
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC

Connected! IP address: 192.168.1.26
HTTP server started
mDNS responder started
Ada
---DEBUG INFO---
Runtime: 10000
Unix time: 10
Free heap: 16992
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 192.168.1.26

raymiec, 2019-03-07:
tried again with the latest, updated to latest esp core and neopixelbus, trying various versions of both still crashing.  Your bin file should do me for now, i only need it for dev at the moment.

1384, room 16 
tail 8
chksum rd $⸮⸮| ⸮l⸮|⸮l⸮c|⸮⸮⸮⸮;⸮c⸮c⸮⸮g'⸮log⸮⸮⸮cx⸮⸮l;l;lx⸮o⸮⸮d⸮⸮#g⸮|d⸮⸮#⸮⸮no⸮ $⸮⸮l ⸮ogl n{⸮⸮⸮gc⸮ls⸮⸮nc⸮d⸮l⸮8⸮⸮⸮⸮ ⸮⸮o⸮
---WLED 0.8.4-dev 1903071 INIT---
esp8266 2_5_0
heap 30016
LEDs inited. heap usage ~1088
Load EEPROM
CSSID: Theinternet NO AP
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC

Connected! IP address: 192.168.1.26
HTTP server started

Exception (2):
epc1=0x3fff5134 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff5134 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb80 end: 3fffffc0 offset: 01a0
3ffffd20:  3fff6ca2 0000002e 00000010 4027daef  
3ffffd30:  3ffffda3 4023edc5 3fff2f70 00000000  
3ffffd40:  3fff33b0 3fff3378 3fff6c5c 4023f047  
3ffffd50:  00000000 3fff3378 3fff6c5c 4023f438  
3ffffd60:  3ffffd90 3fff3330 3fff68c4 4023f047  
3ffffd70:  00000002 3fff3330 3fff68c4 3fff6c5c  
3ffffd80:  3fff6c5c 3fff6c48 3fff3378 40246d31  
3ffffd90:  005e0001 fffffb00 ffffffff 3fff68c4  
3ffffda0:  3fff6cb0 3fff6c48 3fff3378 40247820  
3ffffdb0:  0000005a 402482ec 00000074 40100c78  
3ffffdc0:  00000018 3fff337c 00000001 00000000  
3ffffdd0:  00000002 3fff32d8 00000004 3fff2e04  
3ffffde0:  00000010 00000010 00000000 3fff6c44  
3ffffdf0:  00000000 3fff6c5c 3fff6cc8 40247868  
3ffffe00:  3fff3378 3ffffe30 00000004 3fff6c44  
3ffffe10:  00000016 3fff3378 3fff6c44 40246f8d  
3ffffe20:  3fff3378 3ffffe30 00000004 4023f48c  
3ffffe30:  00000494 00000000 3fff353c 4023f4c8  
3ffffe40:  3fff6c48 3fff3378 000006f6 4024705e  
3ffffe50:  3fff6bd4 00000001 3fff6bf4 00000020  
3ffffe60:  3ffffec8 3fff3378 3fff6c44 40247237  
3ffffe70:  3ffffec8 3ffebb7c 3fff3378 402472c3  
3ffffe80:  3fff6bd4 00000001 40100376 3ffebb7c  
3ffffe90:  3fff2a0c 00000000 3fff2a0c 402219a8  
3ffffea0:  00000000 00000000 3fff2a50 40221efc  
3ffffeb0:  3fff29f0 00000000 3fff2a44 40222a30  
3ffffec0:  3fff29f0 00000001 fb0000e0 402202b5  
3ffffed0:  3fff29f0 4023ec80 3fff2a20 3fff19b8  
3ffffee0:  3fff29f0 00000000 3fff2a0c 4022030e  
3ffffef0:  3fff29f0 00000000 3fff2a0c 4022007d  
3fffff00:  3fff5f64 3fff6ae4 4021fbac 4021fc8c  
3fffff10:  00000000 00000000 3fff2a94 40233524  
3fffff20:  00000000 3fff2a0c 3fff2a94 40229dc4  
3fffff30:  3fff3ba4 3ffe8746 402296d0 402342c8  
3fffff40:  3fff2a94 3ffe8746 3fff2b88 4021b6f9  
3fffff50:  3fff5f64 1a01a8c0 4020d460 4020d3bc  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  00000001 00000000 00007540 feefeffe  
3fffff80:  feefeffe feefeffe feefeffe 3fff2cac  
3fffff90:  3fffdad0 00000000 3fff2c7c 4021b7f8  
3fffffa0:  feefeffe feefeffe feefeffe 40232f48  
3fffffb0:  feefeffe feefeffe 3ffe8880 40100ec9  
<<<stack<<<

debsahu, 2019-03-07:
Could you decode stack using https://github.com/me-no-dev/EspExceptionDecoder to get useful information?

Aircoookie, 2019-03-09:
Finally been able to reproduce this issue. Seems to be a potentially memory-related issue which happens during mDNS initialisiation. Maybe it doesn't like advertising two services (wled and http), but I'd like to keep it that way if possible. Anyways, this issue only occurs with lwip 2 and neopixelbus 2.4.x (for whatever reason the neopixelbus version matters for this). Choosing either lwip 1.4 or neopixelbus 2.3.5, the issue doesn't occur.

```
---WLED 0.8.4-dev 1903091 INIT---
esp8266 2_5_0
heap 30016
LEDs inited. heap usage ~184
Load EEPROM
CSSID: csnet NO AP
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC

Connected! IP address: 10.10.1.176
HTTP server started

Exception (2):
epc1=0x3fff4d2c epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff4d2c depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb80 end: 3fffffc0 offset: 01a0
3ffffd20:  3fff691a 0000002e 00000010 4027d54f  
3ffffd30:  3ffffda3 4023e759 3fff2f70 00000000  
3ffffd40:  3fff33b0 3fff3378 3fff68d4 4023e9db  
3ffffd50:  00000000 3fff3378 3fff68d4 4023edcc  
3ffffd60:  3ffffd90 3fff3330 3fff653c 4023e9db  
3ffffd70:  00000002 3fff3330 3fff653c 3fff68d4  
3ffffd80:  3fff68d4 3fff68c0 3fff3378 402466c5  
3ffffd90:  005e0001 fffffb00 ffffffff 3fff653c  
3ffffda0:  3fff6928 3fff68c0 3fff3378 402471b4  
3ffffdb0:  0000005a 40247c80 00000074 40100c80  
3ffffdc0:  00000018 3fff337c 00000001 00000000  
3ffffdd0:  00000002 3fff32d8 00000004 3fff2e04  
3ffffde0:  00000010 00000010 00000000 3fff68bc  
3ffffdf0:  00000000 3fff68d4 3fff6940 402471fc  
3ffffe00:  3fff3378 3ffffe30 00000004 3fff68bc  
3ffffe10:  00000016 3fff3378 3fff68bc 40246921  
3ffffe20:  3fff3378 3ffffe30 00000004 4023ee20  
3ffffe30:  00000494 00000000 3fff353c 4023ee5c  
3ffffe40:  3fff68c0 3fff3378 e49ba5e3 402469f2  
3ffffe50:  00000000 00000000 4bc6a7f0 00000020  
3ffffe60:  3ffffec8 3fff3378 3fff68bc 40246bcb  
3ffffe70:  3ffffec8 3ffebb3c 3fff3378 40246c57  
3ffffe80:  001388a8 00000001 3fffff10 3ffebb3c  
3ffffe90:  3fff2a0c 00000000 3fff2a0c 402219bc  
3ffffea0:  00000000 00000000 3fff2a50 40221f10  
3ffffeb0:  3fff29f0 00000000 3fff2a44 40222a44  
3ffffec0:  3fff29f0 00000001 fb0000e0 402202c9  
3ffffed0:  3fff29f0 4023e614 3fff2a20 3fff1978  
3ffffee0:  3fff29f0 00000000 3fff2a0c 40220322  
3ffffef0:  3fff29f0 00000000 3fff2a0c 40220091  
3fffff00:  3fff5bdc 3fff675c 4021fbc0 4021fca0  
3fffff10:  00000000 00000000 3fff2a94 40232ee4  
3fffff20:  00000000 3fff2a0c 3fff2a94 40229dd8  
3fffff30:  3fff3ba4 3ffe8746 402296e4 40233c88  
3fffff40:  3fff2a94 3ffe8746 3fff2b88 4021b711  
3fffff50:  3fff5bdc b0010a0a 4020d460 4020d3bc  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  00000001 00000000 00007540 feefeffe  
3fffff80:  feefeffe feefeffe feefeffe 3fff2cac  
3fffff90:  3fffdad0 00000000 3fff2c7c 4021b810  
3fffffa0:  feefeffe feefeffe feefeffe 40232908  
3fffffb0:  feefeffe feefeffe 3ffe8840 40100ed1  
<<<stack<<<
[... 29 lines cut ...]
0x40246bcb: igmp_start_timer at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 695
:  (inlined by) igmp_joingroup_netif at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 522
0x40246c57: igmp_joingroup at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 465
0x402219bc: esp8266::MDNSImplementation::MDNSResponder::_allocUDPContext() at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266mDNS\src/LEAmDNS_Helpers.cpp line 202
0x40221f10: _ZN7esp826613polledTimeout15timeoutTemplateILb0ENS0_11YieldPolicy9DoNothingEE5resetEv$isra$7 at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/PolledTimeout.h line 83
0x40222a44: esp8266::MDNSImplementation::MDNSResponder::stcProbeInformation::clear(bool) at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266mDNS\src/LEAmDNS_Structs.cpp line 1177
0x402202c9: esp8266::MDNSImplementation::MDNSResponder::_resetProbeStatus(bool) at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266mDNS\src/LEAmDNS_Control.cpp line 1155
0x4023e614: operator new(unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3-nox/libsupc++/../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x40220322: esp8266::MDNSImplementation::MDNSResponder::_restart() at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266mDNS\src/LEAmDNS_Control.cpp line 95 (discriminator 1)
0x40220091: esp8266::MDNSImplementation::MDNSResponder::begin(char const*) at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266mDNS\src/LEAmDNS.cpp line 117
0x4021fbc0: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager  const&, std::_Manager_operation) at c:\users\cschwinne\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x4021fca0: std::_Function_handler ::_M_invoke(std::_Any_data const&, WiFiEventStationModeDisconnected const&) at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266mDNS\src/LEAmDNS.cpp line 106
[... cut ...]
0x4020d3bc: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\cschwinne\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x4021b810: setup at C:\Users\cschwinne\git\Arduino\WLED02\wled00/wled00.ino line 508
0x40232908: loop_wrapper() at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/core_esp8266_main.cpp line 122
0x40100ed1: cont_wrapper at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/cont.S line 81
```

raymiec, 2019-03-09:
With lwip2 and neopixelbus 2.3.x i still get the same, tried all versions really.  Soon as i changed to lwip 1 its been fine with neopixelbus 2.3.5.

Cheers man

closed 2019-03-09 by raymiec