wled/WLED #121: WLed Hard-reset?

lightrunner86, 2019-03-08:
Hi, 1.Big thx for your help here last time guys, Great Project 凸(¬◡¬)凸 - i'm still new here, but learning a Lot :-) 

Is there a possibility Hard-resetting the device (wifi-settings, restarting like new) without OTA/USB?? got the 0.8.3 Running on ESP-01, with later no possibility of reaching USB (OTA off for security) Is there a Designated Pin for a hard resetting push button??

BTW:
from a cheap chinease wifi rgb controller i saw a very elegant idea to hard reset:
 if the device notices it got Powered off 5sec,  Poweron>3sec, ......--5 X after each other, it would restart as WifiAP with default value. 
...Sorry I'm more the "Hardware Dude"  in electronics & Circuits ;-)

Aircoookie, 2019-03-08:
Hi! Great that you like WLED!

Here is a custom solution I made some time ago. If power is disconnected within 5 seconds after powerup twice, settings are reset. If you want to use it, just replace the wled06_usermod.ino:
[wled06_usermod.zip](https://github.com/Aircoookie/WLED/files/2947791/wled06_usermod.zip)

With the push button on gpio 0, this is also possible! By default, if you hold the button for 7 seconds, it will open the access point so you can do recovery (if you haven't changed the AP password). You can change this to do a hard reset by changing line 21 in wled09_button.ino:

```cpp
if (buttonReleasedTime - buttonPressedTime > 7000) {initAP();}
//to
if (buttonReleasedTime - buttonPressedTime > 7000) {clearEEPROM(); reset();}
```

Hope one of these solutions will help you :)

lightrunner86, 2019-03-09:
Hey, danke dir vielmals, du bist ja auch aus DE ^^
ich versuche es morgen mal mit einzubinden C++ ist bei mir leider schon ein
paar Jahre her...
aber saucool von dir echt . gibt's eigentlich ein Board wo man sich auch
über die Praktische Anwendung von WLed austauscht? Ich bin selber mehr auf
der Praktischen schiene und versuche die Bude & jede neu "upgecycelte" alte
Lampe gleich mit ESPxxxx upzugraden. ^^
Und Dein Prog füllt das 100% aus! :-)
Ich weis leider nicht was dein/euer Ziel ist, und kann euch im Gegenzug
leider aktuell nur hilfe bei elektronischen Problemen oder Bugreports auf
verschieden ESP anbieten.
Grüße
Terry Baker

Aircoookie, 2019-03-09:
Gerne, freut mich sehr, dass es dir so gut gefällt :)

Klar, ich habe einen Discord-Server, da diskutieren wir alles! Link ist ganz oben in der Readme. 
Bin genauso, dieses Jahr war das erste Mal der komplette Weihnachtsbaum mit WS2811!

Im Moment ist WLED mein Hobby, die Software wird auch für immer kostenlos und Open Source bleiben, aber eventuell verkaufe ich bald mal selbst Lampen, mal schauen ^^

Deine Unterstützung und dass dir mein Projekt gefällt, ist doch schon alles, was ich als Gegenleistung brauche! Dankeschön :)

Viele Grüße
Christian

lightrunner86, 2019-03-09:
Klar helfe ich dir gerne hinterher bei einem Kunden bereits das Vergnügen
hatten die drecks Lampen von Philips einzeln auseinander zu sägen um die
Tiefe zu reduzieren um sie anschließend in einer Trockenbau Decke einbauen
zu können. habe erst gedacht dass du eigentlich mit denen zusammen
arbeitest dann dein Code aus irgendeinem Grund im kaltstart mit der
gleichen Philips Farbe startet. ^^
mein ursprungsgedanke war wie gesagt wegen einer glasbausteinwand die ich
Stein mäßig einzeln adressierbar gestaltet habe um sie später mit neomatrix
anzusprechen und z.b. tetris Demonstrationen durchlaufen zu lassen oder
statusmeldungen zum Stand von Wasser im Keller oder Heizöl anzeigen zu
lassen ohne dass es direkt ein Geräusch gibt, so quasi nach dem Motto und
die Wand leuchtet nur noch bis zur Hälfte hoch Zeit nach dem Öl zu gucken.
Es gibt von ELV bereits einen diamex player dieser unterstützt auch schon
die grafisch einfach bedienbare jinx Oberfläche. wobei dann wieder die
fremd Signale wegfallen da ist deine Idee bisher wirklich mit Abstand die
beste.
Und wenn ich dir bei Lampen oder Elektronik irgendwie helfen kann jederzeit
gerne. :-)
 ich hoffe nur dass deine Idee nicht am Ende als ein billiges china Produkt
endet und am Ende nur von baumarktketten unter dem Namen paulmann oder von
Philips zu so scheiß Preisen vertrieben wird, dass man zuerst im Internet
schaut wird kann man das selber programmieren ;)
Aber ich muss sagen mach das hauptsächlich auch wenn ich merke dass mir das
Licht im Leben einfach total gut tut :-)

lightrunner86, 2019-03-09:
Gibt's hier PN?

lightrunner86, 2019-03-09:
Alles klar super falls ich dir irgendwie mal helfen kann einfach auf die E-Mail schreiben.
:-)

closed 2019-03-09 by lightrunner86

meelow, 2019-03-09:
Hey Terry,
Ich selbst bin auch neu und begeistert von wled. Probier doch mal das
‚discord‘ aus- ist mehr wie ein Chat Programm. Link ist auf der wled github
Seite... bis gleich dort!
Lg
Milo

Tyrian81, 2021-06-01:
which button on esp32 is the reset ? i've got EN and BOOT ?