wled/WLED #122: Feature request: Toggle UDP streaming

valki2, 2019-03-09:
Hi! First off - great software! Works like a charm out-of-the-box and I love the code style!

I am running a massive LED strip as main ceiling light with the possibility to link it to an existing ambilight/hyperion installation. 
Works fine so far - but I would love to externally control the setting "receive  udp realtime" setting as the light should do normal illumination when the tv is off :)

I noticed the bool "realtimeActive" but there seems no API command to switch this one. Did I miss a setting?

If not then I am planning to add it quick and dirty... is this ok and are there any special commit rules?

Aircoookie, 2019-03-09:
Hi! Great that you like WLED!

There is an option in sync settings to "Receive realtime", which does what you want! There is no API call for that one yet, so I'm going to add one, since it makes sense to toggle that more often :)
"realtimeActive" is a state bool which is true if data is received and normal effects are blocked until the timeout. The bool for deactivating realtime is "receiveDirect".

Yes, you can either just open a PR or wait for me to add it (takes two minutes).
By the way, are you running latest master or the 0.8.3 release?

valki2, 2019-03-09:
Ahoi!
Running the latest master (except your changes yesterday with the color wheel fix)...

If it's 2mins, then I will happily wait as it's Saturday what makes beer first :-)

Cheers, valki

valki2, 2019-03-10:
Thanks - works like a charm! :-)

closed 2019-03-16 by Aircoookie

enriquefernandez, 2019-05-04:
Was this added to the API in the end?
I would also love to disable UDP listening with the gui, but also via MQTT or HTTP call.
That way I could ignore the data coming from hyperion whenever I want.

Thanks!

valki2, 2019-05-04:
Hi!
Yes it was added - works here like a charm :)
"Alexa, turn the additional ambilight on" and it switches from normal light mode to ambilight :)
Am Sa., Mai 4, 2019 08:07 schrieb Enrique Fernández : Was this added to the API in the end?
 I would also love to disable UDP listening with the gui, but also via MQTT or HTTP call.
 That way I could ignore the data coming from hyperion whenever I want.
 Thanks!

enriquefernandez, 2019-05-04:
Ok, great!
How do you activate/deactivate UDP realtime with the API?

I don't see anything for that in the [JSON](https://github.com/Aircoookie/WLED/wiki/JSON-API) or [HTTP](https://github.com/Aircoookie/WLED/wiki/HTTP-request-API) api wiki docs..

I see that there's a parameter `live` in the `info` object of the JSON api that tells me whether we are receiving UDP realtime data, but it doesn't look like I can disable it.
There's the `udpn.recv` parameter that I can disable, but it looks like that's not for UDP realtime control, but for the (different?) WLED broadcast functionality.

The only method I know to temporarily disable UDP realtime is to go to the Sync Interfaces GUI menu and uncheck the box. This takes quite a lot of clicks to do, and I'd also love to be able to do it via API (MQTT, JSON or similar). My objective would be to tie this to HomeAssistant and have an action that activates or deactivates UDP realtime control that would in reality activate or disable Hyperion.
Is this possible with the current API?

This project is awesome by the way.

Thanks!

valki2, 2019-05-04:
I do the magic via openhab...

Switch Zusatzambilight "Zusatzambilight" [ "Switchable" ] { http=">[ON:POST:http://192.168.3.9/win&RD=1] >[OFF:POST:http://192.168.3.9/win&RD=0]"}

In the urls you can see the Syntax!

Have fun

Aircoookie, 2019-05-04:
@enriquefernandez Awesome that you like WLED! Oh, just realized, I totally missed adding the `&RD=` HTTP api call to the doc! Sorry!

enriquefernandez, 2019-05-06:
Great!
It looks like this API call is exactly what I need.
Unfortunately, I'm having some issues at the moment with the ESP becoming unresponsive. I've opened a new issue  #163 .

Thanks for the help!