wled/WLED #123: Wemos D1 mini flashing doesn't work

Tig3rch3n, 2019-03-09:
Hey ho,
I tried different methods and different boards but no success u_u
3 different Wemos D1 mini, 2 used ones 1 fresh out of the esd bag

I tried atom/platform.io, uncommented the NodeMCU and changed it to wemos d1 mini in the code build/upload, (platform. Io sais wemos d1 mini 'Succes') it restarts and has the old code running I used before u_u

I tried the http upload method, get the firmware flashing page, upload... 
Nothing changes
Reset button/power cycling doesn't change anything.

I'm at a loss now, no clue what todo.
Any idea what I missed?
Cheers

Aircoookie, 2019-03-09:
Hey,
that is very strange. So, even though you are definitely building and uploading the firmware, the ESPs retain their old code. I would have blamed broken flash memory, but you tried with 3 separate ESPs, so it must be a systematic error. Can you flash any other project but mine fine?

With HTTP upload method, do you mean https://github.com/Aircoookie/ESP8266MinimalHTTPUpdater , together with the 0.8.3 release binary?

Other than that, I'm out of ideas as to why this happens as well.
But don't worry, we'll get WLED working for you somehow!
Did you download the latest master or the 0.8.3 release?

P.S. changing NodeMCU to Wemos D1 is not necessary, they both have the identical ESP-12E module and pinout. You can try once more with the unmodified platform.ini and building for nodemcu and see if that helps.

Tig3rch3n, 2019-03-09:
That's right I tried the 0.8.3 release binary via the minimal updater... It's doesn't do anything after the upload... I had it sitting for 30min power cycled it... No change

For the build with platform.io I used the Latest Git Master

Just tried it with the original ini file with the NodeMCU setting... No change

Said 'NodeMCU Success'
Power cycle... Still my old MQTT client running u_u

Aircoookie, 2019-03-09:
You could try [esptool](https://github.com/espressif/esptool):

```
esptool.py erase_flash
//and then
esptool.py write_flash 0 WLED_0.8.3_ESP8266.bin
```

This should do a complete reset of the entire flash, then flash WLED. If you still have the MQTT client after that, the flash must be broken...

Tig3rch3n, 2019-03-11:
Okay... Just tried...
Flash wiping works

So my guess is somehow the wemos don't like the code

What dev-Board would you recommend for WLED?
I'll just get one, I'm not really happy with the wemos anyway... Got me in trouble with another project u_u

I have some leftover esp-01 (no S) should that work?

Aircoookie, 2019-03-11:
Well, I almost use wemos exclusively, I don't exactly have anything against recommending them, so I don't know why they don't work for you. I will also flash the binary tomorrow to verify that it's not bad in some way. You can also try nodemcu, but that uses the same esp-12e wemos does.

ESP01 does work, but I might have trouble supporting OTA on it in the future due to growing flash usage of WLED. There is a separate binary for it.

Another alternative is an ESP32 board, which also works quite nicely with wled but will cost about double an esp8266.

closed 2019-10-25 by Aircoookie