wled/WLED #124: Audio Reative LED Strip

sjude68, 2019-03-09:
I have seen a project for Sound Reactive LED Strip from Natural-Nerd
https://github.com/hansjny/Natural-Nerd/tree/master/SoundReactive2

He uses a Mic connected to NodeMCU for the Master module and Slave module connected to the LED Strip.  Multiple slave modules can be connected to the same module.

Can I use the WLED as a slave connected to the Master module for audio reactive.

raymiec, 2019-03-10:
No, not out the box, the slave sketch is doing the bulk of the work, the master is just sending the raw audio i think.  You could modify the slave sketch to send the computed rgb data onto a wled device. No idea about lag and stuff, would have to try it out i guess.

Theres also an option instead of using an esp as master module with mic you could probably use the android app im doing to send the audio data over. I myself shall try this later tonight and report back if i get time.

sjude68, 2019-03-17:
I installed ledfx on Windows 10 using Anaconda. Works great with wled.

raymiec, 2019-03-17:
@sjude68 OP wants to use a mic attached to ESP, so he doesn't have to use a separate device.

debsahu, 2019-03-17:
The ADC sampling rate (1000Hz) is too low for ESP8266 to get any useful information above 8000Hz sound. See https://github.com/G6EJD/ESP32-8266-Audio-Spectrum-Display or one could use external hardware (MSGEQ7) with ESP8266/32 like described in my video https://www.youtube.com/watch?v=8YaeUYZ_Ex8

raymiec, 2019-03-17:
There is some sketches that work pretty well, the only way to get it to play properly with WLED is by sending the computed colours, I've had some success on android, over wifi it's not great though, Asus tinker board on ethernet is pretty good. I'm going to try over bluetooth with esp32 and see how that fairs.

SteamRaider, 2019-03-26:
@raymiec 
Where can I get this Android App?
Because I am currently looking for a similar Audio Solution for WLED.
Thank you

mike2nl, 2019-04-24:
Hey all,
i just jump in because we do a lot of things with the esp8266 on Tasmota. 
And today i got the info about WLED. Really a very cool and amazing project.
- The only thing missing i think is that sound(mic) controlled part.
2 possibillities:
speed controlled by sound
color controlled bu sound

@Aircoookie 
You have some ideas to set it up? Need help (stuff) with hardware to test?

Anothermetrial, 2019-09-25:
@Aircookie would you pick up this audio reactive led light subject? It would be really awesome. I am searching for a good solution for 2 years already. A mic connected to esp32 should be ...a killing feature.
Thanks for considering this.

miathedev, 2019-11-23:
I built something similar in the past.

I made a FHT(Fast Hartley Transform) on Audio Input and i devided the spectrum into three parts.

a-b =>bass
b-c =>midtones
c-d =>"tweets" - high tones

I took the average of those three "bands" and mapped them to respective Red, Green, Blue.
I used the http://wiki.openmusiclabs.com/wiki/ArduinoFHT library.

Keep in mind if you want to meassure audio from an audio jack you have to use following circuit:

OPAmp 1:1 amplifier => voltage devider (AIn -> ADC -> GND). 

I prefer using the opamp because you dont have to match the impedance in this case because I+ = I- =0 and ΔU+- =0.

IF YOU DONT do it like this, you will loose a lot of bass and mid tones because you dont match the impedance. Dooes someone know a good library for a DFT/FFT/HFT on ESPXXXX?

Of course you can do it like mentioned earlier above in this issue measuring voltage peaks from a microphone. But you will get not really satisfying results.

Aircoookie, 2020-03-12:
I believe sound reactivity on chip would be a really cool feature to have!
Right now it's a bit of a chicken and egg problem. There are no effects capable of reacting to sound, so  a sound reactivity driver (either using external chips like MSGEQ7 or via software, like FHT) needs to be added first. Then again, what is the point of that if there are no effects to take advantage of it?

I would really love for WLED to have build in sound to visual capabilities. If somebody makes a usermod I will surely consider enhancing a few effects to support sound input! (or maybe i can get to it myself in a few months haha)

sansillusion, 2020-03-12:
I think that a good beat detection could be used to cycle trough effects
and color palettes. ie: change effects every 8 beat and cycle color
palettes every beat.

Just a thought.

RAMBORG, 2020-03-13:
Hello that's the link what I mean.
https://youtu.be/mK9VdvVI_4s
Video is 3 years old, in his playlist are newer clips. ( The code is there too) cinelight is great and he create always projects for free. Base on a Arduino mega

Next example is this one https://youtu.be/8bFgW-mpwdg
He has one demo version and one Pro version (cost something) base on a esp8266

CMaxim77, 2020-03-13:
Thanks for pointing me in the right direction m8, I’ll keep a eye on this post!

luminxa, 2020-04-06:
following for updates :)

RichardA1, 2020-05-11:
@Aircoookie an easy way to start with how the sound effects existing animations would be to have the sound level effect the Brightness, Color, Pallet, Effect Speed, or Effect Intensity. Maybe even have a setting so you can check off with one of those you want the audio signal to control. Heck, if you just have it go by an analog signal you could connect things like potentiometers, temperature sensors, or a light sensor.

crbello89, 2020-05-13:
I have a weird setup, I am using up2stream wifi audio boards to make a poor man's Sonos system.. but I stream spotify to my home assistant.. which in turn gives me audio to send to said up2stream boards, but simultaneously LEDFx is taking the pulse audio to react my led strips. Problem I have now is, my led strip is reacting faster than the audio from the slowest device.. so how do I slow down the LED strip?

raresserban, 2020-05-13:
> I have a weird setup, I am using up2stream wifi audio boards to make a poor man's Sonos system.. but I stream spotify to my home assistant.. which in turn gives me audio to send to said up2stream [...]

Wow, that sounds awesome! Might be off topic, but could you share more details about this setup you have?I also run Home Assistant and am interested in a multi room DIY audio system.

closed 2020-05-26 by sjude68

ChristianWaechter, 2020-06-03:
Hello everybody,

I would like to join this discussion (and maybe development ;) ), as I also want to have my LEDs sound sensitive controlled with a Wemos D1 Mini and without any further devices. Also I'm familiar with hardware and firmware development. My setup is somehow overkill, but it should be possible. I have 500 WS2812 LEDs build in the ceiling of my combined kitchen/dinning/living room that should be used for ambient light when watching movies. But also, it would be nice to have the light react to the sounds from the movie, at least to the three frequency ranges low (shots, explosions), mid (voices) and high (screams, shouts...). Or use a MSGEQ7 to get at least 7 frequency ranges. If you havent seen it yet, this is what I found on youtube and should be possible with these ranges: https://youtu.be/yhxa9k6pvNc

As far as I read through here, the internal ADC is limited because of its samling rate to frequencys of 5000 Hz. This might still work, if you define the frequency ranges correctly:
Low: 0 - 300 Hz
Mid: 300 - 4000 Hz
High: Everything above 4000 Hz

There actually might be a difference between the ESP32 and ESP8266 concerning the sample rate.
ESP32 (ADC1) up to 105 k samples per second (https://www.esp32.com/viewtopic.php?f=19&t=1215)
ESP8266: up to 40 k samples per second (http://g6ejd.dynu.com/index.php/micro-controllers/esp32-8266-spectrum-analyser/)

But IMHO it would be better to use of a external ADC that is connected via I2C (or even SPI). For example the ADC121C021 can do up to 188k samples per second, which should be enough.

I also do not think, that the ADC is the only weak point when using the ESP for sound visualisation. I do not think that anyone would like to use a big stage microphone for that ;) So a small microphone like the Ekulit EMY-63M/P has a given frequency range starting at only 300 Hz but at least up to 20 kHz. The Ekulit EMY-9765P starts at 100 Hz, but only goes up to 10 kHz.

For the sound sensitive pattern or effect, how about a "Cinema"-pattern that reacts to the measured frequencys?

PS: The image was taken in a very early state, the LEDs are hidden in the ceiling right now ;)
[image]

Dreamoffice, 2020-06-30:
are there any news about the music reactiv wled feature?

Bart-van-Ingen, 2020-07-14:
there is a fork of wled that allows for sound reactive leds: [https://github.com/atuline/WLED/wiki](https://github.com/atuline/WLED/). May be worth checking out!

DJprasenjit, 2022-10-21:
i managed to do it. wire less VU meter using WLED programme. and its working great . i will share video here. its so simple setup.
its only works on windows laptop or desktop pc. its shows the display color background or the sound effect vu meter. both are looks great.