wled/WLED #125: More docs on usage

splitbrain, 2019-03-09:
I'm just starting with this fantastic software and having trouble understanding the interface. I think some more wiki pages on the following topics would help:

* mobile vs. classic interface (why not show the mobile interface on Desktop, too - it seems more powerful?)
* basic UI elements (I understand the power and settings icons, but what does the moon do?)
* FX Mode vs FastLED Palette (Are they applied at the same time? How do they combine?)
* List of FX Modes and FastLED Modes (with a short description and the number to use in the settings and macros)

I'm happy to help. I could start with the last point, but need some help with the rest.

Aircoookie, 2019-03-09:
Hi! I totally agree with you, we really need more docs!

I'm going to answer your questions very briefly, but I will soon add some more docs.
- I also mostly use the mobile interface on desktop (you can switch in UI settings). Next major version will add segment support, and at that point I will probably remove the classic UI altogether. Basically classic was the initial UI. A year ago, an actual webdesigner made the mobile UI for WLED, which i absolutely prefer.
- The moon icon activates the "nightlight" mode. Basically the light will gradually get darker over the course of an hour (target brightness and duration configurable in settings)
- FX mode is the main FX mode and defines the pattern or animation that's running on the LEDs. FastLED palette defines the colors. Almost all effects support palettes (the default "Solid" being a notable exception) On any setting but default it overrides the top color wheel.
- I wanted to create a list of all FX with a short description since forever now. Your help would be greatly appreciated :)

Let me know if you have more questions or suggestions!

splitbrain, 2019-03-09:
Thanks for your answers!

I started by going through the the list of effects and trying them (after having set a primary and a secondary color). I left some out where I couldn't really figure out what exactly is going on. This list needs some cleanup by someone who knows what's going on a little better (and maybe is a native speaker). Some intro on how primary and secondary color are picked and how the FastLED palettes influence that might help, too.

I can't edit the wiki directly, but this should be a start to be copied:

| Effect ID | Name | Description |
| ---: | --- | --- |
| 0 | Solid | primary color fixed for all LEDs |
| 1 | Blink | blinks the primary color on and off |
| 2 | Breathe | primary color oscillates between dark and bright |
| 3 | Wipe | Switches between primary and secondary, switching LEDs one by one, start to end |
| 4 | Wipe Random | Same as 3, but uses random colors |
| 5 | Random Colors | Switches all LEDs to the same random color |
| 6 | Sweep | Switches between primary and secondary, switching LEDs one by one, start to end to start |
| 7 | Dynamic | Sets each LED to a random color |
| 8 | Colorloop | Fades all LEDs at the same time to a random color |
| 9 | Rainbow | Displays the colors of the rainbow along the whole strip |
| 10 | Scan | Sets the strip to the secondary color and a single primary colored light wanders between start to end to start |
| 11 | Dual Scan | Same as 10 but uses two lights starting at both ends |
| 12 | Fade | Fades between primary and secondary color |
| 13 | Chase | Primary colored LEDs move across a secondary colored background (ratio 1:2) |
| 14 | Chase Rainbow | Same as 13 but the primary color is replaced by the colors of the rainbow |
| 15 | Running |  |
| 16 | Saw | |
| 17 | Twinkle | Random LEDs light up in the primary color with secondary as background, resets after 50% :question: are lit primary |
| 18 | Dissolve | |
| 19 | Dissolve Rnd | |
| 20 | Sparkle | single random LEDs light up in the primary color for a short time, secondary is background |
| 21 | Dark Sparkle | All LEDs are lit in the primary color, single random LEDs turn off :question: for a short time |
| 22 | Sparkle+ | All LEDs are lit in the primary color, multiple random LEDs turn off :question: for a short time |
| 23 | Strobe | All LEDs are lit in the secondary color, all LEDs flash in a single short burst in primary color |
| 24 | Strobe Rainbow | |
| 25 | Mega Strobe | All LEDs are lit in the secondary color, all LEDs flash in several short bursts in primary color |
| 26 | Blink Rainbow | |
| 27 | Android | Similar to 10 , but about a third of the strip uses the primary color |
| 28 | Chase | 2 LEDs in primary color move from start to end, turning the visited LEDs to secondary color |
| 29 | Chase Random | Like 28 but leaves a random color behind |
| 30 | Chase Rainbow | Like 28 but leaves the colors of the rainbow behind |
| 31 | Chase Flash | 2 LEDs flash in secondary color while the rest is lit in primary. The flashing LEDs wander from start to end  |
| 32 | Chase Flash Rnd | Like 31, but the 2 LEDs flash in random colors an leave a random color behind (restarts after the whole strip has one color) |
| 33 | Rainbow Runner | Like 28, but the 2 LED light up in the colors of the rainbow and leave the primary color behind |
| 34 | Colorful | The strip lights up in a color full pattern that repeats over the strip, the pattern changes |
| 35 | Traffic Light | The secondary color is used as background while other LEDs lights up in red, red+yellow, green or yellow |
| 36 | Sweep Random | Like 6 but uses random colors |
| 37 | Running 2 | pattern of two LEDs primary and two LEDs secondary moves along the strip |
| 38 | Red & Blue | Like 37 but always uses red and blue |
| 39 | Stream | |
| 40 | Scanner |  |
| 41 | Lighthouse | |
| 42 | Fireworks | The strip is turned off, random parts light up and move then fade again |
| 43 | Rain | The strip is turned off, random parts light up and fade |
| 44 | Merry Christmas | Like 37 but always uses red and green |
| 45 | Fire Flicker |  |
| 46 | Gradient | Moves a saturation gradient of the primary color across the whole strip |
| 47 | Loading | |
| 48 | In Out | Similar to 3 but starts from the center of the strip and moves to both ends, then back to center |
| 49 | In In | Like 48, but always starts from the ends |
| 50 | Out Out |  |
| 51 | Out In | |
| 52 | Circus | |
| 53 | Halloween | Like 37 but always uses orange and secondary color |
| 54 | Tri Chase |  |
| 55 | Tri Wipe | Like 3 but turns LEDs off as "third color" |
| 56 | Tri Fade | Fades the whole strip from primary color to secondary color to off |
| 57 | Lightning |  |
| 58 | ICU | |
| 59 | Multi Comet | |
| 60 | Dual Scanner |  |
| 61 | Stream 2 | |
| 62 | Oscillate | The strip is turned off, areas of primary and secondary colors move between opposite ends, combining colors where they touch. |
| 63 | Pride 2015 | Each LED cycles through the colors of the rainbow |
| 64 | Juggle | |
| 65 | Palette | |
| 66 | Fire 2012 | Simulates flickering fire in red and yellow, startin at the start of the strip |
| 67 | Colorwaves | |
| 68 | BPM | |
| 69 | Fill Noise | |
| 70 | Noise 1 | |
| 71 | Noise 2 | |
| 72 | Noise 3 | |
| 73 | Noise 4 | |
| 74 | Colortwinkle | LEDs light up randomly in random colors and fade off again |
| 75 | Lake |  |
| 76 | Meteor | The primary color moves over a background of secondary, pulling a "tail" of brighter, desaturated primary color |
| 77 | Smooth Meteor | Like 76, but smoother :question: |
| 78 | Railway | Shows primary and secondary color on alternating LEDs. All LEDs fade to their opposite color and back again.
| 79 | Ripple | The strip shows secondary color as background, the primary moves along the strip by fading in |

Aircoookie, 2019-03-09:
Thank you very much! I will complete the table and add it to the wiki soon!

raymiec, 2019-03-10:
I'll fire this into a database and put a form up to edit / export, will be good to have the descriptions in the app.

splitbrain, 2019-03-10:
BTW it's a bit confusing that some of those effects have the same name despite being different ones.

roaoro, 2019-05-02:
Can anyone point me in the direction of how to use the timer for this cool project? I simply want the lights to turn off in 30 minutes, I found the section for timed light and put in 30 minutes, and assume hitting the night button on top would make it turn off in 30 minutes but doesn't seem to work that way.

brentbrooks70, 2019-05-03:
@roaoro Did you change the **Default Target brightness** in the **Timed Light** section of the **LED setup** page? If you want the light to completely turn off after 30 minutes you should set the target to 0 and uncheck **Fade down**

roaoro, 2019-05-04:
I was able to get it to work after a few adjustments, not sure what I did or didn't do but it is working now. Thank you very much. Awesome project!

closed 2019-10-25 by Aircoookie