wled/WLED #127: TP-Link HS210 causes wled to turn on

neilyboy, 2019-03-11:
Oddball issue im having.  I have a set of TP-Links HS210 switches in my shed.  I also have my huge setup of~850+ leds running beautifully off of your code. One issue im having however is when I turn on my lights to the shed via the kasa android app (provided by tp-link) about two seconds after the main lights turn on,  my leds string turns on.  I see a bright white flash on my controller and the wled strip lights up. 

Is there a way to modify something to stop this action from happening.  It's not a huge deal other than I then have to log into the web gui and kill the leds manually (as they do not turn off with the kasa app). I figure this is just an odd coincidence.  No worries if it can not easily be fixed.  I just found it interesting.  Thanks again for your brilliant code!

Neil

Mariu86, 2019-03-11:
You have a option in settings to power up the leds after power up / after wled connect to WiFi or to be off after the power on .

Aircoookie, 2019-03-11:
Mariu is probably right, the HS210 control packet seems to make WLED reset instead of turning it on, the "white flash" is common if this happens. It would be interesting to know why the reset happens, probably the HS210 or app are sending a flood of discovery packets or something. You can try whether the issue persists after turning off Alexa in Sync Settings and rebooting.

neilyboy, 2019-03-11:
You guys rock!  Il give those settings a try.  I know the hs210 kit is a 3-way switch configuration.  So one switch talks to the other switch over wi-fi . I have to agree that the flash of white light behavior is the controller resetting.  I will pop back into the config and turn lights off after reboot.  Thanks again
Neil

neilyboy, 2019-03-11:
Is there a discord or forum elsewhere to discuss goofy little issues like this or is creating an issue ok in this scenario? Thanks again for the quick help!
Neil

Mariu86, 2019-03-11:
You will find the discord at the end of the wiki page .

closed 2019-03-12 by neilyboy