wled/WLED #128: Add Relay support for Sonoff like Devices

aenertia, 2019-03-12:
Hi there; I am happily using WLED on sonoff basics; but would like to see addition of support for GPIO12 (the relay) built in. 

This would be connected to the 5V AC/DC High Amp PSU for the LED driver so in effect provides a 'real' on/off switch for the LED strips (I am using SK6812 connected on GPIO4).

Currently I am driving this via tapping into the on-board 5v source but this is only useful for strips of around 10-30 LED's for anything longer an external driver is needed.

Aircoookie, 2019-03-12:
Seeing that high power PSUs can have an always-on fan and a non-negligible standby current, as well as the fact that each addressable LED uses about 1mA when "off", I think this is a useful feature, so I will add it in time for the 0.8.4 release! GPIO12 seems like a good option for default pin as well.

Aircoookie, 2019-03-16:
This is now supported and should work without any code changes on GPIO12! If one wants to use a different pin or invert the logic level, that can be set in `NpbWrapper.h`

aenertia, 2019-03-16:
Did you add button support and state remember bits? There is some nice denounce/state stuff you can probably lift from tasmota

Aircoookie, 2019-03-16:
Yes, button was already supported, but I extended it to also optionally support double click. Button pin is gpio0, i don't know if that works for the Sonoff as well, you can also change it in NpbWrapper.h. Right now, only Low-is-pressed is supported (pullup, normally open pushbutton), but i want to add support for pulldown soon!

sjude68, 2019-03-19:
> This is now supported and should work without any code changes on GPIO12! If one wants to use a different pin or invert the logic level, that can be set in `NpbWrapper.h`

The GPIO12 is not toggling with LED ON or OFF.  It stays ON. When I change RLYMDE to 0, the GPIO12 stays OFF.

Aircoookie, 2019-03-19:
That's peculiar, I've just tried this again with the latest master and it (still) works fine. AFAIK the sonoff basic has a green LED on gpio14. Can you try if toggling that works?

Aircoookie, 2019-03-19:
Ah ok, great :D yes, the version before had it not working. my bad!

dhami220, 2019-10-19:
I have not been able to make sonoff basic work by uploading the binary. Do I need to compile the code with specific settings for the sonoff? I have sonoff basic as well as SV. I prefer to use SV to keep it all low voltage. Is that is feasible?

raoulteeuwen, 2019-10-19:
@aenertia could you please provide some details on how to combine a Sonoff with WLED? I'm interested since i found sometimes my 1st 6 LEDs do strange things after a power reset (after which i manually disconnect power and connect it again to make the 1st LEDs go off), and since i read in this thread adding a Sonoff could save power. I have flashed my Sonoff basic with ESPhome. I'm using Home Assistant as a control of all the stuff.

aenertia, 2019-10-27:
Sonoff basic can be a bit temperamental. 

My 'recipie' is this :

Desolder fresh sonoff basic SPI module - replace with 4MB SPI
Solder on LED harness wire to either side of the D5 Diode (5V rail) - you now have a 5v power able Sonoff basic. Data line goes to GPIO3 pin (RX).

At this point you can flash WLED - I generally build with d1_mini as target but you can use esp01_1m as well. Recently I've added an additional board definition for esp01 but with 4M flash-layout. 

Sometimes it won't take - generally I will start with latest sonoff.bin from tasmota release and check wifi is working etc and I don't have a dud module. Then do an erase_flash with esptool then upload the wled binary. 

Sometimes i've noticed that PIO doesn't do the spiffs image so it looks like it works but doesn't until you do a : pio run --target uploadfs

You then run your 5v driver source wired up to the relay as normal.

aenertia, 2019-10-27:
Also you can safely check it's working by just connecting the 5v harness connector to a seperately powered 5v strip - as the 5v and gnd go to the 5v rail inside the sonoff basic and it has a regulator you effectively have a LED controller that leeches power from the strip at this point.

Adding the 5v AC/DC driver in-line with the relay can mean that on quick physical resets there is enough residual current in the driver source to confuse the esp8266 - it seems a couple of seconds is needed between full cycles if you are powering the sonoff in-line off the ac/dc

DotNetDann, 2019-10-31:
> If one wants to use a different pin or invert the logic level, that can be set in `NpbWrapper.h`

Working well however had to build the bin. Would love the invert setting via the UI if posible

CaptInsano, 2019-12-07:
I came here looking for this option.

I have a few tasmota sonoff S20 sockets which control both the power to NodeMCUs running WLED & 5V 10A LED power supplies. Is it possible to install WLED directly onto the S20 socket and use one of the pins from the esp8266 in the S20 (GPIO1 & GPIO3) as a data pin to drive the LEDs?

If this is possible, I have 3 questions:
Can I just use the provided binary from the releases page?
Can I set it that when I click the power icon in the WLED UI that it also toggles the S20 relay?
Will the physical button on the S20 still toggle the relay?

Thanks a million for any advice!

raddi, 2020-01-07:
@CaptInsano Thats exact the same setup as I want to use. There are already .bin files for the LED-PIN on GPIO 3 (RX) but I tried 0.8.5, 0.8.6 and 0.9.0b1 versions but all seem to have a problem with Sonoff like devices, I connect to the WLED-AP and change the Wifi Network, but after a reboot, all settings are lost. I tried Sonoff S26, Sonoff S20, Sonoff Basic but all seem to have the same problem, they won't remember the settings after a reboot. 
I tried:
WLED_0.9.0-b1_ESP8266_ledpin3.bin
WLED_0.9.0-b1_ESP8266_1M_full.bin
WLED_0.9.0-b1_ESP8266.bin
WLED_0.8.6_ESP8266.bin

and some others but the problem still exists. As for the Sonoff S26: there is a GPIO2 pin directly on the board, so I tried to use the standard .bin files which work very nicely with my wemos but do not work with Sonoff devices. Has anybody got it working on a Sonoff? 
By the way, the GUI shows "WLED error 2"

Aircoookie, 2020-01-07:
@raddi "WLED error 2" means that the settings could not be written to flash. The reason for that is that the Sonoff Basic (and probably S20 and others) only have 512k flash chips. You can try the 512k binary, it should save the settings just fine. I could send a custom build compiled for gpio3 later.

raddi, 2020-01-09:
@Aircoookie Thank you for your reply, i built my own firmware.bin and now it is working like a charm, thank you so much!

jkobin, 2020-01-15:
I'm using an ESP8266 with GPIO12 to drive a relay to control a power supply and it works great!  The challenge I have is when the ESP8266 is power cycled, GPIO12 is set high even though WLED is "Off".  I see I can invert logic in NpbWrapper.h but that doesn't solve the issue.  How do I get the ESP8266 to reboot with GPIO12 low?

sailhobie, 2020-02-10:
> @raddi "WLED error 2" means that the settings could not be written to flash. The reason for that is that the Sonoff Basic (and probably S20 and others) only have 512k flash chips. You can try the [...]

@Aircoookie 
Hi, I'd be very interested in the 512K binary with GPIO 3. I took a look at the wrapper and found where the setting for GPIO is but wasn't sure how to compile for 512K. Happy to learn how though.

Thanks

closed 2020-06-17 by stale[bot]

raddi, 2020-11-17:
@Aircoookie with 0.10.2 there is no 512k environment/buildflag in platform.io anymore. Using the 512k.bin file you're providing results in the Sonoff device not booting. How do i get WLED 0.10.2 running on Sonoff devices? :)

Aircoookie, 2020-11-17:
@raddi there still was an environment at the point 0.10.2 was released (although even then I had to remove a few extra effects to get it to fit) https://github.com/Aircoookie/WLED/tree/v0.10.2

Are you getting an exception printed to the Serial monitor? That binary should boot fine...