wled/WLED #129: Support Homie MQTT Convention

nichtmax, 2019-03-13:
First things first: your software is the holy grail. I use it with Hyperion and Ledfx and MQTT and I don’t know what I’d do without it. Thanks for your work! 

So I use MQTT with OpenHAB and that’s always a pain in the ass because every channel needs to be setup manually, HSB to RGB-Hex and vice versa calculations need to be done, I need a huge rule set and so on...

Have you heard of [Homie](https://homieiot.github.io/specification/)? I haven’t tested it due to a lack of supporting devices but the promise they make is automatic discovery, configuration and usage of devices and services. OpenHAB and HomeAssistant support Homie, any thoughts on this? I can try to help but I’m really bad in C 😅

Aircoookie, 2019-03-14:
Hi! It makes me very happy to hear that you love WLED :)

I've took a quick look at Homie and it's definitely interesting and worth adding!
Right now I'm running into slight memory issues though, so I would like to put adding more interfaces on hold until that is resolved by further optimizations. That said, it could still be added as an optional interface! I'll keep Homie in mind, it has a lot of potential for "cross-system" IoT and I hope I can add support for it someday!

There is another request for OpenHAB integration here, by the way: #35
A user managed to get all functionality working with a quick NodeRED script, apparently.
Just in case you'd be interested in that!

Aircoookie, 2020-03-29:
Similar to the structure proposed in #207 :)

closed 2020-03-29 by Aircoookie