wled/WLED #130: SPIFFS issue

Adrian8x, 2019-03-14:
I have the following issue when I activate SPIFFS. I installed and uninstalled the ESPasyncwebserver library but without success.

`

Build options changed, rebuilding all
C:\Users\rpup\AppData\Local\Temp\arduino_modified_sketch_268563\wled04_file.ino: In function 'String getContentType(String)':

wled04_file:55: error: base operand of '->' has non-pointer type 'AsyncWebServer'

   if(server->hasArg("download")) return "application/octet-stream";

            ^

C:\Users\rpup\AppData\Local\Temp\arduino_modified_sketch_268563\wled04_file.ino: In function 'bool handleFileRead(String)':

wled04_file:80: error: base operand of '->' has non-pointer type 'AsyncWebServer'

     size_t sent = server->streamFile(file, contentType);

                         ^

C:\Users\rpup\AppData\Local\Temp\arduino_modified_sketch_268563\wled04_file.ino: In function 'void handleFileUpload()':

wled04_file:88: error: base operand of '->' has non-pointer type 'AsyncWebServer'

   if(server->uri() != "/edit") return;

            ^

wled04_file:89: error: 'HTTPUpload' was not declared in this scope

   HTTPUpload& upload = server->upload();

   ^

wled04_file:89: error: 'upload' was not declared in this scope

   HTTPUpload& upload = server->upload();

               ^

wled04_file:89: error: base operand of '->' has non-pointer type 'AsyncWebServer'

   HTTPUpload& upload = server->upload();

                              ^

wled04_file:90: error: 'UPLOAD_FILE_START' was not declared in this scope

   if(upload.status == UPLOAD_FILE_START){

                       ^

wled04_file:96: error: 'UPLOAD_FILE_WRITE' was not declared in this scope

   } else if(upload.status == UPLOAD_FILE_WRITE){

                              ^

wled04_file:100: error: 'UPLOAD_FILE_END' was not declared in this scope

   } else if(upload.status == UPLOAD_FILE_END){

                              ^

C:\Users\rpup\AppData\Local\Temp\arduino_modified_sketch_268563\wled04_file.ino: In function 'void handleFileDelete()':

wled04_file:108: error: base operand of '->' has non-pointer type 'AsyncWebServer'

   if(server->args() == 0) return server->send(500, "text/plain", "BAD ARGS");
[... 304 lines cut ...]

                              ^

C:\Program Files (x86)\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:410:30: note:   no known conversion for argument 3 from 'void()' to 'ArRequestHandlerFunction {aka std::function<void(AsyncWebServerRequest*)>}'

C:\Program Files (x86)\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:411:30: note: AsyncCallbackWebHandler& AsyncWebServer::on(const char*, WebRequestMethodComposite, ArRequestHandlerFunction, ArUploadHandlerFunction)

     AsyncCallbackWebHandler& on(const char* uri, WebRequestMethodComposite method, ArRequestHandlerFunction onRequest, ArUploadHandlerFunction onUpload);

                              ^

C:\Program Files (x86)\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:411:30: note:   candidate expects 4 arguments, 3 provided

C:\Program Files (x86)\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:412:30: note: AsyncCallbackWebHandler& AsyncWebServer::on(const char*, WebRequestMethodComposite, ArRequestHandlerFunction, ArUploadHandlerFunction, ArBodyHandlerFunction)

     AsyncCallbackWebHandler& on(const char* uri, WebRequestMethodComposite method, ArRequestHandlerFunction onRequest, ArUploadHandlerFunction onUpload, ArBodyHandlerFunction onBody);

                              ^

C:\Program Files (x86)\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:412:30: note:   candidate expects 5 arguments, 3 provided

exit status 1
base operand of '->' has non-pointer type 'AsyncWebServer'

`

Aircoookie, 2019-03-14:
Hi! Do you want to add a custom functionality that depends on SPIFFS? I haven't been using it for more than 2 years, and EspAsyncWebserver requires different code to work with spiffs. Once i implement a better preset/custom palette/playlist feature, I will need SPIFFS again. Right now it doesn't work. If you have a good reason for it, i might re-add support sooner :)

Adrian8x, 2019-03-14:
Hi,

It's easier for me to develop/debug the HTML if I upload html's to FS and not to convert every time from gzip to array.

Aircoookie, 2019-03-14:
That makes sense! Since I soon will need the SPIFFS again anyways, I'll restore the functionality very soon!

Adrian8x, 2019-03-14:
That's great! I'm home designer and I create personalized decors sometimes with lights. I don't have so much time and knowledge to develop too deep firmware's. Lights are a little parts of what I do. I'm focused more at forms. I just want to create interface(logo, themes...) more personalized to users, to change their architectural lights so I need to change easy the HTML and to upload on ESP via HTML interface with a single button.

Thank you so much! You made a great library!

Aircoookie, 2019-03-16:
SPIFFS is now supported again in latest master! You need to toggle the 3 headers in wled00.ino to enable it. After that, you can use the `/edit` page to upload or change SPIFFS content. Serving arbitrary files (including gzipped) from SPIFFS (and index.htm/index_mobile.htm) is also supported if you toggle on `#define WLED_ENABLE_FS_SERVING` :)

Adrian8x, 2019-03-18:
I tested your last repository with SPIFFS and it works perfect.

Thank you a lot for your help and for this peace of firmware art!

closed 2019-03-18 by Adrian8x