wled/WLED #131: "Sync-Lights"-state not saved

YeonV, 2019-03-14:
Hello :)

First of all:
Thank you sooo much for your AWESOME WORK!!!

2 Things:
1. After reboot the "Sync-Lights"-button defaults back to "on". Is it possible to save the "off"-state upon Power-loss and power-on? Or could i change the default value to off in the arduino code (preferably in the usermod-section)
2. Is it possible to extend the xml-response (preferably in the usermod-section) with the following values? WiFi.localIP(), ledCount, mqttDeviceTopic, mqttGroupTopic

A small code example would be soo nice :)

P.S.:

I included the wled in "Home Assistant" with all effect-, color- and brightness-change functionality.
Everything controlable via Google Home.
With a small component its now also possible to animate over multiple strips via udp within homeassistant itself.
If its of any use for you, let me know...

Aircoookie, 2019-03-14:
Hi!

Thank you, I'm very glad that you like it :)

You can set the boot default in the web settings -> Sync Settings -> WLED broadcast! I recommend just unchecking the "Send notification on direct change" checkbox, that way you can easily temporarily re-enable it with the "Sync Lights" button.

Would be awesome if you shared your config! (if it's no extra trouble for you). I bet there are a lot of HA users who would benefit from it!

Mariu86, 2019-03-15:
I want to start using home assistant and anything that can help me it's very good. Thanks

YeonV, 2019-03-15:
> You can set the boot default in the web settings -> Sync Settings -> WLED broadcast! I recommend just unchecking the "Send notification on direct change" checkbox, that way you can easily temporarily re-enable it with the "Sync Lights" button.

Perfect mate!!
That was what i was looking for!!

> Would be awesome if you shared your config! (if it's no extra trouble for you). I bet there are a lot of HA users who would benefit from it!

I dont know if its the right place to share some of my things, but here you go... 
(it got a bit longer than I tought, sorry for that)

WLED in HA:
[image]

Here is the HA-configuration using the official mqtt light:
```
light:
  - platform: mqtt
    name: "LED-Strip_1"
    state_topic: "blade/ledx/strip1/c"
    command_topic: "blade/ledx/strip1"
    rgb_state_topic: "blade/ledx/strip1/c"
    rgb_command_topic: "blade/ledx/strip1/col"
    brightness_command_topic: "blade/ledx/strip1"
    brightness_state_topic: "blade/ledx/strip1/g"
    brightness_value_template: "{{ value }}"
    rgb_command_template: "{{ '#%02x%02x%02x' | format(red, green, blue)}}"
    rgb_value_template: "{{ value[1:3] | int(base=16) }},{{ value[3:5] | int(base=16) }},{{ value[5:7]| int(base=16) }}"
    qos: 0
    optimistic: true
    on_command_type: "first"
    payload_on: "ON"
    payload_off: "OFF"
    effect_list: !include lighteffects.yaml
    effect_command_topic: "blade/ledx/strip1/api"
    effect_state_topic: "blade/ledx/strip1/api"
    effect_value_template: "{{ value }}"
```
With this setup you got your WLED in HA:
[image]

With working effects:
lighteffects.yaml attached as txt (please care with indentations...)
[lighteffects.yaml.txt](https://github.com/Aircoookie/WLED/files/2972861/lighteffects.yaml.txt)
[image]

Since its the official mqtt light, its pretty straight forward exposing it to Google Home.

WLED configuration (for this example):
MQTT-Device-Topic:  `blade/ledx/strip1`
MQTT-Group-Topic:  `blade/ledx`
You might want to turn off "sync-lights" to properly control each strip via voice:
"Ok Google, set `LED-Strip_1` to blue"
Ofc you would use a proper name.

---

As additional information for you, here my changes to the arduino code:
(NOT NEEDED for the normal usecases)

1. Added simple username & password into MQTT
2. Extended the XML-response-function to return JSON instead of XML only if requested from within the MQTT-context
3. Added condition to only publish on `/c` if length of the value is >5 to filter

now my mqtt output is:
[image]

with this you can setup a sensor in HA like:
[image]

I saw too late that you already got a JSON response. Im gonna change it to that instead of my custom response.

Here is the configuration for that sensor:
```
sensor:
  - platform: mqtt
    name: "LED-Strip_1"
    state_topic: "blade/ledx/strip1/v"
    unit_of_measurement: 'LEDs'
    value_template: "{{ value_json.leds }}"
    json_attributes:
      - leds
      - ip
      - topic
      - color
```
After adding some, they show up in HA like:
[image]
---
Here is my current HA:
[image]
---
I think this comment is already exploding, so lets not talk about the python-udp thing here ^^
In short: it can produce "color-swipe" through multiple strips and can be triggered like a "normal light": Hey Google, set `Transition` to blue. 
Or set it as an "Ambient".

closed 2019-03-15 by YeonV

Mariu86, 2019-03-16:
Thanks , when I will finish to install all I hope I will understand all the settings :D I am a beginner. Thanks guys !

aidbish, 2019-03-19:
@YeonV 
Just trying your config in HA. i have found upon any change (colour, effect etc) the brightness resets to 0 (off). is it possible to overcome this, so any change only affect the change and not reset s brightness. 
i am looking at using the grup topic to control an number of wled setups

YeonV, 2019-03-19:
> @YeonV
> Just trying your config in HA. i have found upon any change (colour, effect etc) the brightness resets to 0 (off). is it possible to overcome this, so any change only affect the change and not reset s brightness.
> i am looking at using the grup topic to control an number of wled setups

You might try to set the `on_command_type` to 'brightness' or 'last'.
(I am using brightness for some of my lights)

debsahu, 2019-03-19:
@aidbish `on_command_type` is depreciated for JSON lights ref: https://github.com/home-assistant/home-assistant/issues/20685, which you already know. I think this is default mqtt light so it is still a valid input.