wled/WLED #137: Problem with Alexa Device Emulation

TheBlackDevWarPixel, 2019-03-22:
Hey,

I have a problem with the Alexa Device Emulator, the Alexa-App and also the Website (alexa.amazon.de) find a other device but not the ledstrip.

**Output from https://drive.google.com/open?id=1d01gRgzTIyjCRMyLydFp0IwuMjQU1YuK :**

`Serial started, waiting 2 sec

Connecting to WiFi.............
WiFi Connected to JLan
IP address: 192.168.188.28
Ask Alexa to discover devices

Connecting UDP
UDP success
HTTP Server started...
Received UDP: 192.168.188.1:36086
:M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 5
ST: urn:schemas-upnp-org:device:avm-aha:1

Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/l2tpv3.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: upnp:rootdevice
NTS: ssdp:alive
USN: uuid:9580
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/l2tpv3.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:95802409-bccb-40e7-8e6c-C80E14C07C10
NTS: 
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/l2tpv3.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-upnp-org:device:l2tpv3:1
NTS: ssdp:
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/l2tpv3.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-any-com:service:l2tpv3:1
NTS: ssdp:
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/fboxdesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: upnp:rootdevice
NTS: ssdp:alive
USN: uuid:12
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/fboxdesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:123402409-bccb-40e7-8e6c-C80E14C07C10
NT
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/fboxdesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-upnp-org:device:fritzbox:1
NTS: s
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/fboxdesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-any-com:service:fritzbox:1
NTS: s
Received UDP: 192.168.188.1:36086
:M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 5
ST: urn:schemas-upnp-org:device:avm-aha:1

Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igd2desc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: upnp:rootdevice
NTS: ssdp:alive
USN: uuid:75
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igd2desc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:75802409-bccb-40e7-9f6c-C80E14C07C10
NTS
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igd2desc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-upnp-org:device:InternetGatewayDev
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igd2desc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:75802409-bccb-40e7-9f6b-C80E14C07C10
NTS
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igd2desc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-upnp-org:device:WANDevice:2
NTS: 
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igd2desc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:75802409-bccb-40e7-9f6a-C80E14C07C10
NTS
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igd2desc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-any-com:service:Any:1
NTS: ssdp:a
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/avmnexusdesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: upnp:rootdevice
NTS: ssdp:alive
USN: uui
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/avmnexusdesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:535502409-bccb-40e7-8e6c-C80E14C07C10
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/avmnexusdesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-upnp-org:device:avmnexus:1
NT
Received UDP: 192.168.188.1:36086
:M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 5
ST: urn:schemas-upnp-org:device:avm-aha:1

Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igddesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: upnp:rootdevice
NTS: ssdp:alive
USN: uuid:758
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igddesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:75802409-bccb-40e7-8e6c-C80E14C07C10
NTS:
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igddesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-upnp-org:device:InternetGatewayDevi
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igddesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:76802409-bccb-40e7-8e6b-C80E14C07C10
NTS:
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igddesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-upnp-org:device:WANDevice:1
NTS: s
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igddesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: uuid:76802409-bccb-40e7-8e6a-C80E14C07C10
NTS:
Received UDP: 192.168.188.1:1900
:NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
LOCATION: http://192.168.188.1:49000/igddesc.xml
SERVER: FRITZ!Box 7362 SL (UI) UPnP/1.0 AVM FRITZ!Box 7362 SL (UI) 131.07.01
CACHE-CONTROL: max-age=1800
NT: urn:schemas-upnp-org:service:WANDSLLinkConfig:1
Received UDP: 192.168.188.1:36086
:M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 5
ST: urn:schemas-upnp-org:device:avm-aha:1

Received UDP: 192.168.188.1:36086
:M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 5
ST: urn:schemas-upnp-org:device:avm-aha:1
`

I already tryed it with https://drive.google.com/open?id=1aN8bO10qiLvRbGyXYBaWixOAP2nyQ46- but also no success

debsahu, 2019-03-22:
You need a physical Alexa device (echo/dot etc) on the same network as WLED.

TheBlackDevWarPixel, 2019-03-22:
okay, so it won't work with a virtual device (configured my raspi as alexa), thought that it would work because my switch is also working.

closed 2019-10-25 by Aircoookie