wled/WLED #138: Problem with Compiling

Shizzzzzy, 2019-03-22:
Hi,
Everytime if i try to compile the code this error appears:

Arduino: 1.8.7 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 4M (1M SPIFFS), ck"

In file included from C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/esp8266_peri.h:24:0,

                 from C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/Arduino.h:38,

                 from D:\Users\Joshua\Desktop\WLED-0.8.3\wled00\wled00.ino:32:

C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther/tools/sdk/include/c_types.h:43:0: warning: "__packed" redefined [enabled by default]

 #define __packed        __attribute__((packed))

 ^

In file included from c:\users\joshua\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\stdlib.h:19:0,

                 from C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/Arduino.h:27,

                 from D:\Users\Joshua\Desktop\WLED-0.8.3\wled00\wled00.ino:32:

c:\users\joshua\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\sys\cdefs.h:250:0: note: this is the location of the previous definition

 #define __packed __attribute__((__packed__))

 ^

In file included from C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/esp8266_peri.h:24:0,

                 from C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/Arduino.h:38,

                 from D:\Users\Joshua\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:28,

                 from D:\Users\Joshua\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBrightnessBus.h:29,

                 from sketch\NpbWrapper.h:42,

                 from sketch\WS2812FX.h:30,

                 from sketch\WS2812FX.cpp:27:

C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther/tools/sdk/include/c_types.h:43:0: warning: "__packed" redefined [enabled by default]

 #define __packed        __attribute__((packed))

 ^

In file included from c:\users\joshua\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\stdlib.h:19:0,

                 from C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/Arduino.h:27,

                 from D:\Users\Joshua\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:28,

                 from D:\Users\Joshua\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBrightnessBus.h:29,

                 from sketch\NpbWrapper.h:42,

                 from sketch\WS2812FX.h:30,

                 from sketch\WS2812FX.cpp:27:

c:\users\joshua\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\sys\cdefs.h:250:0: note: this is the location of the previous definition

 #define __packed __attribute__((__packed__))

 ^

In file included from C:\Users\Joshua\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/Arduino.h:242:0,
[... 2310 lines cut ...]

   oappend(";--sCol:#");             oappend(cs[4]);

                      ^

D:\Users\Joshua\Desktop\WLED-0.8.3\wled00\wled18_server.ino:265:22: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   oappend(";--tCol:#");             oappend(cs[5]);

                      ^

D:\Users\Joshua\Desktop\WLED-0.8.3\wled00\wled18_server.ino:266:20: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   oappend(";--cFn:");               oappend(cssFont);

                    ^

D:\Users\Joshua\Desktop\WLED-0.8.3\wled00\wled18_server.ino:267:15: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   oappend(";}");

               ^

exit status 1
no matching function for call to 'ESP8266WebServer::arg(String&)'

Aircoookie, 2019-03-23:
Hi!
I'm sorry you are experiencing compilation issues!
What version of ESP8266 Arduino core do you use? I currently recommend 2.4.2.
Also, what ESP board do you have? If you have a NodeMCU or Wemos D1 mini, please select that in the Board menu, generic module is mostly meant for ESP01 boards.
Please also ensure that you have installed and updated the NeoPixelBus and FastLED libraries.
I hope we'll get WLED to work for you!

Shizzzzzy, 2019-03-23:
Hi,
Thanks for your reply, I already fixed the issue with the compilig errors, I just reinstalled my whole IDE, but to my annoyance a second problem occurs. After compiling and uploading

Der Sketch verwendet 511748 Bytes (48%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.
Globale Variablen verwenden 48572 Bytes (59%) des dynamischen Speichers, 33348 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
Uploading 515888 bytes from C:\Users\Joshua\AppData\Local\Temp\arduino_build_656071/wled00.ino.bin to flash at 0x00000000
................................................................................ [ 15% ]
................................................................................ [ 31% ]
................................................................................ [ 47% ]
................................................................................ [ 63% ]
................................................................................ [ 79% ]
................................................................................ [ 95% ]
........................                                             [ 100% ]

the Wlan AP doesn´t appear (I already read this thread https://github.com/Aircoookie/WLED/issues/85), so I tried it with a other, blank ESP8266, the same problem.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

---WLED 0.8.4-dev 1903191 INIT---
esp8266 2_4_2
heap 28072
LEDs inited. heap usage ~184
Load EEPROM
CSSID: Your_Network NO AP
C_NC
C_NC
Can't connect. Opening AP...

Connected! IP address: 0.0.0.0
HTTP server started
Ada

Shizzzzzy, 2019-03-23:
---DEBUG INFO---
Runtime: 10000
Unix time: 10
Free heap: 22400
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 0.0.0.0
---DEBUG INFO---
Runtime: 20001
Unix time: 20
Free heap: 22400
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 0.0.0.0
---DEBUG INFO---
Runtime: 30002
Unix time: 30
Free heap: 22400
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 0.0.0.0
---DEBUG INFO---
Runtime: 40003
Unix time: 40
Free heap: 22400
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 0.0.0.0

Aircoookie, 2019-10-25:
Hi, I just redone the WiFi logic and hope this is fixed with the latest master :) If not, feel free to reopen

closed 2019-10-25 by Aircoookie