wled/WLED #141: APA102

JanBosNL, 2019-03-25:
I dont see options to use my apa102 leds?

Aircoookie, 2019-03-26:
There isn't any option for that at the moment. 
Since the underlying NeoPixelBus library supports them though, I can add support for APA102 tomorrow!

debsahu, 2019-03-26:
@Aircoookie NeoPixelBus has APA102 support, its referred to as **DotStarSpiMethod**. Uses software and hardware SPI support. Hardware is async though and software is slow!

Aircoookie, 2019-03-26:
@debsahu thanks! I think WS28xx will remain the primary choice although APA102 definitely seem to have some advantage because of the CLK pin and additional global brightness possibility.

@JanBosNL just got an APA102 version to compile (completely untested though)
Just replace line 21 in NpbWrapper.h with `#define PIXELMETHOD DotStarSpiMethod` and delete the second parameter LEDPIN in lines 71 and 76.

JanBosNL, 2019-04-03:
Thank you @Aircoookie , I will try it out. 

Been trying lots of stuff, still learning a lot. First few days the only way to get some leds burning was to throw them in a campfire... At the moment though, I am actually compiling some nice first tests that seem to work OK.. 

When I have tested your code, I will let you know.

JanBosNL, 2019-04-03:
Compiled and uploaded, 

changed line 71 and 76 to:

```
  void Begin(NeoPixelType type, uint16_t countPixels)
  {
    cleanup();
    _type = type;

    switch (_type)
    {
      case NeoPixelType_Grb:
        _pGrb = new NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>(countPixels);
        _pGrb->Begin();
      break;

      case NeoPixelType_Grbw:
        _pGrbw = new NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>(countPixels);
        _pGrbw->Begin();
      break;
    }
  }
```
So far no luck, I first thought that the clock pin needed to be defined as well. Since the apa102 has 4 wires. GND and 5V and DATA + CLOCK

First though I tried to read up on the  DotStarSpiMethod you tried 

> 
> DotStarSpiMethod
> 
> This method should only be used with DotStar color features.
> It will use the available hardware SPI support for the board you build with.
> Generally the clock and data pins are not changeable for hardware support and thus the Pins argument is omitted on the constructor. If the pins are changeable like on the ESP32, then provide the pins to Begin.
> The hardware pins on the board are often labeled as MOSI for data, and MSCLK or CLK for the clock.

So seems defining the pins freely is not possible with the DotStarSpiMethod on the esp8266 platform. Since I use a NodeMCU on the esp8266platform, I checked the datasheet to find the MOSI and CLK pins. The data pin is called SD1 on the NodeMCU and the clockpin CLK

Plugged the DATA and CLK wire from the ledstrip to those pins and we suddenly got leds lighting up.

So it works. Not tested the rest of the webserver and functions. will try that another day.

JanBosNL, 2019-04-03:
OK,,, checked the webserver controls.. They are not working. Leds seems to hang in a random multicolor confetti mode that might just be random noise from the clk and mosi spi pin.

Seems adjusting any of the settings nor the amount of leds on the strip has any effect on the strip.

T-Arens, 2019-05-01:
I have implemented a working solution, based on DotStarMethod (software). It works great with 301 LEDs and an ESP-01. Note that you probably need a level shifter (I use a SN74AHC126D). 

I have created a pull request. Meanwhile, the code is available at [my place](https://github.com/T-Arens/WLED). I hope that is the correct way to contribute, I am new to GitHub.

salopette, 2019-05-01:
Where do you get such a level shifter and do you have to connect this?
I still have here 1 meter apa102 Stripe, would like to try!
I have it connected without it, but it just blinks, and the Esp8622 does not reboot / start at all.

T-Arens, 2019-05-01:
If it doesn't boot and behave normally, the level shifter is not your problem.
Where you can get a SN74AHC126N (N is easier to solder) depends on where you live, I guess (Digikey?).
The connection would look like this:
[image]

T-Arens, 2019-05-01:
You could also order a TXS0108E level shifter board from e.g. AliExpress.
[image]
VA = 3.3V
VB = 5V
OE = 3.3V
A1, A2 = Data/Clock from ESP8266
B1, B2 = Data/Clock to LEDs over 100 Ohms Resistors
GND = GND

salopette, 2019-05-01:
Which is better? can I then control synonymous LPD6803 ic Led Band?

https://www.amazon.de/Converter-Pegelwandler-BiDirektional-Raspberry-Mikrocontroller/dp/B07DKT9JXT/ref=pd_sim_147_5/259-6435335-5384064?_encoding=UTF8&pd_rd_i=B07DKT9JXT&pd_rd_r=3579f4a2-6c27-11e9-ba4e-017bb8e0c721&pd_rd_w=spxae&pd_rd_wg=o7FWo&pf_rd_p=b0773d2f-6335-4e3d-8bed-091e22ee3de4&pf_rd_r=67Q18ZX27A92A7X88BR2&psc=1&refRID=67Q18ZX27A92A7X88BR2

https://www.amazon.de/AZDelivery-TXS0108E-Converter-Arduino-Raspberry/dp/B07N7FFY2Q/ref=pd_lpo_sbs_200_t_1?_encoding=UTF8&refRID=D0WB7WQ665QB31P1NY0N&th=1

T-Arens, 2019-05-01:
Definitely the second one, the TXS0108E. Will be better for high speeds.
I don't know the LPD6803 and don't find much information about those LED strips, so I don't know.

salopette, 2019-05-02:
https://github.com/area3001/esp8266-arduino-lpd6803

T-Arens, 2019-05-03:
The level shifter should work. The protocol looks different from the APA102s.

salopette, 2019-05-03:
Hi, do I have to change something in your library for apa102? or is it already finished?

//PIN CONFIGURATION
#define BTNPIN 0  //button pin. Needs to have pullup (gpio0 recommended)
//#define USE_APA102 // Uncomment for using APA102 LEDs.

T-Arens, 2019-05-04:
Look [here](https://github.com/Aircoookie/WLED/pull/160).

salopette, 2019-05-05:
I have it so wired now, but it does not work.
Did I do something wrong?

[image]

debsahu, 2019-05-05:
Hardware SPI for ESP8266 are on GPIO13(MOSI/DATA) and 14(SCK/CLOCK) pin.

T-Arens, 2019-05-05:
Try swapping clock and data, I got that mixed up, too. You uncommented #define USE_APA102, right?

T-Arens, 2019-05-05:
And what is D3 and D4? You have to use GPIO 0 and 2. Don't know what board you are using.

Aircoookie, 2019-05-05:
@salopette hmm, the wiring looks fine. Nothing comes to mind but trying to swap, either.
@T-Arens D3 and D4 are GPIO0 and 2 on Wemos D1 and NodeMCU, so no issue there.

salopette, 2019-05-05:
Hello, when I swap data and clock, nothing happens.

[image]

T-Arens, 2019-05-05:
And what does happen, when you don't swap them? Are they doing anything? Directly after power up of the ESP, they should flicker wildly. If that doesn't happen, something is electrically wrong. Are you sure you connected to the right end of the strip? It happened to me once, I connected to the wrong end, and then, nothing happens. 😉

salopette, 2019-05-05:
When I trade, the LEDs only stay light white.
I followed the arrows for direction

T-Arens, 2019-05-05:
You mentioned LPD6803 LEDs earlier. Are you using them, or APA102s?

salopette, 2019-05-05:
@T-Arens 
 i use apa102.

I did it!

It does not seem to be ok with my ESp8622?
Have now taken a Witty Cloud and it works!
But without level shifter! Do I need that umbedingt?

T-Arens, 2019-05-05:
If it works, no. ;-) Just if you get into problems (flicker, not working at all...), put it in.

salopette, 2019-05-06:
many thanks for the support!
You have been great!