wled/WLED #142: Can not connect to other WiFi

zwiebelslayer, 2019-03-26:
Hello.
When i am entering the Credentials it is updating them, but not connecting to the Wifi. Eventhought the wifi credentials are right it starts up the AP again.

debsahu, 2019-03-26:
Is your WiFi network 2.4GHz? ESP8266 and ESP32 only work in that frequency and not 5GHz.

zwiebelslayer, 2019-03-26:
Yes it is. I have seen that some other people have the same issue with the FRITZ!BOX 7049.

zwiebelslayer, 2019-03-26:
I have tried to access an other WiFi hot spot this does not work either. Where do i have to flash the .bin file

debsahu, 2019-03-26:
Instructions are here: https://github.com/Aircoookie/WLED/wiki

zwiebelslayer, 2019-03-27:
I think the problem might be the .local adress i tried this out with an other sketch and i did not connect to the WiFi eihter. Is there a way to just have a adress like this 192.168.178.XX?

rborkhuis, 2019-03-28:
I have the exact same problem as @zwiebelslayer .
I've tried to edit the source code, but nothing works.
I even did it OTA and it didn't work.

I have a esp32.

Aircoookie, 2019-03-28:
@rborkhuis Sorry that it's not working right away for you! You say you have exactly the same issue, so i suppose the AP works, but connecting to your home network doesn't. I suppose you've checked that password, signal range or 5Ghz are not the culprit, so we'd need to do some deeper debugging to find out what the issue is. Yesterday I had a case on discord (esp8266) where lwip v2 (the tcp stack) caused watchdog resets. Lwip 1.4 worked flawlessly for the user though. Then again, Lwip 1.4 causes extreme slowdowns on some ESPs when compiled with PIO... it's all a bit messed up!

Looking forward to your esp8266 results tomorrow!

rborkhuis, 2019-03-29:
@Aircoookie 
No problem, its fun to learn some more C.
Im not sure if i can test it on the esp8266 wemos d1 today.
It might be the board version.
I know for sure that the LOLIN32 works, because i compile it on the arduino ide.
If i select the esp32 dev board it doesnt connect on the arduino ide.
The EEPROM doesn't reset when i recompile it.

Im new to arduino btw :).

Aircoookie, 2019-03-29:
@rborkhuis So if you compile for LOLIN32, WLED works and connects, but if you compile for esp32 dev board, it doesn't? Have you tried if the same issue arises with other WiFi (example) sketches?

Because both 0.8.3 and 0.8.4 are affected, I don't think a recent change to my code has caused the issue, but it's more of an underlying problem.

You could also try to flash a [binary](https://github.com/Aircoookie/WLED/releases) using esptool.

Hope we'll be able to get it to work nicely :) same for you, @zwiebelslayer !

rborkhuis, 2019-03-29:
I've tried to compile it for LOLIN32 with PIO but it still doesn't work.
I've tried OTA sketches and the wifi worked, but when i flashed wled to it OTA it doesn't connect.

So i think the problem is in the code.
Im going to take a look at the issue page of the esp32 arduino library.
maybe i can find something usefull there.

debsahu, 2019-03-29:
I have a LOLIN D32 and can check this, in PIO replace `board = esp32dev` to `board = lolin_d32`

rborkhuis, 2019-03-29:
That's new, POI says that i need to use LOLIN32...
I will try it when im home.

debsahu, 2019-03-29:
Ref: https://docs.platformio.org/en/latest/boards/espressif32/lolin_d32.html

zwiebelslayer, 2019-03-29:
Ok i now flashed the programm with the arduino sketch and everything works fine. I dont know where the Problem was but it worked. Thank you for your help. Danke!

zwiebelslayer, 2019-03-29:
I now got only one more problem the effects do not load only when i am using the AP. How can i fix this.
I am connect with this adress http://192.168.178.36/ not the other one: wled-cc50e3c4906e.local:8266 .
[image]

zwiebelslayer, 2019-03-29:
It also works in the Browser when i change the theme to mobile.

rborkhuis, 2019-03-29:
my esp8266 won't connect to my home network too...

zwiebelslayer, 2019-03-29:
Have you flashed the esp over arduino or the .bin file?

rborkhuis, 2019-03-29:
flashed it with PIO

rborkhuis, 2019-03-29:
I have them both working.
It was a board problem...
Maybe put this in the readme?
or make different bins?

closed 2019-04-17 by zwiebelslayer