wled/WLED #143: UDP switch LEDs on/off

bboehmke, 2019-03-26:
I tried the UDP realtime functionality and it is working great.
The only thing I noticed is that the LEDs must be turned on and also the brightness must be set before.

Is it possible to switch the LEDs on/off and maybe also change the brightness via UDP?

Actually I want to have the LEDs by default turned off and switch them on only if I send packets.

Aircoookie, 2019-03-26:
Hi! There are two options to achieve what you want:

1. Check "Force max brightness" in the Realtime section of sync settings. It has the drawback of always being full brightness though as the name suggests.

2. This is more of a workaround, but it might be an option for you if you mostly use UDP instead of the normal lighting mode. Just set the color to black and leave the brightness on the level you'd like it to be when receiving UDP!

bboehmke, 2019-03-27:
Both options seems to solve my problem.

One thing I don't understand: 
Is there a different between a lower brightness and an RGB value of for example (#001000)?
So I can even reduce the brightness this way for option 1 by using "lower" RGB values.

Aircoookie, 2019-03-27:
That's nice!

Yes, you can do that. The brightness is a global multiplier, it affects colors, effects, UDP packets, everything. For displaying a solid color, there is no difference, but once you use some effects, it gets clear.

bboehmke, 2019-03-28:
Perfect.
That is enough for my use case.

closed 2019-03-28 by bboehmke