wled/WLED #145: No Web UI when connected as client

yloose, 2019-04-01:
So I already have ESP with the WLED firmware working perfectly fine, so I tried to flash another ESP. The new one also opens the AP and I can connect and control the strip, but when I connect it to my WiFi network it doesnt have a web server running.

This one is working:
[image]
This one is 
[image]
not

Aircoookie, 2019-04-02:
Hi! You probably have compiled the project from source using the latest 2.5.0 arduino core. Most likely you are experiencing https://github.com/esp8266/Arduino/issues/5784 . Installing Arduino core 2.4.2 or one of my binaries should solve the issue!

yloose, 2019-04-06:
Yeah, that was it. It works now, thanks!

closed 2019-04-06 by yloose