wled/WLED #146: Save settings via web interface

cruunnerr, 2019-04-02:
Hi there,

first i want to say, that here i found the best way to power ws2812 stripes with ESP8266. Great software. Thank you for that :)

I wish to save the settings i made (like macros, default values etc.).
Is there a way to save my settings? When the ESP lost the configuration or crashes it would be nice to have a save file and i just need to upload on a fresh WLED-System :)

Thank you

Oh ehm.... and i want to ask, if it is possible, to publish the on/off state via Mqtt as boolean (true/false) and a separate brightness and effect value. 

That would be great for the integration with iobroker.

Thank you again and sorry for my bad english :(

Aircoookie, 2019-04-10:
Hi!
Awesome that you like WLED!

Yes, I agree that it would be very useful to be able to backup and restore (parts of) the settings! I'll try and think of a way to achieve that :)

About MQTT, I can easily add setting on/off with a boolean value. Adding more topics to publish (like a separate topic for brightness, effect, effect speed, each color...) is something I'm still hesistant about, because it might impact performance too much. Every value is changable via the `wled/devicetopic/api` topic, but I realize that might not be the easiest solution for home automation platforms. We'll see!

Don't worry, your english is absolutely error free :)

YeonV, 2019-04-15:
@cruunnerr You know that you get the XML response as MQTT Message on `device/v`:
```<?xml version="1.0" ?><vs><ac>255</ac><cl>0</cl><cl>255</cl><cl>0</cl><cs>0</cs><cs>0</cs><cs>0</cs><ns>0</ns><nr>1</nr><nl>0</nl><nf>1</nf><nd>60</nd><nt>0</nt><fx>0</fx><sx>220</sx><ix>128</ix><fp>0</fp><wv>-1</wv><ws>0</ws><md>1</md><cy>0</cy><ds>MyLight-1337A3</ds></vs>```

I changed the arduino code to return a custom JSON-response on MQTT /v:
```
 {"ip":"192.168.1.172","leds":21,"topic":"blade/ledx/strip2","color":[0,0,255]}
```
Also there is a the json http-API with http://[ip]/json/info

Aircoookie, 2020-01-07:
There is a way to back up everything, macros, presets and settings using esptool! See https://github.com/Aircoookie/WLED/issues/479

Downloading them as .json files will be available in 0.10.0 :)

closed 2020-01-07 by Aircoookie

jangrewe, 2020-12-27:
Hi @Aircoookie , i recently came across WLED when building my own Ambilight, and i LOVE it!
Along the process, a couple of ESP8266 got fried (one while "in production"), and i was wondering if that config backup/export (and preferably also import) via JSON is available by now? We're a bit past 0.10.0 ;-)

JakeShirley, 2021-02-16:
Resurrecting old thread.  Is it possible to back these settings up now via JSON? I'd love to do it daily just in case.  I don't want to lose my dozens of presets!

Aircoookie, 2021-02-18:
@jangrewe @JakeShirley yes, this is now supported in 0.11.0 and up! Just go to the `[WLED-IP]/edit` page and download the cfg.json and presets.json files (right click on file name)! To restore, you just delete the existing ones, upload the ones you downloaded for the backup and reboot :)

JakeShirley, 2021-02-18:
To expand after looking at the /edit page, here are the two files you can hit:
- `[WLED-IP]/edit?download=cfg.json`
- `[WLED-IP]/edit?download=presets.json`

n0valis, 2021-05-06:
How do I restore from cfg/presets.json?

Aircoookie, 2021-05-07:
@n0valis hi, just go to the [WLED-IP]/edit page and use the "Choose file" and then "Upload" buttons for both `presets.json` and `cfg.json`. You might need to reboot WLED afterwards in order to apply the restored configuration.

n0valis, 2021-05-08:
nice!

n0valis, 2021-05-08:
maybe I am just to stupid. But if I upload my saved cfg with different settings, there has nothing changed in the settings. Even after reboot.