wled/WLED #147: Keep state after UDP

YeonV, 2019-04-07:
Would it be possible to add a flag to keep the last UDP state?

Context:
At the moment the strip falls back into the previous state after an UDP-stream.
My momentary workaround is to send the last state also via MQTT.
This bug sometimes, so I need to restart the strips.

Posible solution-approaches:
- setting a flag via web interface: "Keep last UDP state"
- adding a new UDP identifier byte 0: 5-WARL+save-state

Thank you soo much for this great project!!!

Aircoookie, 2019-04-07:
Awesome, I'm really glad that you like WLED!

That addition is definitely possible and useful! I'll add it very soon!
The only issue is that strip contents are reset if turning off and on the strip, so the implementation would probably look like:
- Send `255` in the second byte of the WARLS packet (seconds to keep the state) to keep the state "indefinitely"
- Any color or effect change in the web UI will return to normal mode. Brightness may be changed, but complete on/off is not possible right now. I'll try to find a way to potentially change that, though!

YeonV, 2019-04-08:
Thank you soo much for your answer.
Both points dont seem to work :/
- 255 is returning to normal mode after 255s
- while those 255s are running, you cant set anything via MQTT (because the UDP-stream is still open)

Aircoookie, 2019-04-08:
Oh, I'm sorry I wasn't clearer. This is just the way i plan to add it, it doesn't work in the current version. Will add it soon!

YeonV, 2019-04-08:
Ah omg sorry for my missunderstanding.
Thank you soo much mate!!!

YeonV, 2019-04-14:
@Aircoookie BOOM you made it!
Tested and working perfectly :)

THANKS SO MUCH!

closed 2019-04-15 by YeonV