wled/WLED #148: Flickering clours

Masterofx2, 2019-04-10:
Hello.
When i connet my LED stripe with 98 Leds i had to change the colour from the standard to RGB. But when i connect the same led stripe with only 10 LEDs the Standard stetting works just fine. Also when i try to dimm the led stripe it flickers a lot this does t happen on the led stripe with 10 diodes. I havent tried a level shifter yet. The wemos is not the problem eiter i have switched both wemoses from the two led stripes

Aircoookie, 2019-04-10:
Hi! Seems like it is a power problem. You shouldn't use as many as 98 LEDs via USB, I highly recommend you use an external 5V ~4A power supply connected to the LEDs. You'll also need to increase the "Maximum Current" setting in LED preferences to the rating of your power supply!

Masterofx2, 2019-04-10:
I have an 10amp power supply

Aircoookie, 2019-04-10:
Ok then that is probably the problem :) Just set "Maximum current" (the field directly below LED count) to "10000" and you should be fine!

Masterofx2, 2019-04-10:
Ok i just did that it already got better but when i want to change the brightness to a higher value it flashes for around 3 times until it reached the set brightness. Also the first led is always green no matter what colour i picked is this intended?

Masterofx2, 2019-04-10:
Also the stripe with 10LED needs the standard gbr mode while my 98LED stripe needs rgb eventough they both are the same stripe just cut into different lenghts.

Masterofx2, 2019-04-10:
Thank you for your fast response. I have tried the first and the second thing. 
1. The first pixel was not broken it happend again to the second one.
2. Then i injected the power in the last Pixel as you said in point 1. and wow it fixed a lot. The colours are now the same as the ones in the 10 LED strip so the standard GBR. Also the breathing effect now works at least half decent.
Problem now: When i display solid colours or other effect the are not displayed properly as you can see here: https://youtu.be/VDR3uasKwXE
In the Video i am using the breathing effect and for around 1ms or so there are just random colours flashing up. (it is not easy to see in the video, but clearly with our eys)

Masterofx2, 2019-04-10:
Ok i solved the problem.
i justed turned down the power supply to the minimum setting ro around 4.2V and it justed worked like it should. So maybe the better soloution would be adding a level shifter but for now this just works fine.

Aircoookie, 2019-04-10:
Awesome that it works now!
So, yes, this is a voltage level issue. Having a variable powersupply like yours eliminates the need for a levelshifter. The LEDs should work fine with a 3.3V signal on up to 4.7V. 5V is just so slightly out of spec, it works 90% of the time, but then there are the 10% where it doesn't.

Masterofx2, 2019-04-10:
Ok i cheered too early no it doesnt work any more so i defenetly will add a level shifter.

Masterofx2, 2019-04-10:
Ok i now added a level shifter and now it finally works consistent. (Problem solved thank you very much)

Aircoookie, 2019-04-10:
Sounds good! Have fun with the software and let me know if any further issues arise! Just out of interest, do you know the model nr. of the levelshifter you use?

Masterofx2, 2019-04-11:
I am using these ones: https://www.ebay.de/itm/5-Stuck-4-Kanal-Pegelwandler-Level-Shifter-Konverter-I2C-5V-3-3V-Arduino-Ra/253050668603?hash=item3aeafec63b:g:2rAAAOSwlAFb4Dr7

closed 2019-10-25 by Aircoookie