wled/WLED #149: Which method has the least overhead?

mentaluproar, 2019-04-13:
I'm trying to tie this into apple homekit and I'm trying to figure out what would have the least impact on this device.  Homekit has some things that might link up to this, and I already have some Hue stuff around the house.  There are so many options I'm confused how exactly I should connect this.

Aircoookie, 2019-04-13:
Hi! I don't have any experience with homekit, so I cannot provide you with "the best way" to connect your WLED lights, unfortunately. I doubt Homekit has provisions for using either the WLED http, udp or json API. MQTT or Blynk are also likely not an option. If you can send HTTP requests from homekit, I do recommend the HTTP api.

The hue sync feature would indeed work well (including voice control), but keep in mind that you can only sync WLED to an existing Hue light and not control it on its own via HomeKit. Hue sync does poll your hue bridge every second or so for updates to the light, but it has no major performance implications in the latest 0.8.4 version.

mentaluproar, 2019-04-13:
so something like this?
https://www.npmjs.com/package/homebridge-web-rgb

mentaluproar, 2019-04-14:
Okay, so I have most of it tied in.  color selection doesn't work though.  I used https://www.npmjs.com/package/homebridge-http-rgb-push

I think the color selection issue is my config being shit, not your side, so I'm going to ask for help from the author there.  Unless you are feeling adventurous :)

bachoo786, 2019-10-15:
Hey @mentaluproar did you manage to tie up wled with homebridge?

mentaluproar, 2019-10-16:
I don't think so.  I'm redoing the config since something went wrong a while back and I'm just now getting around to fixing it but I have absolutely no idea how I did it.

THATDONFC, 2020-02-11:
I managed to get this working with this plugin https://github.com/QuickSander/homebridge-http-rgb-push
So far I can turn the LED's on/off and change the brightness. Might see if I can fork this project and create a WLED specific plugin or create a PR to add some features to the original.

The biggest issue is the lack of the ability to change the colors. This plugin requires a HEX value for the color and I don't see a method for extracting the HEX color via the HTTP or JSON API's.

mentaluproar, 2020-02-11:
There’s also a way to interact with it over Json in homebridge but I never could get that working cleanly. We need a homebridge plugin for this. The existing ones work but are really lacking compared to what WLED could do with a dedicated plugin. It already exposes a lot already.

ceeeekay, 2020-05-01:
There's a basic homebridge WLED plugin [here](https://github.com/fabiopigi/homebridge-wled-simple)

THATDONFC, 2020-05-01:
Thanks! That must be new. I’ll test is out this weekend

gilshahar7, 2020-05-08:
Hey, i just stumbled upon this issue and i was wondering if its possible to somehow merge this project into WLED:

https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266

I have gotten WLED to work with HomeKit using homebridge which is great but not ideal.

the best way would be to not use a homebridge on a RPI and have it straight on the esp8266 since i have another WLED strip at my parents house.

closed 2020-09-12 by stale[bot]

dennis777, 2023-10-20:
Hey all, wondering if anyone has ever came to a plausible conclusion here?

sommo, 2023-10-20:
yes, there is a plug-in for this

defektu, 2024-01-11:
Hi all, I managed to connect homekit with another spare ESP8266 to WLED without homebridge. It just reroutes homekit request to WLED via WebSocket. 
You can set multiple homekit devices with single ESP8266 with routing. Works like cheap homebridge.

sommo, 2024-01-12:
so without an rpi that run homebridge, could you link the project? thank youInviato da iPhone

defektu, 2024-01-12:
Hi @sommo! Sorry for deleting the comment. I have not prepare a repo for that yet. I might be able to tell you the process.
I have a Wemos D1 with WLED running. Also, have an ESP01 that I programmed. 

I also do not have HomeKit Hub. I downgraded my Home Architecture so my iPad is now able to act as Hub. This means I can easily change my lights even if my iPhone is not connected to the local network. [This](https://discussions.apple.com/thread/254934792?sortBy=best) still works at iOS 17.12 

Here are the steps I did:
- ESP01 uses https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266
- I modified this example: https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/tree/master/examples/Example05_WS2812_Neopixel
- Added ArduinoWebsockets. Prepared JSONs for WLED.
- It connects to WLED WebSocket when a wifi connection is present. Keeps itself alive on the socket.
- Just converts and redirects the HomeKit messages (on-off, hsv, brightness) to WLED (on-off state, RGB data, brightness).

So ESP01 acts like a simple bridge.

gilshahar7, 2024-01-12:
Thats a cool implementation! 
ultimately if someone is able to combine the ESP-HomeKit project into a usermod on WLED we can omit the bridge part and have everything on the led controller

defektu, 2024-01-12:
I can change hue, saturation, brightness and on-off state straight from my iPhone Home. On 1MB Flash Size ESP01, almost realtime with sockets... Slight delay when iPhone is not connected to local wifi. possibly the delay is from Apple Home communication.

I think this seems possible with [Mixiaoxiao/Arduino-HomeKit-ESP8266](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266) repo. [ErrorErrorError/Arduino-HomeKit-ESP8266](https://github.com/ErrorErrorError/WLED)  implemented ESP-Homekit to WLED actually. But it only works for ESP-32 cause crypto part. Mixiaoxiao's port is lightweight. I might get another 4MB ESP8266 and try to implement it in my spare time. (Don't wanna ruin my current setups lol)

defektu, 2024-01-13:
Well, tried something with https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266. ESP shuts down when trying to pair. But It sees it as a HomeKit device while running.

htx996, 2024-06-25:
> Hi @sommo! Sorry for deleting the comment. I have not prepared a repo for that yet. I might be able to tell you the process. I have a Wemos D1 with WLED running. Also, have an ESP01 that I programmed.
> 
> I also do not have HomeKit Hub. I downgraded my Home Architecture so my iPad is now able to act as Hub. This means I can easily change my lights even if my iPhone is not connected to the local network. [This](https://discussions.apple.com/thread/254934792?sortBy=best) still works at iOS 17.12
> 
> Here are the steps I did:
> 
> * ESP01 uses https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266
> * I modified this example: https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/tree/master/examples/Example05_WS2812_Neopixel
> * Added ArduinoWebsockets. Prepared JSONs for WLED.
> * It connects to WLED WebSocket when a wifi connection is present. Keeps itself alive on the socket.
> * Just converts and redirects the HomeKit messages (on-off, hsv, brightness) to WLED (on-off state, RGB data, brightness).
> 
> So ESP01 acts like a simple bridge.

It's great how to modify it