wled/WLED #150: [Feature Request] Option to disable ESP8266 onboard LED

kaiomatico, 2019-04-13:
Hi, first of all thank you for this great project und for providing ESP8266 binaries. The only thing which I really like to see added is an option to disable the ESP8266's onboard LED. Is this possible? Thanks in advance!

Aircoookie, 2019-04-13:
Hi! Great that you like WLED! Unfortunately, due to hardware limitations, the best pin for driving LEDs is the hardware UART pin GPIO2 - which is the same pin the onboard LED uses, so it cannot be turned off.

There are three workarounds to the problem:
- Cover the LED so it doesn't shine through as bright.
- Use a screwdriver or soldering iron to physically remove the LED
- Use a different pin for driving the LEDs. Software bitbang driving and DMA work surprisingly well. If you have less than about 250 LEDs, DMA driving on pin GPIO3 works well. You can change the pin in the file `NpbWrapper.h`. Let me know if you want a binary with a different pin configured.

Have fun using WLED!

kaiomatico, 2019-04-13:
Thank you for your fast and helpful answer. It would be super great if you could provide me a binary 👍

Aircoookie, 2019-04-14:
Sorry for keeping you waiting! Here is the binary for DMA support: https://drive.google.com/open?id=1kwkm9I0pe0HoQa0brNBaQAmr6jblBxsb 

Keep in mind you need to connect the strip to GPIO3 (labeled RX on Wemos D1 and NodeMCU). You can just flash this binary via the WLED software update function in the Security settings web UI.

kaiomatico, 2019-04-15:
Thank you very much, it works perfectly fine!

closed 2019-04-15 by kaiomatico