wled/WLED #151: MQTT Set RGB sometimes leads to FFFFFF

StefanRu1, 2019-04-14:
Hi,

first of all thank you so much for this great software for the esp8622! 

It seems i have a problem with MQTT.
I use FHEM as home automation server and connect WLED via MQTT.
It works well but sometimes when i set an RGB value with:
wled/moodled/col H<HEX>
The response in a020a621c6c3:wled/moodled/c:.* is #FFFFFF and the lamp turns also white.
I can not see a pattern. Sometimes it works a view times good but than it switch to FFFFFF. 
Sometimes it nearly always sets FFFFFF when i set a hex value.

Do you have an Idea whats happening? Can i somehow help in finding the issue?

Thanks and best regards,
Stefan

StefanRu1, 2019-04-14:
One more comment, i have also problems with ON,OFF,T in wled/moodled.
OFF always works, but ON and Toggel only sometimes works. I have to do it 1-10 times until it works.

Thanks,
Stefan

Aircoookie, 2019-04-14:
Hi Stefan!

great that you like WLED!
I'm sorry to hear that MQTT seems to be an issue for you, however I can't seem to reproduce it.

Are you using WLED 0.8.4? It uses a more reliable MQTT library than 0.8.3. Also please make sure that the ESP8266 has good reception.

The easiest way to figure out whether FHEM sends out wrong data or if it is an issue with WLED is to install a 3rd party MQTT client (for example [MQTT Dash](https://play.google.com/store/apps/details?id=net.routix.mqttdash) for android) to your phone or PC and subscribe to the topics `wled/moodled`, `wled/moodled/col`, and `wled/moodled/c`. If the `col` topic, additionally to the `c` topic, already contains `#FFFFFF` instead of the selected color, it is an issue with the FHEM configuration. If not, send me the color hex you tried so I can attempt to reproduce the issue!

I hope we can get this working well for you!

StefanRu1, 2019-04-14:
Wow, thanks for the fast answer 👍 
I use 0.8.4, but if you do not know about such issues i think FHEM is causing the trouble. 
I using a FHEM build in MQTT Broker. Perhaps there is an issue.

Thanks alot for the info about the 3rd party client, i will try this and will post my findings as soon as did a little bit of testing.

Thanks,
Stefan

StefanRu1, 2019-04-14:
Hi,

ok i did a short test. When i use the topic wled/moodled/c for receiving and wled/moodled/col for sending in MQTT Dash and send from the dash it is working fine 👍 

But when i set the color from the webgui or alexa the  wled/moodled/c topic says invalid format.
I used color Color topics in MQTT Dash. 

So i added also a text type for wled/moodled/c. 
I can see that when i set the color with MQTT dash color type in wled/moodled/col everything works fine, but as soon as i set the color externally (web, Alexa) i get #FF reading in wled/moodled/c.

Perhaps this is causing troubles in my FHEM.
Can you reproduce this? Is this an issue or a feature?

Thanks,
Stefan

StefanRu1, 2019-04-14:
Oh and even now i got a #FFFFFFFF even when i set the color out of MQTT Dash to some kind of purple.
But it is more unlikely to happen in MQTT Dash.
Nevertheless i still communicate over the FHEM MQTT server. 
Do you think this can be an issue?

Lets see if you can reproduce it or not, if not i will change the MQTT Server on my Raspberry.

Thanks a lot,
Stefan

Aircoookie, 2019-04-15:
I got news on this!
Turns out there is an issue with the /c topic data (not with setting color via /col), and that is that leading zeros are truncated. So blue `#0000FF` is reported as `#FF` for example. Many programs might have issues with that, luckily @YeonV has created a fix today which is already in the latest master version!

StefanRu1, 2019-04-15:
Oh, perfect! That was fast 👍 
I will test it as soon as possible!

Thank you and @YeonV so much for the help.
Stefan

StefanRu1, 2019-04-19:
Ok, i compiled today the latest sources and updated my WLEDs.
What should i say it works perfect now with MQTT and FHEM.

Thank you so much and again thank you for this great software!

closed 2019-04-19 by StefanRu1