wled/WLED #153: Compiled for ESP32 and no effect list to select

Beliar-666, 2019-04-17:
Hello, 
i´m running Wled on a 8266 and there it works like charm. No i Compiled it to use it on my ESP32 and i can not select any effects. On the selection menu it only shows up loading all the time. There where no errors while compiling.

Any Ideas where the Error is?

Beliar-666, 2019-04-18:
I now tried your hints from Post [118](https://github.com/Aircoookie/WLED/issues/118) but also this won't help. i can see the List under /json/effects and also downgrades to version 0.8.3. My Arduino ide is 1.8.9 and my Core ? and LW IP ?.

Beliar-666, 2019-04-18:
next hint: tried on my pc with opera, firefox, chrome and on both, also the nodemcu8266, no effects list. On my smartphone in Chrome on both, esp32 and 8266, there is the list. strange issue.

closed 2019-04-18 by Beliar-666

Aircoookie, 2019-04-18:
Hi!
Great that you managed to make it work already!
What exactly fixed the issue for you?

Beliar-666, 2019-04-18:
i Don´t know what it was exactly. Now i try to go again to the 8.4

Beliar-666, 2019-04-18:
Both chips r now running with 0.8.4 without errors i think. Is there any possibility to change mqtt port? I used the binarys, and i think there is the Port on 1883 nut i need 1884.

Aircoookie, 2019-04-18:
Sounds good!

For mqtt, using port 1883 is standard in almost all configs, so there is no web setting for it. You chan change it in the top of `wled18_mqtt.ino` (`#define WLED_MQTT_PORT 1884`), but then you need to build the project from source. If you'd like, i can send you binaries for esp8266 and/or 32 with port 1884 tomorrow.

Beliar-666, 2019-04-19:
Ready binary for 8266 and 32 with port 1884 would be fine. Already i can say that both are running with the 0.8.4 like charme since yesterday.

Beliar-666, 2019-04-22:
@Aircoookie do you have the bins ready with port 1884?

Aircoookie, 2019-04-22:
Sorry for making you wait! (this happens way too often, I try to respond quickly and end up forgetting it :) )

[wled084_mqtt1884.zip](https://github.com/Aircoookie/WLED/files/3103458/wled084_mqtt1884.zip)

Beliar-666, 2019-04-23:
Many thanks @Aircoookie 
I Now only have one Problem with 8266. I Updated it, and after that there short was the Gui of Wled and then without doing anything the Connection broke and the Stripe goes on and off all the Time. I tried to flash other Versions also and tried other Software like Tasmota, EasyESP etc. but nothing works. Also erased the whole 8266 and then flashed. Flashing works, but after flashing there is no AP and no thing to see in my router. so the 8266 is no more reachable.

sansillusion, 2019-04-23:
Are your LEDs on their own power supply. It sounds like mabe a brownout ?

Beliar-666, 2019-04-23:
LED Strip is ok. it works on other ESP ;-) I Think the 8266 is bricked. And no, npo own power supply, and it isnt necessary cause only 15 Pixel. and it runs like charme before the flash of the file provided by @Aircoookie

Beliar-666, 2019-04-23:
Good news, the Device seems not to be bricked. I have flashed Espeasy and after the xxx flash and delete it works. now i can again try to flash Wled on the Device.

Beliar-666, 2019-04-24:
@Aircoookie The 8266 runs again with the .bin provided with Port 1884. Maybe the flashprozess before was faulty.