wled/WLED #154: Setting MQTT broker causes continous wdt reset

aidbish, 2019-04-19:
Just downloaded the latest commit, everything works fine till i set a MQTT broker. After rebooting in never comes back online and the output from the serial connection shows wdt resets

Ada

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0fd86a07
~ld

StefanRu1, 2019-04-19:
That is strange,
i just compiled a version and it works fine.
I would add my compiled binary, but i do not know is this is allowed and wanted here.

Best regards,
Stefan

Aircoookie, 2019-04-19:
Hi,
this is open-source, everything is allowed :)
Those wdt resets are worrying, 99% likely that this is a memory fragmentation problem again.
I'm glad that you got it working now, but here are some points that might cause it:
- Running an older arduino core than 2.4.2, they use too much memory
- The more LEDs are configured, the more RAM is used and issues become more likely
- Sometimes certain lwip versions cause issues. I recommend lwip 2 in general, but sometimes switching to lwip 1 may solve some issues

Anyways, have fun with WLED and let me know if there are any new issues!

aidbish, 2019-04-20:
Still an issue for me. 
Not using older core
only 16 ws2812b
using lWIP 2 higher bandwidth

StefanRu1, 2019-04-20:
Hi, 
did you erase the ESP before flashing?
What kind of ESP do you have?

Here is my working build from yesterday for a ESP8622 4M. 
[wled00.ino.d1_mini.zip](https://github.com/Aircoookie/WLED/files/3099875/wled00.ino.d1_mini.zip)

Best regards,
Stefan

aidbish, 2019-04-20:
Yeah i set to erase all contents, same board a D1 mini

StefanRu1, 2019-04-20:
I have not set a gpio, so it is gpio2 for the led stripe.

Aircoookie, 2019-04-20:
@aidbish so you are getting wdt resets on gpio3 with only 16 leds configured? I'll need to look into this.

aidbish, 2019-04-20:
@Aircoookie yeah mate

aidbish, 2019-04-21:
Update
my comment regarding d1 mini was part correct it is a d1 mini pro. but after changing board type there was no change
i did however go back over commits and if i comment out the the line sendHADiscoveryMQTT(); in wled17_mqtt.ino i do not get any wdt reset and everything is stable

Aircoookie, 2019-04-22:
Confirmed that HA autodiscovery (which is always enabled when MQTT is) causes WDT resets after connection to the broker when used with DMA method.
Will need to investigate further.

aidbish, 2019-05-09:
@Aircoookie Just wondering if you have had a chance to see whats causing the issue?

Aircoookie, 2019-05-09:
@aidbish since there is no exception (ESP crash), but a direct wdt reset, this is 99% a stack overflow issue since the 3kB buffer for the autodiscovery packet is too large. A possible fix would be to dynamically allocate the buffer in heap, but I'm afraid of it causing too much fragmentation.

SmartMediaRLP, 2019-05-23:
@Aircoookie i got this Reset Loop too. Everything is fine, until i add the MQTT Data. Then i have to flash again. Is there a solution for it? Do i have to comment out the Line with sendHADiscoveryMQTT(); ?

zewelor, 2019-05-30:
Are there any recommendations for any workaround, to still have working mqtt ? Dont use DMA method ?

Aircoookie, 2019-05-30:
@zewelor Yes, either don't use DMA or disable the HA autodiscovery by commenting out line 48 in wled17_mqtt.ino (`sendHADiscoveryMQTT();`)

SmartMediaRLP, 2019-05-30:
@aircoookie commenting out Line 48 won't help. It won't works furthermore. There are still wdt crashes again.
What is DMA?

aidbish, 2019-05-31:
Commenting out the line does work for me

From what i understand of DMA, If you are using NodeMCU and Wemos and using LEDpin 3 (RX) then this is DMA, Change to LedPin 2 (D4) to use Uart. Haven't tried it yet, but will later to see if it plays nice. Wonder if using a ESP32 maybe an option?

Aircoookie, 2019-10-25:
Should be fixed now!

closed 2019-10-25 by Aircoookie