wled/WLED #156: two questionsss(ip adress and ws2811 strip)

richyxi, 2019-04-21:
hiii a have a little issue or a missunderstanding, first it's the ip, when i upload the code on the esp, first i type a static ip adress but when i turn on the lights it appears to go with an ip client different from the static one, how can i change that?

second, i saw that this project can work with ws2811 strips and tried to use it but with no [luck](url
[image]
)
i connected them in the GPIO2, any advice of how can i get them work?

thank u in advance:)

Aircoookie, 2019-04-21:
Hi, for IP please make sure you filled the subnet and gateway fields correctly and that the static ip is actually in the range of your network (most use `192.168.X.X` or `10.10.1.X`).

About the LEDs, they seem to light up in the correct color, just too dim. Probably you forgot to set the auto-brightness limiter in LED settings to the rating of the 5V power supply you use for driving the LEDs. That is a must with such a large spool as yours (i'm guessing it is 5m). Let me know if this solves the issue, otherwise there are further troubleshooting tips!

richyxi, 2019-04-22:
thanks a lot! i fix my issue with the ip adress, and for the strips i realized that the ws2811 works with 12v and not 5, so thats why it so dim so i get them working but with ws2812b and works freaking good! awsomme project and thks for the feedback, cheers!

closed 2019-04-22 by richyxi