wled/WLED #157: Domoticz and MQTT

salopette, 2019-04-21:
Hii!
Yesterday I came across the project randomly and of course immediately realized! It's really great, I love something like that !!!
I use Domoticz as a home assistant. My question is, can the project be controlled via MQTT and Domoticz?

https://www.domoticz.com/wiki/MQTT

THX

Aircoookie, 2019-04-21:
Hi!
I'm happy that you like WLED!
Controlling it via MQTT is possible, but I'm afraid it doesn't support the Domoticz protocol.
https://github.com/Aircoookie/WLED/wiki/MQTT

I sadly cannot add support for every protocol and platform since memory resources on the esp8266 are so limited... You could try something like NodeRED as an adapter between WLED and Domoticz though.

frustreermeneer, 2019-10-14:
@Aircoookie @salopette 
I'm working on a WLED plugin for Domoticz. 
Maybe you can give it a try: https://github.com/frustreermeneer/domoticz-wled-plugin

salopette, 2019-10-16:
Thanks, I will test it on the weekend !!!

closed 2020-04-05 by stale[bot]