wled/WLED #158: Can't access setting pages in own build. Official binary release works.

fiechr, 2019-04-22:
Since I wanted to add some custom code to the binary I cloned the repo in Visual Studio Code. Before changing anything at all, I only uncommented env_default = nodemcuv2 (ESP3266 board is a "Lolin NodeMCU v3") and started a build to test if everything works for default settings.
The build finish with "[SUCCESS]". After uploading to the chip and resetting it, the wifi station mode is running and allows devices to connect - which also works fine. Accessing the first page and main page works fine, but as soon as I want to access any settings (like WiFi or LED settings) the page loads forever and the connection is reset after a while.

Here is the log of the build process: https://pastebin.com/qi4mAk1W

I tried different dev boards (same and similar) - same problem.

I then downloaded the latest binary release, flashed it with esptool and everything works as expected.

What could be the issue? Shouldn't - in theory - the same libs and code produce basically the same binary? Or was the official release built with different default settings?

fiechr, 2019-04-22:
Selecting arduino core 2.5.0 instead of 2.4.2 seems to solve the issue for me.

Aircoookie, 2019-04-22:
Hi!
Settings pages not working often happens when the esp runs out of heap memory - with how many features WLED has, that unfortunately can occur in some circumstances.

2.5.0 has optimized core library memory usage, so that explains why it now works for you. Unfortunately, there is an [issue](https://github.com/esp8266/Arduino/issues/5784) with it which causes some ESPs to become unresponsive, so I still have to recommend 2.4.2.

Thank you for your feedback! So am I right that this happened while connected to WLED-AP without any settings in the web UI actually changed? In that case, the ESP shouldn't really be able to run out of heap. Just tried to reproduce the issue using PIO in VSC, nodemcuv2 target works just fine though.

aenertia, 2019-06-18:
Sometimes the spiff's doen't get uploaded/written with platform IO. 

try running:

$pio run --target uploadfs

I have also noticed the eprom settings get retained in situations where I would not expect them too (i.e setting a new AP password string in the src)

fbnts, 2019-06-19:
Hi, I am wanting to try out WLED (to replace a custom MQTT NeoPixel RBGWW implementation) but I am having the same problem.

My LEDs are on pin 14 so I need to compile it myself.  I am using 2.5.2 ESP8266 core.

When I click on any settings pages the ESP crashes.  I have tried pasting the Exception into the Exception Decoder but it doesn't return anything.

Is there anything else I can try?

fbnts, 2019-06-19:
It looks like I have solved it by downgrading to 2.5.0 core

Aircoookie, 2019-07-15:
I believe I know what is causing this issue (as well as a related one where the HTTP API doesn't send a valid response). I will publish a fix in a few days time!

closed 2019-10-25 by Aircoookie