wled/WLED #159: MQTT connection is lost after multiple hours of inactivity

StefanRu1, 2019-04-23:
Hi,
i am facing an issue with the MQTT connection.

I am using Mosquitto right now but also used FHEM MQTT server before.

On both servers i have the problem that after some hours of inactivity (> 12h) of the LED stripe the WLED client is disappearing from the server and will not come back when using it again. 
It is also not possible to send MQTT messages to the devices.

Is it possible that the MQTT client crashes on the ESP8266?
I am using a Wemos D1 Mini and the latest WLED Binaries 0.8.4.

So it was not possible to get the connection back to life. Not by sending MQTT messages from the server and not by changing values on the WLED.

The only way to get the client back, is to enter the "settings" tab on the Wled, go to "security settings" and press "Save & Reboot".

After the reboot the Wled is directly recognized again.
Did someone had such issues before?
Can i somehow monitor this?

Thanks and best regards,
Stefan

Aircoookie, 2019-04-24:
Hi!
I easily can imagine this happening because the MQTT doesn't yet re-connect, meaning that if the connection is broken due to any problem with the ESP, WiFi, network routing or broker, it only reconnects when it is rebooted. I will try and implement a reconnection logic soon!

StefanRu1, 2019-04-25:
Ah, ok. 
That explains my problem very good. 
Perhaps i shortly lost WIFI in between.

So reconnect feature would be great.

Thanks a lot,
Stefan

Aircoookie, 2019-05-17:
Finally an update on this, sorry for taking so long! I've implemented the MQTT reconnection logic! Still need to do some testing, but expect the fix this weekend!

StefanRu1, 2019-05-17:
Oh perfect!
Thank you so much.

Have a nice Weekend...

StefanRu1, 2019-05-21:
Is there something i can test for you?
I can easily reproduce the problem and can tell you if it is fixed.
Also i control the stripes with a motion sensor via my smarthome with help of MQTT. 
So i can also do a excessive testing of MQTT.

Best regards,
Stefan

Aircoookie, 2019-05-21:
Just pushed the reconnect to master! I hope it will solve the issue for you :)

StefanRu1, 2019-05-22:
Thanks again.
I have flashed the new build. Looks good i will do a bit more testing and write back.
Can you tell my why you changed the LEDPIN in NpbWrapper_h from 2 to 5?
This was a bit confusing ;-)

Thanks and best regards,
Stefan

Aircoookie, 2019-05-22:
Good to hear!
Oh,  I have a device using pin 5 and i really need to remember to change this sort of this sort of thing back before pushing ;) Sorry for the confusion!

StefanRu1, 2019-05-22:
No problem! 
There was a clear message in the buildlog telling that pin 5 might cause flickering.
That was the point I recognized something is not as expected 😊
I will write a feedback how it works as soon as a has time to test it excessive. 

Thanks, 
Stefan

StefanRu1, 2019-05-24:
It works perfect. 
Even after MQTT Server restarts the LED is sending and receiving MQTT messages.

Thank you so much!

closed 2019-05-24 by StefanRu1