wled/WLED #162: Assistance with Custom Code

fish3602, 2019-05-05:
Looking for assistance with implementing some custom functionality.  Any help appreciated!

Project is an edge-lit acrylic sign for a sailing club, Community Boating in Boston.  They rate wind speed similar to a traffic signal, Green, Yellow, Red for Light -> Strong wind.  Rating is displayed as a Flag Color on their website.  Goal is to light up the sign with the current flag color (wind speed rating) per the website.

I've implemented this already in python on a raspberry pi.  Looking to port the functionality to C for the ESP-01 and WLED.

Draft code is below.  First issue is with the urllib.urlopen function, likely need to move to a HTTP GET function of some sort for C.  Please advise on that and any other issues noted.  This code is being implemented in wled06_usermod.ino  

long lastTime = 0;
int delayMs = 1000*60*2; //we want to do something every 2 min
char response;

void userLoop()
{
  if (millis()-lastTime > delayMs)
  {
    lastTime = millis();
    response = urllib.urlopen('https://api2.community-boating.org/api/flag')  //get CBI flag color
    html= response.read()
    
    if html == 'var FLAG_COLOR = "C"': //Closed = White
        col[]{255, 255, 255};
    elif html == 'var FLAG_COLOR = "G"': //Green
        col[]{0, 255, 0};  
    elif html == 'var FLAG_COLOR = "Y"': //Yellow
        col[]{255, 255, 0};  
    elif html == 'var FLAG_COLOR = "R"': //Red
        col[]{255, 0, 0}; 
    else:
        col[]{0, 255, 255};   //Purple = ERROR

    colorUpdated(2)  //direct change
             
  }
 
}

Aircoookie, 2019-05-06:
Hi!
Cool project! You should be able to do this quite easily using the `HTTPClient` library. (here's an [example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino) on how to send a GET request)
Let me know how it goes and if you need further help :)

fish3602, 2019-05-07:
Thanks very much for the pointers!  Will let you know how it turns out.

closed 2019-10-25 by Aircoookie