wled/WLED #163: Deactivating UDP realtime while hyperion.ng sending data causes WLED to hang

enriquefernandez, 2019-05-06:
Hello,

Whenever the UDP realtime setting is active and hyperion.ng is sending data, everything works fine.
However, some times I want to deactivate UDP realtime to take back control of the strip without touching hyperion (which is still sending data and thinking that the strip is still there). Whenever I try to do that, it looks like WLED hangs.
I have tried deactivating the setting using both the settings page and the `&RD=0` HTTP call. I both cases, it seems like WLED stops responding to the hyperion data and shows the last color I had selected. However, I cannot do anything after that. I cannot access the webserver or control the strip in anyway (the website never loads and looks like it stops responding). The only way to take back control is to restart the ESP with the reset button.

This only happens when hyperion is sending data. If hyperion is not running, I can enable and disable the UDP realtime setting with either the settings page or the HTTP API call without issues. Is the UDP data that hyperion.ng is sending making the esp hang somehow?

I'm using an ESP32, and not having any other issues besides that. I also have MQTT enabled (but wasn't using it directly the times this has happened).

Any ideas?
Thanks!

Aircoookie, 2019-05-06:
Hi!
I'll need to look into this, it seems like turning off realtime with `RD=` works, but the esp hangs if there are some incoming packets which are ignored.

closed 2020-04-05 by stale[bot]

musicaways, 2020-04-16:
Same problem for me! Any solution?