wled/WLED #164: Enable MQTT overrider while UDP-stream

YeonV, 2019-05-06:
Hey mate,

you added the feature to keep the UDP stream "alive".
While the Web-UI is still working for changing colors, you cannot use MQTT to change color anymore.
Any possibilities to make WLED still react to MQTT, while the UDP-stream is set to 255?

Cheers
Blade

Aircoookie, 2019-05-06:
Need to look into it, changing colors via MQTT if UDP is retained with 255 (but not actively sending) should absolutely work. Does MQTT work again if you change the color via the web UI once? (until you send new UDP packets)

YeonV, 2019-05-06:
Hey mate, 
thank you soo much for answering me!!

No the color was not changing.
After rebooting all strips, its working again.
I bet it is the MQTT-connection-lost-bug im running into.
A reconnect on connection-lost should solve that.

I saw this other issue:
https://github.com/Aircoookie/WLED/issues/159

Shall we close this one as it is not due to the UDP stream?

Aircoookie, 2019-05-06:
Yes, I'm pretty sure it is #159, as well. I'm prioritizing adding the MQTT reconnect function now :)

closed 2019-05-06 by Aircoookie