wled/WLED #165: MQTT On/Off/Toggle

DuschdrBabbe, 2019-05-06:
Hi,
my mqtt command ON is not working anymore... toggle also not working (just OFF not ON) OFF is working...
I must set the brightness so switch the light on...

HTTP Api listed in Wiki didnt match 
(WIKI) &T= | 0, 1, or 2 | Master On/Off/Toggle 
(in Real) 0=off 1=on t=toggle this works

I have used the latest bin (WLED_0.8.4_ESP8266.bin) from here...

Greetings...

Pascal

Aircoookie, 2019-05-06:
Hi,
I believe there might have been a bug at the time the binary was made. I'll send you a new binary soon!

Aircoookie, 2019-05-24:
Sorry, I totally forgot to send this to you until I remembered because another user requested a new binary today :( Hope the wait wasn't too inconvenient for you and that this binary will solve the issue!

Here you go!
https://github.com/Aircoookie/WLED/issues/172#issuecomment-495752717

DuschdrBabbe, 2019-06-04:
Hi Funktioniert! 

Grüßle

(Issue kann geschlossen werden)

closed 2019-06-04 by Aircoookie