wled/WLED #167: Logitech harmony support

mentaluproar, 2019-05-18:
Since I was able to tie this into HomeKit. I was wondering if there’s a way to tie it into the Logitech harmony system as well.

Aircoookie, 2019-05-21:
Hi! I'm not very familiar with Logitech Harmony, but that should be possible with IFTTT: https://github.com/Aircoookie/WLED/wiki/Remote-Access-and-IFTTT

It may be that the Harmony has a local API to interface it directly. But in that case you'd need some additional code to do what you need.

brentbrooks70, 2019-05-29:
I would check out these posts that discuss the closing of the local API last December
[](https://community.logitech.com/s/question/0D55A00008D4bZ4SAJ/harmony-hub-firmware-update-fixes-vulnerabilities)
 and the optional firmware path you can opt-into going forward 
[](https://community.logitech.com/s/question/0D55A00008OsX3CSAV/update-to-accessing-harmony-hubs-local-api-via-xmpp)
that allows an XMPP local API interface.

closed 2020-04-05 by stale[bot]