wled/WLED #168: Crash on Wifi connection

StormPie, 2019-05-21:
Hey!

Keen to get back into some WLED goodness.

I went and got the latest version, uploaded to my board but when connecting I got "Error Loading" in the top left of the Mobile UI.

I then enabled Debug and can see it's crashing and restarting before it loads any presets/palettes 

Any ideas? Below is the console readout.

---WLED 0.8.4 1904141 INIT---
esp8266 2_4_2
heap 32664
LEDs inited. heap usage ~640
Load EEPROM
CSSID: NETGEAR20-5G USING AP8
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
Can't connect. Opening AP...

Connected! IP address: 0.0.0.0
HTTP server started
Ada
---DEBUG INFO---
Runtime: 11055
Unix time: 11
Free heap: 26800
Wifi state: 0
State time: 11055
NTP last sync: 999000000
Client IP: 0.0.0.0
---DEBUG INFO---
Runtime: 21063
Unix time: 21
Free heap: 26352
Wifi state: 0
State time: 11055
NTP last sync: 999000000
Client IP: 0.0.0.0
---DEBUG INFO---
Runtime: 31067
Unix time: 31
Free heap: 25800
Wifi state: 0
State time: 11055
NTP last sync: 999000000
Client IP: 0.0.0.0

Exception (3):
epc1=0x40228a50 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025b653 depc=0x00000000
[... 292 lines cut ...]
3ffffec0:  00857a00 42803333 3fff1698 402088bd  
3ffffed0:  00857a00 3fffdad0 3fff1ea8 00000000  
3ffffee0:  00000000 42803333 3fff1698 40208d48  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  04340737 0000008f 3fff1698 40209110  
3fffff20:  40201dd1 4bc6a7f0 000080b4 3fff1ea8  
3fffff30:  00000000 4bc6a7f0 000080b4 3fff1ea8  
3fffff40:  00000000 00000000 4bc6a7f0 00000000  
3fffff50:  00000000 00000000 3ffe8970 00000000  
3fffff60:  3fffdad0 3fff1ea8 402252bc 3fffefb0  
3fffff70:  3fffdad0 00000000 3fff1ea2 40225376  
3fffff80:  3fffdad0 00000000 3fff1ea2 4021a70a  
3fffff90:  40227078 00000000 00000004 3fff1ea8  
3fffffa0:  00000000 00000000 00000001 3fff1ea8  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Aircoookie, 2019-05-21:
Hi, welcome back!

Seems like your module crashes when it sends the current state to the UI.
You can try two things:
- Switch between LwiP 1.4 and 2 (compile setting), this solved a similar issue for another user
- reset the board ("Erase Flash: All flash contents" compile setting)

I hope it'll help :)

StormPie, 2019-05-22:
Hey @Aircoookie :)

That reset of flash contents is a great trick, thanks for that. 

This time WLED was responsive in the desktop UI, but as soon as I went to the mobile UI or the settings page in the desktop UI it crashed and rebooted. I've uncommented all the options to disable features to reduce memory (except mobile UI) and I still get the crash.

Switching between LwiP 1.4 and 2 didn't seem to make a difference in this case.

I don't know how to select the arduino core (2.5, 2.4.2 etc) so maybe I am using the wrong one?

Thanks heaps for the help!

Aircoookie, 2019-05-23:
Wow, I thought that would help. Sorry for the issue :)
I can see from the log that you already have the correct Arduino core installed (2.4.2)

Do you have a spare ESP to try? You can also use the [ESP exeption decoder](https://github.com/me-no-dev/EspExceptionDecoder) to get some helpful info about why the crash happens.

I hope we can get WLED running for you again :)

StormPie, 2019-05-27:
@Aircoookie I've been using the desktop UI for hours on end with no trouble but as soon as I swap to the mobile version, or try on my iPhone, the ESP reboots and after a little while the colour wheel may take a few inputs then it resets again.
It also never loads any of the effects or presets (probably because it only tries looking for them on load, which is when the ESP has crashed).

I don't have a spare ESP at the moment, is it likely to be faulty if it works so well with the desktop UI? Also, I've been messing with the music syncing program I made using WLED HTTP API calls and it works with no issues.

I tried the decoder but really don't know how to interpret what I am seeing as a cause haha! Here it is incase you can decipher anything:

`
Connected! IP address: 192.168.0.21
HTTP server started
mDNS responder started
Ada
---DEBUG INFO---
Runtime: 11196
Unix time: 11
Free heap: 26488
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 192.168.0.21
Not-Found HTTP call:
URI: /win&P1=1&P2=2
API req: /win&P1=1&P2=2

Exception (3):
epc1=0x402255f0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40256e2b depc=0x00000000

ctx: sys 
sp: 3fffeb30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffecd0:  00000400 3fff6500 3fffed30 3fff12d4  
3fffece0:  3ffe9d7f 3fff6674 3fff6674 3fff12d4  
3fffecf0:  3ffe9d7f 0000000d 3fff12d4 40213a49  
3fffed00:  3ffe9d7f 3fff6674 3fff6840 40213aa8  
3fffed10:  3ffe9d7b 3fff6674 3ffe9d75 3fff61c4  
3fffed20:  0000000b 3ffe9bdb 3fff6674 40213ca8  
3fffed30:  40256e2b 00000000 00000000 3fff61c4  
3fffed40:  3fff63d8 3fff5bc4 3fffed60 40213f28  
3fffed50:  00000000 00000000 00000000 4022189f  
3fffed60:  3fff176c 000006e5 000006e5 00000000  
3fffed70:  00000000 00000000 3fffedb0 401006dc  
3fffed80:  3fff6240 00000000 00000000 4022185c  
3fffed90:  00000002 3fffedb0 00000001 3fff5e73  
3fffeda0:  00000002 3fff5bc4 3fff5bc4 40213fc4  
3fffedb0:  00000000 00000000 00000000 4021f97c  
3fffedc0:  3fff5bd4 3fff5bc4 3fff4af4 4021fcac  
3fffedd0:  3fff59c4 3fff19fc 3fff5bf8 3fff5e73  
3fffede0:  00000002 00000002 3fff5bc4 4021e0d4  
3fffedf0:  00000002 00000002 3fff5bf8 40221bdc  
3fffee00:  3fff5e73 3fff5bc4 3fff5b74 3fff5e73  
3fffee10:  00000002 00000002 3fff5bc4 4021e36a  
3fffee20:  00000000 4021de16 3fff5bc4 40222a70  
3fffee30:  00000000 4bc6a7f0 00002ebd 3fff1a14  
3fffee40:  00000000 00000000 3fff5bf8 00000000  
3fffee50:  3fff5ae4 402221ac 40106772 00000000  
3fffee60:  3fff5ae4 3fff5ce4 00000000 4021e399  
3fffee70:  3fff5ae4 3fff5ce4 3fff19f4 4021bf1c  
3fffee80:  3fff59c4 00000001 3fff5414 401006dc  
3fffee90:  0000111c 3fff59c8 00000001 3fff1a14  
3fffeea0:  3fff59c4 3fff19f8 3fff19f4 4021bf60  
3fffeeb0:  3fff59c4 3fff19f8 3fff19f9 4022d6ec  
3fffeec0:  1500a8c0 00000000 00000000 00000000  
3fffeed0:  00000000 40231ec0 00000194 3fff5d02  
3fffeee0:  3fff1b6c 3fff5ce4 3fff1c54 4023164d  
3fffeef0:  00000014 000006e8 000006e8 3fff1b6c  
3fffef00:  3fffdc80 3fff39e4 3fff53b4 3fff3a2c  
3fffef10:  00000008 3fff1b6c 3fff5ce4 4022aab8  
3fffef20:  3fffdc80 3fff39e4 3fff53b4 4022a8f3  
3fffef30:  4024821a 3fff39e4 3fff53b4 4024822b  
[... 250 lines cut ...]
3ffffee0:  402222bb 3fffefb0 3fff5844 40219ec3  
3ffffef0:  3fff5808 00000000 3fff5584 40218417  
3fffff00:  00000000 00000002 0000016d 3ffe8964  
3fffff10:  00000000 4bc6a7f0 00002e87 3fff11bc  
3fffff20:  00000000 4bc6a7f0 00002ebc 3fff1624  
3fffff30:  00000000 00000000 4bc6a7f0 00000000  
3fffff40:  00000000 4bc6a7f0 00002ebc 3fff1624  
3fffff50:  00000000 00000000 4bc6a7f0 00000000  
3fffff60:  3fffdad0 00000000 40106772 71a9fbe7  
3fffff70:  00000000 00000000 3fff161e 3fff1624  
3fffff80:  3fffdad0 00000000 3fff161e 40218be7  
3fffff90:  40223d40 1500a8c0 00000004 3fff1624  
3fffffa0:  00000000 00000000 00000001 40222321  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

`

Aircoookie, 2019-05-27:
Ok, it seems like the ESP crashes (most likely running out of memory) when getting the `http://ipaddress/json` page containing the current light state (only the mobile UI partly uses the new JSON api, the desktop one still uses the old HTTP one).
Try opening that and see if it also leads to a crash! You can also try to update to this binary: https://github.com/Aircoookie/WLED/issues/172#issuecomment-495752717
If it still crashes with that binary, it is perhaps a hardware issue.
Sorry for all the trouble! :)

StormPie, 2019-05-28:
Yes, as soon as I load http://ipaddress/json the ESP crashes and reboots. Is it needed or can it be disabled somehow?

Unfortunately I have two tiny lines of code I add to UserMod to power up my LEDs so .bin files aren't super useful (unless they can be edited and I don't know how haha)

Aircoookie, 2019-05-28:
Unfortunately, yes, the json API is required in 0.8.4 because it loads the effects and palette lists dynamically (so you don't have to change the html if FX are changed). Also for stuff coming soon like segment support!
What pin are you using for powering up the LEDs? Just to make sure nothing is conflicting...
You can try the binary anyways (even if the LEDs won't power up), if it does crash again we know that the cause is some WLED setting or ESP hardware. If it works, it is your compiler somehow deciding to do something different then mine (probably a different version of some library, although I checked everything that could lead to such a crash). Sorry for bombarding you with questions like this, but how many LEDs do you have configured? And did you use the 0.8.4 release source or the latest github master? Let's get this working again! :)

StormPie, 2019-05-31:
@Aircoookie the .bin worked! I could use both UI's without any issue, except of course without my extra code the lights didn't come on.

I am using PIN 4, that must be an issue? Normally I make the following edit in each sketch version:
`//boot starts here
void setup() {
  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH);
  wledInit();
  }`

Powering up PIN 4 closes a switch which powers the LEDs and prevents any random flashes from ESP reboots.

StormPie, 2019-06-04:
@Aircoookie I have a hardware system and boards that all work assuming I can use PIN 4... can I change WLED's need for it or do I need to get my boards redone?

Aircoookie, 2019-06-04:
I'm very sorry! I did in fact reply to you, but it seems like GitHub has not submitted the comment for some reason... :/

No, using Pin 4 is absolutely fine! It is the default pin for the IR sensor though, so that might interfere (You can change or disable it in line 9 of NpbWrapper.h) It doesn't make sense though that it only crashes when requesting the /json.

If all else fails, i will send you a binary with your power up code once I get home :)

StormPie, 2019-06-12:
That would be very appreciated :)

siedi, 2019-07-28:
I pretty much have the same crash. Must be related to the json library and the way of using it. Maybe I do have a different version of the library, esp8266 code or whatsoever. Works with your prebuild binary, not with the latest code and compiling it myself (can start it, see the main page, but once I click on the settings, it crashes)

siedi, 2019-07-28:
It is related to this line https://github.com/Aircoookie/WLED/blob/master/wled00/wled18_server.ino#L308

With 2.5.0 it is working, with 2.5.2 not. Seems related to https://github.com/esp8266/Arduino/issues/6109

Changing the above mentioned line to:
`if (var == "SCSS") return String(FPSTR(PAGE_settingsCss));`
worked for me.

Aircoookie, 2019-08-20:
Thank you very much for the fix! Implemented into master just now and WLED seems to work just fine with the 2.5.2 core now.

closed 2019-10-25 by Aircoookie